Klipper3d/klipper #3568: ignore Bed Level sensor when far away from Bed

JustAnother1, 2020-11-22:
I use Klipper on a Delta printer. I have a piezo elctric bed level sensor somewhere above the nozzle. I do a G28 followed by DELTA_CALIBRATE. That failes after the first move towards the bed( The nozzel stops a few mm above the centre of the Bed. The Message I get from Klipper is : "!! Probe triggered prior to movement". 

That is correct.

The stop at the end of that first movement is rather abrupt so the piezo pressure sensor triggers. If I tune the sensor to be less sensitive then I get into issues that the impact on the bed is not registered. I think that Klipper should be able to ignore this signal as it knows that an impact is not possible as that height. (Or maybe it could decelerate more smoothly?)

To explain it more. The Sensor triggers better on fast pressure changes than on slow ones. So an abrupt stop and then a slow approach towards the bed doesn't work at all. If the sensitivity is low enough to ignore the stop then it is not high enough to detect the bed. And if it is high enough
 to trigger on the bed impact it will also trigger on the abrupt stop,...

[klippy.log](https://github.com/KevinOConnor/klipper/files/5580416/klippy.log)

JustAnother1, 2020-11-22:
I use:
FIRMWARE_VERSION:v0.9.1-43-gfcb78e50
On Octopi 0.18 (nightly)

Hywelmartin, 2020-11-23:
add a small pause before each probing
```
[probe]
...
activate_gcode:
      G4 P500
```
slow probing is actually better even with piezo.. recommended frpm Precision Piezo is "Probing Speed: 4-7mm/s"

JustAnother1, 2020-11-24:
@Hywelmartin I use the Precision Piezo sensor. I set the speed to 5mm and added the delay you proposed.

It helps a bit. The first move is now no issue any more. The algorithm always starts. But if I have the sensor too sensitive(I see it triggering on the first move) then I get detections before the bed gets hit.

When I make the sensor less sensitive I get timeouts on the detection. The sensor does not trigger even though it is pushed hard into the glass. Observing that behaviour got me thinking that a fast change is what makes the sensor trigger.

I suppose the sensor is just bad,....

closed 2020-11-24 by JustAnother1

Hywelmartin, 2020-11-25:
the tension of the sensor in combination with sensitivity is a balance act... sort of.