Klipper3d/klipper #3570: Motors do not move during homing // Z-Stepper move at all

SkyNotFound, 2020-11-23:
[klippy.log](https://github.com/KevinOConnor/klipper/files/5584011/klippy.log)

Hello dear community!

At first I'd like to introduce myself shortly since this is my first post here to provide a bit background.
I'm a 3d printing enthusiast since 2017 and started off with a Tevo Tarantula which was modified over the last few years until nothing of the original parts but the frame were left. Until now I was using Marlin but got intrigued by klipper's many features and benefits, so here I am, giving it a shot. But admittedly I have fairly limited knowledge or experience with hardware/software-programming besides the Arduino IDE. 
So far I ran in a few minor problems setting klipper up, but thanks to the well written documentation I was always able to resolve them myself. But this one is a bit different, so I'd like to ask you guys if you could point me in the right direction.

- - - 
The setup: 
- Raspberry Pi 3 B+
- MKS Gen 1.4
- 5x TMC2130
- 5x NEMA 17 1.8°/step Motors
- E3D Chimera+ Hotend and Titan extruders
- Cheap SD Card 2 Line Display with Button

To my problem:
When trying to home the printer, none of the axes will move, except the Z-motor making a "click" noise one time and then is going silent. I found out, that there's a possibility to force moves for diagnostic purposes - so I tried and to my pleasure found that the X and Y motor were indeed moving (even the right direction and correct distance), but the Z-motor did the same as with a homing command: Clicked one time, the Z-rod seemed to have turned about 3-4° or so and then did nothing.

At first I thought the pins were assigned wrong and double checked them, next I ruled out a calculation error (dist/step too low), but that seems to be both right. 

Any hint in the right direction would be much appreciated! As well as if you find anything else odd in the config that shouldn't be there or seems questionable - still a bit new in this metier and happy to learn from your experience!

Best regards from Germany,
Sky

KevinOConnor, 2020-11-23:
Sounds like the FAQ item at https://www.klipper3d.org/FAQ.html#i-converted-my-config-from-marlin-and-the-xy-axes-work-fine-but-i-just-get-a-screeching-noise-when-homing-the-z-axis

-Kevin

SkyNotFound, 2020-11-23:
Thank you very much for your fast reply!

I lowered the maximum Z speed to 1mm/s and the max acceleration to 5mm/s², but the motor still won't move, instead it seems like there is holding current flowing.

But albeit the X/Y motors move after a forced move command, they still perform no homing if promted - is this a normal klipper behaviour if the z-motor did not reach the end stop or should they be moving regardless what's wrong with the z-motor?

closed 2020-12-22 by github-actions[bot]