Klipper3d/klipper #3572: Skr 1.3 complete Pinout

goldjunge91, 2020-11-24:
Hi
i have on my skr. 1.3 a e3d pt1000 amplifier board on  my aux-1  connector,  pin is  P0_03_A6 
can i use this pin as normal pin for klipper ?

Roycinger, 2020-11-24:
Here's the pinout for the SKR 1.3:
https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/BTT%20SKR%20V1.3/hardware/SKR-V1.3-PIN.pdf

goldjunge91, 2020-11-24:
> Here's the pinout for the SKR 1.3:
> https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/BTT%20SKR%20V1.3/hardware/SKR-V1.3-PIN.pdf

I know the PIN number that I need, P0_03_A6 
And I can not find any description how I setup an analog pin. 

> Hi @goldjunge91,
>
> It did not look like there was a Klipper log file attached to this ticket. The log file has been engineered to answer common questions the Klipper developers have about the [...]

Hi,
I have no log file I have start to write my config before I flash my board

DeviousPenguin, 2020-11-25:
The AUX1 connector only has a serial link (TX0 and RX0)

goldjunge91, 2020-11-26:
on marlin i use the RX0 Pin (P0.03.A6) as analog pin and its work on my tronxy.

closed 2020-12-01 by klipper-gitissuebot