Klipper3d/klipper #3573: Multi samples support for safe_z_home

belotv, 2020-11-24:
Hello,

The current implementation of safe_z_home is quite limited by the fact it is based on a single measurement. There is little added value to allow samples in the bed mesh, while a bias in the homing can easily offset the first layer.

Most low cost probes have a standard deviation of which is not that bad, but the range is sometimes not great (>0.025mm). By allowing a safe_z_home based on an average of samples, it would be a huge improvement for probes that are not top quality.

It would, from my point of view, really make sense to add samples/sample_dist_retract options to the [safe_z_home] section, or to use the values from the [bltouch] section. Then, the zero would be defined based on the average value rather than a single point.

PS: Not attaching klippy.log report as the issue is not printer specific.

Kind regards

KevinOConnor, 2020-11-26:
An interested developer would need to implement and submit support for that.  The support is not simple because the homing code is notably different from the probing code (and has notably different requirements).

Separately, it is often possible to use z_tilt, bed_tilt, or bed_mesh after each home to obtain higher accuracy using those functions.

-Kevin

closed 2020-12-25 by github-actions[bot]

menelike, 2021-02-24:
I'd love to see that feature as well since the z offset occasionally is misaligned on homing (using a genuine bltouch).
Running calibration before each print is a valid solution, but not convenient as it just takes too much time to complete in comparison with a multi-sample probe on one spot like proposed by @belotv .

Esteban16899, 2021-11-09:
Hey! I would vote anywhere to reopen this ticket, it's a valid feature and I think would benefit a large group!

ntcong, 2021-11-26:
@Esteban16899 I took a look at the relevant code, it's probably a lot of work to implement multiple samples because of the difference in homing logic. 

PR is welcoming though, maybe someone with enough interest in the feature will invest some of their time in it, in the meantime I think it's best to get a better probing device.