Klipper3d/klipper #3574: BL-Touch v3.1 sensor pin on SKR v1.4 Turbo

morganfw, 2020-11-25:
Hello,
I've connected an original BL-Touch v3.1 to SKR v1.4 Turbo following this pinout schemes for SERVOS (yellow cable on 2.0 pin) and Probe (white cable on 0.10 pin) pins:
https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/BTT%20SKR%20V1.4/Hardware/SKR-V1.4-pinout.jpg
https://github.com/bigtreetech/BIGTREETECH-SKR-V1.3/blob/master/BTT%20SKR%20V1.4/Hardware/BTT%20SKR%20V1.4PIN.pdf

[image]
then configured BL-Touch recognition on printer.cfg
```
[bltouch]
sensor_pin: P0.10 #Probe
control_pin: P2.0 #SERVOS
z_offset: 2.95
```
when I start Klipper, the BL-Touch doesn't start and cannot be recognized (No Led on BL-Touch).

Analysing klippy.log that seems the 0.10 pins is used by **BUS_PINS_i2c2** (see following klippy.log):

> Starting serial connect
> Loaded MCU 'mcu' 84 commands (v0.9.1-43-gfcb78e50-20201122_164423-octopi-rpi3b-plus / gcc: (15:7-2018-q2-6) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907] binutils: (2.31.1-11+rpi1+11) 2.31.1)
> MCU 'mcu' config: BUS_PINS_i2c1=P0.1,P0.0 BUS_PINS_i2c0=P0.28,P0.27 BUS_PINS_i2c2=P0.11,**P0.10** BUS_PINS_i2c1a=P0.20,P0.19 BUS_PINS_ssp1=P0.8,P0.9,P0.7 STEP_DELAY=2 ADC_MAX=4095 BUS_PINS_ssp0=P0.17,P0.18,P0.15 RESERVE_PINS_USB=P0.30,P0.29,P2.9 CLOCK_FREQ=120000000 MCU=lpc1769 STATS_SUMSQ_BASE=256
> Sending MCU 'mcu' printer configuration...

No references about BL-Touch in klippy.log.

I've done another test trying to insert Pin 1.27 which is related to the Z-Endstop on sensor pin (black cable on GND and white cable on 1.27 pin):

[image]

```
[bltouch]
sensor_pin = P1.27 #Z-Endstop
control_pin = P2.0 #SERVOS
z_offset = 2.95
```

but BL-Touch cannot start and in klippy.log I see following:

> =======================
> Extruder max_extrude_ratio=0.266081
> Starting serial connect
> got {u'count': 317, '#receive_time': 1530.520047541, u'sum': 351322, u'sumsq': 3119247, '#name': u'stats', '#sent_time': 1530.517195614}
> Loaded MCU 'mcu' 84 commands (v0.9.1-43-gfcb78e50-20201122_164423-octopi-rpi3b-plus / gcc: (15:7-2018-q2-6) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907] binutils: (2.31.1-11+rpi1+11) 2.31.1)
> MCU 'mcu' config: BUS_PINS_i2c1=P0.1,P0.0 BUS_PINS_i2c0=P0.28,P0.27 BUS_PINS_i2c2=P0.11,P0.10 BUS_PINS_i2c1a=P0.20,P0.19 BUS_PINS_ssp1=P0.8,P0.9,P0.7 STEP_DELAY=2 ADC_MAX=4095 BUS_PINS_ssp0=P0.17,P0.18,P0.15 RESERVE_PINS_USB=P0.30,P0.29,P2.9 CLOCK_FREQ=120000000 MCU=lpc1769 STATS_SUMSQ_BASE=256
> Sending MCU 'mcu' printer configuration...
> Configured MCU 'mcu' (1024 moves)
> Starting heater checks for heater_bed
> **Failed to verify BLTouch probe is raised; retrying.**
> **Failed to verify BLTouch probe is raised; retrying.**
> **BLTouch raise probe error: BLTouch failed to raise probe**
> Starting heater checks for extruder

Can you check if the firmware level pin mapping is wrong?
And maybe I could have burned the BL-Touch with this wrong mapping?

morganfw, 2020-11-25:
Following log is from Klipper with no BL-Touch installed and configured:
[klippy_no_bltouch.log](https://github.com/KevinOConnor/klipper/files/5598450/klippy_no_bltouch.log)

Following log is from Klipper with BL-Touch installed on 0.10 pin on SKR V1.4 Turbo board:
[klippy.log](https://github.com/KevinOConnor/klipper/files/5598495/klippy.log)

morganfw, 2020-11-27:
After retrying connecting BL-Touch v3.1 to Z-Stop pin 1.27 i'm getting following error:

> Unable to read tmc uart 'stepper_x' register IFCNT

but if I try to disconnect BL-Touch cables, all working fine, klippy.log attached
[klippy.log](https://github.com/KevinOConnor/klipper/files/5608632/klippy.log)

I'm on Raspberry Pi 3B+ with Raspbian 10 OS

>$ uname -a
>Linux octopi-rpi3b-plus 5.4.77-v8+ #1371 SMP PREEMPT Tue Nov 17 13:39:18 GMT 2020 aarch64 GNU/Linux

>$ cat /etc/issue.net
>Raspbian GNU/Linux 10

>$ python -V
>Python 2.7.16

>$ python3 -V
>Python 3.7.3

May help me to resolve issue?
Thanks in advance

paulhew69, 2020-11-28:
Hi
I use the Z-Stop input instead of the probe input, successfully.
I have seen many people ask, then swap to the Z-Stop and it works.

P.

morganfw, 2020-11-28:
> Hi
> I use the Z-Stop input instead of the probe input, successfully.
> I have seen many people ask, then swap to the Z-Stop and it works.
> 
> P.

Thank you for answer, I tried to connect to pin 1.27 as well, but no way, Klipper returns the following error (see my last post):

`Unable to read tmc uart 'stepper_x' register IFCNT`

The BL-Touch I own is following: https://www.creality3dofficial.com/collections/bl-touch/products/creality-bl-touch?variant=31412867498057 for Creality V1 Mainboard.

Could it be that the BL-Touch is modified and therefore not compatible with SKR V1.4 Turbo?

morganfw, 2020-11-28:
After removed the connector pins on the cable from the plastic housing, checked them and then reinserted, I've reconnected plugs on board and now like magic the BL-Touch works, maybe a defective plug on the cable.

0.10 pin (Probe) doesn't seem to make the BL-Touch work well, so I connected to pin 1.27 which is the z-stop one and everything works perfectly!

closed 2020-11-28 by morganfw