Klipper3d/klipper #3578: NoSectionError: No section: 'mcu'

MacGyverBeta, 2020-11-26:
[klippy.log](https://github.com/KevinOConnor/klipper/files/5601548/klippy.log)

Hello together,

I did update the latest version of klipper on my RaspberryPi and updated firmware on my SKR mini e3 V1.2.
Since then my printer.cfg file seems to be invalid... (see log file)
"NoSectionError: No section: 'mcu'"

I hope you can help me.

Here is my printer.cfg

[stepper_x]
step_pin: PB13
dir_pin: !PB12
enable_pin: !PB14
step_distance: .0125
endstop_pin: ^PC0
position_endstop: 0
position_max: 235
homing_speed: 50

[tmc2209 stepper_x]
uart_pin: PB15
microsteps: 16
run_current: 0.580
hold_current: 0.500
stealthchop_threshold: 250

[stepper_y]
step_pin: PB10
dir_pin: !PB2
enable_pin: !PB11
step_distance: .01249875
endstop_pin: ^PC1
position_endstop: 0
position_max: 235
homing_speed: 50

[tmc2209 stepper_y]
uart_pin: PC6
microsteps: 16
run_current: 0.580
hold_current: 0.500
stealthchop_threshold: 250

[stepper_z]
step_pin: PB0
dir_pin: PC5
enable_pin: !PB1
step_distance: 0.002488
endstop_pin: ^PC2
position_endstop: 0.0
position_max: 250

[tmc2209 stepper_z]
uart_pin: PC10
microsteps: 16
run_current: 0.580
hold_current: 0.500
stealthchop_threshold: 5

[extruder]
step_pin: PB3
dir_pin: !PB4
enable_pin: !PD2
step_distance: 0.002214
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: PC8
sensor_type: EPCOS 100K B57560G104F
sensor_pin: PA0
control: pid
pid_Kp: 29.795
pid_Ki: 2.048
pid_Kd: 108.380
min_temp: 0
max_temp: 250
max_extrude_only_distance: 1000.0

[tmc2209 extruder]
uart_pin: PC11
microsteps: 16
run_current: 0.650
hold_current: 0.500
stealthchop_threshold: 5

[heater_bed]
heater_pin: PC9
sensor_type: ATC Semitec 104GT-2
sensor_pin: PC3
control: pid
pid_Kp: 54.027
pid_Ki: 0.770
pid_Kd: 948.182
min_temp: 0
max_temp: 130

[fan]
pin: PA8

[mcu]
serial: /dev/serial/by-id/usb-Klipper_stm32f103xe_35FFD6054153393022590557-if00 
restart_method: rpi_usb

[printer]
kinematics: cartesian
max_velocity: 300
max_accel: 3000
max_z_velocity: 5
max_z_accel: 100

[static_digital_output usb_pullup_enable]
pins: !PC13

[board_pins]
aliases:
    # EXP1 header
    EXP1_1=PB5, EXP1_3=PA9,   EXP1_5=PA10, EXP1_7=PB8, EXP1_9=<GND>,
    EXP1_2=PB6, EXP1_4=<RST>, EXP1_6=PB9,  EXP1_8=PB7, EXP1_10=<5V>

[neopixel my_neopixel]
pin: PC7
chain_count: 9
initial_RED: 0.0
initial_GREEN: 0.0
initial_BLUE: 1.0

[virtual_sdcard]
path: ~/.octoprint/uploads/

[display]
lcd_type: st7920
cs_pin: EXP1_7
sclk_pin: EXP1_6
sid_pin: EXP1_8
encoder_pins: ^EXP1_5, ^EXP1_3
click_pin: ^!EXP1_2
#kill_pin: ^!EXP2_8

[output_pin beeper]
pin: EXP1_1

[bed_screws]
screw1: 32,32
screw2: 203,32
screw3: 203,203
screw4: 32,203
screw5: 117,117

[pause_resume]

[gcode_macro M600]
default_parameter_X: 50
default_parameter_Y: 0
default_parameter_Z: 10
gcode:
    PAUSE
    M25
    G91
    G1 E-.8 F2700
    G1 Z{Z}
    G90
    G1 X{X} Y{Y} F3000
    G91
    G1 E-100 F1000

[gcode_macro RESUME_PRINT]
gcode:
    RESUME
    M24

Arksine, 2020-11-26:
The log is showing an empty config file.  So either file at `/home/pi/klipper_config/printer.cfg` is empty or you have some whitespace errors in that file have resulted in a parsing issue.

MacGyverBeta, 2020-11-26:
@Arksine thank you, you were right... supid me. The klipper startup script did load another printer.cfg - and my correct one was just shown in octoprint. - Works again!

closed 2020-11-26 by MacGyverBeta