Klipper3d/klipper #3579: Klipper SKR PRO v1.2 with TMC5160 drivers Unable to connect

evandene, 2020-11-26:
DELTA printer configuration
Controller board: SKR PRO v1.2
stepper drivers TMC5160 (Copied pin numbers from TMC2130)
BLTOUCH
ReprapDiscount full graphic smart controller

Try to connect to octopi.
The STATUS command gives me:
Recv: // Unable to connect
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: // Error configuring printer
Recv: // Klipper state: Not ready
Recv: !! Unable to connect

I tried may things but this old guy can't fix it.
Please help me

[KLIPPER_klippy and config.zip](https://github.com/KevinOConnor/klipper/files/5602091/KLIPPER_klippy.and.config.zip)

evandene, 2020-11-26:
Sorry
I thought the zip file did contain the klippy.log file plus the configuration file.
Please find these files separately attached.

[klippy.log](https://github.com/KevinOConnor/klipper/files/5602677/klippy.log)

[printer.zip](https://github.com/KevinOConnor/klipper/files/5602685/printer.zip)

Sineos, 2020-11-26:
The 5160 config on my SKR 1.4 looks something like:

[tmc5160 stepper_y]
cs_pin: P1.9
spi_software_miso_pin: P0.5
spi_software_mosi_pin: P1.17
spi_software_sclk_pin: P0.4

Going by the SKR Pro 1.2 pin diagram (https://github.com/bigtreetech/BIGTREETECH-SKR-PRO-V1.1/blob/master/SKR-PRO-V1.2/manual/SKR-PRO-V1.2PIN.PDF) you should use following pins:

[image]

E.g.:

[tmc5160 stepper_x]
cs_pin: PA15
spi_software_miso_pin: PC11
spi_software_mosi_pin: PC12
spi_software_sclk_pin: PC10

closed 2020-12-25 by github-actions[bot]