Klipper3d/klipper #3581: Getting i2c for a ssd1306 to run on a SKR Mini V2.0

AlessandroEmm, 2020-11-26:
Hey there

So I tried getting this to work but I always end up in a i2c comm error, which makes sense since default pins are set to the values for 1.2 board it seems:
https://github.com/KevinOConnor/klipper/blob/5773654a00443dc5e1b6 e3ce195c48a5688c3aac/src/stm32/i2c.c#L25
-> 
[image]

Whereas on the 2.0 it's PA15/ and PB15
https://user-images.githubusercontent.com/31118252/85957631-ed71a800-b95c-11ea-9776-2bb7f3304a1d.png

So how would I change this via config (given its the actual problem)? I tried changing the source and recompiling but still would run into the same problem. Display-config is bone stock.

Thanks a lot
Alessandro

AlessandroEmm, 2020-11-27:
MCU 'mcu' config: BUS_PINS_i2c1=PA15,PB15 RECEIVE_WINDOW=192 BUS_PINS_i2c2=PB10,PB11 BUS_PINS_spi2=PB14,PB15,PB13 SERIAL_BAUD=250000 INITIAL_PINS=!PA14,!PC15 STEP_DELAY=2 ADC_MAX=4095 BUS_PINS_spi3=PB4,PB5,PB3 CLOCK_FREQ=72000000 BUS_PINS_spi1=PA6,PA7,PA5 MCU=stm32f103xe STATS_SUMSQ_BASE=256 RESERVE_PINS_serial=PA3,PA2

So apparently setting the pins worked.

[https://pastebin.com/bcSH9MuY](klippy.log)

AlessandroEmm, 2020-11-27:
So apparently by looking into other STM32 pinouts (https://www.electronicshub.org/wp-content/uploads/2020/02/STM32F103C8T6-Blue-Pill-Pin-Layout.gif)  PA15/PB15 are digital, so i changed to PB8/9 which seem to be aliases for the regular 1st. Unfortunately with no change, still getting the timeout. I must be understanding something basic wrong, please enlighten me. 👍 

[klippy (5).log](https://github.com/KevinOConnor/klipper/files/5606406/klippy.5.log)

KevinOConnor, 2020-11-27:
I2c requires using pins that have hardware i2c support - https://www.klipper3d.org/FAQ.html#do-i-have-to-wire-my-device-to-a-specific-type-of-micro-controller-pin

Klipper doesn't currently support the PB8/9 aliases (and changing the pins in the C code is not sufficient to add support).  So, you'll either need use PB6/7 or PB10/11 or a developer will need to add support for PB8/9.

-Kevin

AlessandroEmm, 2020-11-27:
Hi Kevin

Thanks - i see that this board, unlike the 1.2 doesn't provide i2c easily. What would a developer need to do for PB8/9 to become an option? (Trying to asses whether I would have enough skill to do it).

Assuming I would want to control the display through the raspberrypi host: Would i need to configure specifc pins in that mode too or will klipper default to the only available two i2c pins of the raspberry anyway?

Thanks,
Alessandro

AlessandroEmm, 2020-11-30:
So the latter question i was able to answer myself, sorry I didnt take the doc for setting up a linux-process based klipper instance as a posibility for the host, when there is actual documentation at https://github.com/KevinOConnor/klipper/blob/3d36ab6841811c6de8535c2705edac9cdd683a7b/config/sample-raspberry-pi.cfg#L28

So the issue can be closed, but i'd still be interested in what would be required to make 8/9 working. 

Thanks again!
Alessandro

KevinOConnor, 2020-12-03:
It would be necessary to download the stm32f103 developer documentation, determine how to enable the PB8/9 alias, and then implement it.

-Kevin

AlessandroEmm, 2020-12-05:
Thanks, will look into it. So this issue can be closed.

Lefuneste83, 2021-01-01:
I have the same issue here, trying to set up a 0.96" OLED screen on a Voron V0 with a SKR Mini E3 V2.0. The proposed wiring of this mod does not work with SKR Mini E3 V2.0

https://github.com/VoronDesign/VoronUsers/tree/master/printer_mods/madros/V0_skirt_OLED

Equivalent wiring on this V2.0 board requires the use of PA15/PB15 in place of PB6/PB7. When configuring the display I get an I2C communication error and a MCU Shutdown.

If I understand correctly, the only way to have this screen to work is to wire it directly to the Raspberry Host with I2C enabled on GPIO headers and declare i2c_mcu: host

This is quite a long shot for activating 2 wires...

AlessandroEmm, 2021-01-09:
@Lefuneste83  yeah I looked into how to activate/use the alternative i2c pins, its not too hard. but attaching to the raspy is so easy that i stopped investigating/developing. Especially since changing the pins may stop other thing on the SKR from working.

Lefuneste83, 2021-01-09:
What's annoying is that it completely changes the way you wire an OLED screen from a V1.2 board. I agree that adding a secondary Pi MCU is not such a difficult thing for us, but many beginners will be puzzled by the procedure as it is scattered among various pages (cfg config, process implementation, make flash of virtual device...). I2C is such an elegant and easy protocol to implement,l it would be great to be able to use the one provided by the SKR board.

closed 2021-02-07 by github-actions[bot]

t3chguy, 2021-05-09:
To wire up an SSD1306 on the SKR E3 Mini v2.0 you can use PB8 (SCL), PB9 (SDA) on the EXP header and in config use

```
[display]
lcd_type: ssd1306
i2c_bus: i2c1a
```

the `i2c1a` bus is the alternate mapping as specified in the datasheet

[image]

https://www.st.com/resource/en/reference_manual/cd00171190-stm32f101xx-stm32f102xx-stm32f103xx-stm32f105xx-and-stm32f107xx-advanced-arm-based-32-bit-mcus-stmicroelectronics.pdf

M1dn1ghtN1nj4, 2021-06-11:
I'm still trying to get this working with mine.  Same hardware and setup.  But I just get an i2c timeout and shutdown.  I think there needs to be some pin definitions somewhere in addition to the code posted above.

SemenHHH, 2021-07-04:
In file: klipper/src/stm32/i2c.c

i2c parametrs:
DECL_ENUMERATION("i2c_bus", "i2c1", 0);
DECL_CONSTANT_STR("BUS_PINS_i2c1", "PB6,PB7");
DECL_ENUMERATION("i2c_bus", "i2c1a", 1);
DECL_CONSTANT_STR("BUS_PINS_i2c1a", "PB8,PB9");
DECL_ENUMERATION("i2c_bus", "i2c2", 2);
DECL_CONSTANT_STR("BUS_PINS_i2c2", "PB10,PB11");

engpedroamorim, 2021-07-11:
klipper/src/stm32/i2c.c is exactly as that excert.  i'm connecting the display to the exp1 port pins PB8 and PB9 i'm using a rpi zero w serial uart by the TFT pins of the skr. always get an error: i2c timeout.
 [display]
lcd_type: ssd1306
i2c_bus: i2c1a
i2c_mcu = mcu

M1dn1ghtN1nj4, 2021-07-12:
> 
> 
> klipper/src/stm32/i2c.c is exactly as that excert. i'm connecting the display to the exp1 port pins PB8 and PB9 i'm using a rpi zero w serial uart by the TFT pins of the skr. always get an error: i2c timeout.
> [display]
> lcd_type: ssd1306
> i2c_bus: i2c1a
> i2c_mcu = mcu

I have it working now.

For the wiring:
SDA = PB9
SCL = PB8
VCC = +5V
GND = GND

For the config:

[display]
lcd_type: ssd1306
i2c_bus: i2c1a
i2c_mcu = mcu