Klipper3d/klipper #3582: Method to access PI temperature sensors

al3ph, 2020-11-27:
Is there a way to access the PI temp sensors like any other temperature sensor ?

It's be handly if the GPU/CPU temps were exposed when running the PI as an MCU, that way we could control case fans based off the temps from klipper.

cheers!

al3ph, 2020-11-27:
This is an example piece of code someone wrote, even just the CPU temp would be handy, GPU not so much.

```
#!/bin/bash
# Script: my-pi-temp.sh
# Purpose: Display the ARM CPU and GPU  temperature of Raspberry Pi 2/3 
# Author: Vivek Gite <www.cyberciti.biz> under GPL v2.x+
# -------------------------------------------------------
cpu=$(</sys/class/thermal/thermal_zone0/temp)
echo "$(date) @ $(hostname)"
echo "-------------------------------------------"
echo "GPU => $(/opt/vc/bin/vcgencmd measure_temp| awk -F\= '{print $2}')"
echo "CPU => $((cpu/1000))'C"

```

KevinOConnor, 2020-11-27:
I agree it would be useful, but querying the host temperature sensor is not currently implemented.  An interested developer would need to take that on and submit the code.

-Kevin

al3ph, 2020-11-27:
Heh, hint hint.. I'll have a look at the code base. Any pointers to parts I'll need to look at would be handy.

BladeScraper-Designs, 2020-12-03:
I'm also interested in this as well, being able to control a fan using the SoC temperature.  OctoPrint is able to access the SoC temperature (there are a couple of plugins that will show you the SoC temp on your dashboard), so I'd think Klipper should be able to as well since it's running on the same Pi.  I know it's probably not a very highly requested feature, but it'd be cool to see.

al3ph, 2020-12-03:
It's in progress..  https://github.com/KevinOConnor/klipper/pull/3587

closed 2020-12-03 by al3ph

nicedevil007, 2021-02-06:
> #!/bin/bash
> # Script: my-pi-temp.sh
> # Purpose: Display the ARM CPU and GPU  temperature of Raspberry Pi 2/3
> # Author: Vivek Gite <www.cyberciti.biz> under GPL v2.x+
> # [...]

do you know a possible solution to output this to the motd? so everytime I login to the shell I can see the temps?

richardklingler, 2021-08-28:
Call this script at the end of the /etc/profile script.