Klipper3d/klipper #3584: ADXL345 graph_accelerometer.py not working

xkuyax, 2020-11-28:
Hi,

i just got my ADXL345 working via the skr pro SPI interface, i was able to run the TEST_RESONANCES AXIS=X & TEST_RESONANCES AXIS=Y command which gave me an .csv file (uploaded as .txt)
[klippy.log](https://github.com/KevinOConnor/klipper/files/5609865/klippy.log)
[resonances_x_20201128_010643.txt](https://github.com/KevinOConnor/klipper/files/5609866/resonances_x_20201128_010643.txt)

If i now i try and run `~/klipper/scripts/graph_accelerometer.py /tmp/resonances_y_20201128_011024.csv -o /tmp/resonances_x.png
`
I get the following error:
```pi@vcore:~/klipper $ ~/klipper/scripts/graph_accelerometer.py /tmp/resonances_y_20201128_011024.csv -o /tmp/resonances_x.png
Traceback (most recent call last):
  File "/home/pi/klipper/scripts/graph_accelerometer.py", line 242, in <module>
    main()
  File "/home/pi/klipper/scripts/graph_accelerometer.py", line 208, in main
    datas = [parse_log(fn) for fn in args]
  File "/home/pi/klipper/scripts/graph_accelerometer.py", line 18, in parse_log
    return np.loadtxt(logname, comments='#', delimiter=',')
  File "/usr/lib/python2.7/dist-packages/numpy/lib/npyio.py", line 1134, in loadtxt
    for x in read_data(_loadtxt_chunksize):
  File "/usr/lib/python2.7/dist-packages/numpy/lib/npyio.py", line 1061, in read_data
    items = [conv(val) for (conv, val) in zip(converters, vals)]
  File "/usr/lib/python2.7/dist-packages/numpy/lib/npyio.py", line 768, in floatconv
    return float(x)
ValueError: could not convert string to float: freq
pi@vcore:~/klipper $
```

Thanks for your help and awesome work!

Sineos, 2020-11-28:
Did you follow https://github.com/dmbutyugin/klipper/blob/adxl345-spi/docs/Measuring_Resonances.md#software-installation?

xkuyax, 2020-11-28:
> Did you follow https://github.com/dmbutyugin/klipper/blob/adxl345-spi/docs/Measuring_Resonances.md#software-installation?

Yeah, python-matplotlib is installed and numpy too (i think otherwise there would be an difference error)

python-matplotlib is already the newest version (2.2.3-6).
python-numpy is already the newest version (1:1.16.2-1).

Im using the sensor not via the rpi like it is shown in the tutorial, but via the skr pro spi interface

Sineos, 2020-11-28:
Ah, sorry, now I see what is going wrong. For the stand-alone processing scripts you have to run your tests with `OUTPUT=raw_data`

xkuyax, 2020-11-28:
> Ah, sorry, now I see what is going wrong. For the stand-alone processing scripts you have to run your tests with `OUTPUT=raw_data`

Ah, now that whole thing makes sense.. :D
I guess this can be closed now

closed 2020-11-28 by xkuyax

davide-molino, 2021-04-09:
@xkuyax can you share your printer.cfg ? I want to connect the adxl345 to the SPI port on the SKR Pro but didn't work