Klipper3d/klipper #3585: Move 2 Stepper Motor Independently

scientistnobee, 2020-11-28:
Hi everyone,

I would like to move two stepper motors independent to each other.

Motor #1 set to rotate at X,
Motor #2 step forward 40 steps, then back 100 and repeat,
Motor #3 backward 100 steps, forward 30, backward 50, forward 10, repeat.

I'd also like to not have acceleration if possible, or have minimal otherwise. 

This is not possible with Marlin, so I installed Klipper with RAMPS set-up. I knew this is inherently possible with Arduino by using the Multi-stepper class, but then I would lose all the good features of Klipper. So please let me know, if this independent move is possible with the Kilpper.

scientistnobee, 2020-11-28:
Hi This query doesn't require the log file.
[klippy.log](https://github.com/KevinOConnor/klipper/files/5611068/klippy.log)

KevinOConnor, 2020-12-03:
https://www.klipper3d.org/Config_Reference.html#manual_stepper

scientistnobee, 2020-12-09:
Hi Kevin, if I

scientistnobee, 2020-12-17:
My issue hasn't been solved. I am attaching here the log file. 
[17122020_klippy.log](https://github.com/KevinOConnor/klipper/files/5711533/17122020_klippy.log)

jakep82, 2020-12-17:
A cartesian printer requires a Z axis.  If you're not using Klipper on a printer with cartesian kinematics, then you need to use `kinematics: none`

closed 2021-01-15 by github-actions[bot]