Klipper3d/klipper #3588: [FR] MMU DIP Board stm32f0c8

kakou-fr, 2020-11-29:
This is a request for the MMU  DIP board from BTT.
It is possible to add the  support for the stm32f0c8 MCU into klipper ?

Thks

KevinOConnor, 2020-11-29:
It's not difficult to add support for the stm32f030 chip (it's very similar to the stm32f070 chip which is already supported).  If someone has the board on hand and is willing to run tests then I can probably assist.

That said, I don't see a schematic for that particular btt board.  Knowing where all the pins are routed will be critical to using it successfully with Klipper.

-Kevin

kakou-fr, 2020-11-29:
I have the board and started a port of the 070, only  problem I have is with the USB support.
Other bi problem, this board use an hc595 to manage some pins like the dir pins of steppers.

kakou-fr, 2020-11-30:
this is the error when i try to compile with (or without) usb : 

```
src/stm32/stm32f0.c: In function 'pll_setup':
src/stm32/stm32f0.c:153:18: error: 'RCC_CFGR3_USBSW' undeclared (first use in this function); did you mean 'RCC_CFGR3_USART1SW'?
         cfgr3 |= RCC_CFGR3_USBSW;
                  ^~~~~~~~~~~~~~~
                  RCC_CFGR3_USART1SW
src/stm32/stm32f0.c:153:18: note: each undeclared identifier is reported only once for each function it appears in
```

kakou-fr, 2020-12-01:
ok i  commented this line, and now i can flash klipper on it  (i use the serial connection USART1).

Some recommandation for the implementation of the hc595 ?

KevinOConnor, 2020-12-03:
Ah - this board has a "shift register" also?  That's a shame, and it severely limits the utility of the board.  (It would have been far simpler, and probably cheaper, for the board to use a micro-controller with sufficient pins then to add a goofy second chip to control a handful of additional pins).

In any case, this was discussed a bit at https://github.com/KevinOConnor/klipper/issues/916#issuecomment-440902897 and at https://github.com/KevinOConnor/klipper/pull/1047 .  Those discussions were about the original (convoluted) prusa mmu2 board, but they would apply to any board using shift registers.

-Kevin

kakou-fr, 2020-12-03:
yes i am agree, they just cloned the prusa  MMU board with  an stm32f0 and keep the 74HC595 istead of proposing a new design ... I am trying to include the stuff from  @trevjonez without modifying the gpio.c

kakou-fr, 2020-12-06:
i submited a PR  here #3606

closed 2021-01-04 by github-actions[bot]