Klipper3d/klipper #3589: MCU 'rpi' shutdown: Unable to issue spi ioctl

ffnull, 2020-11-29:
Try to setup resonance messure.
RPI4. Latest branch
SPI enabled and tested accelerometer via script.
But when im isuue in console
ACCELEROMETER_QUERY

Im got
22:45:44 Internal error on command:"ACCELEROMETER_QUERY"
22:45:50 MCU 'rpi' shutdown: Unable to issue spi ioctl
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown
Klipper state: Not ready
22:45:50 MCU 'rpi' shutdown: Unable to issue spi ioctl
[klippy.log](https://github.com/KevinOConnor/klipper/files/5612990/klippy.log)

ffnull, 2020-12-10:
any solution?

KevinOConnor, 2020-12-10:
I don't know why it would do that.  It works fine for me.  Have you customized the OS install in some way?  FWIW, when I've seen "weird OS error" reports in the past, they were often resolved by a fresh OS installation.

-Kevin

ffnull, 2020-12-10:
> I don't know why it would do that. It works fine for me. Have you customized the OS install in some way? FWIW, when I've seen "weird OS error" reports in the past, they were often resolved by a fresh OS installation.
> 
> -Kevin

Im use oficial octoprint image. just update it via apt update & upgrade

KevinOConnor, 2020-12-10:
Well, your log shows it couldn't gather basic OS / version info (`Git version: '63077484'` and `CPU: 4 core ?`) so something very strange is occurring.  It sounds like the problem is at the OS level.

-Kevin

ffnull, 2020-12-10:
> Well, your log shows it couldn't gather basic OS / version info (`Git version: '63077484'` and `CPU: 4 core ?`) so something very strange is occurring. It sounds like the problem is at the OS level.
> 
> -Kevin

Can problem be in that im use RPI 4 and this branch: https://github.com/Desuuuu/klipper ?

KevinOConnor, 2020-12-15:
This github issue tracker is only for use with the code in this github repository.  You'll need to either reproduce the issue with the pristine code from this repo, or report the problem to the owners of the code you are running.

-Kevin

closed 2021-01-13 by github-actions[bot]