Klipper3d/klipper #3591: Board restart after connect to usb

anonizmous, 2020-11-30:
hello.
I have a board: skr 1.3 with tmc2209*4
I am build klipper and flash it. System detect new USB device, in kernel log see:
```
ноя 25 12:24:56 valik-comp kernel: usb 2-13: new full-speed USB device number 98 using xhci_hcd
ноя 25 12:24:57 valik-comp kernel: usb 2-13: New USB device found, idVendor=1d50, idProduct=614e, bcdDevice= 1.00
ноя 25 12:24:57 valik-comp kernel: usb 2-13: New USB device strings: Mfr=1, Product=2, SerialNumber=3
ноя 25 12:24:57 valik-comp kernel: usb 2-13: Product: lpc1768
ноя 25 12:24:57 valik-comp kernel: usb 2-13: Manufacturer: Klipper
ноя 25 12:24:57 valik-comp kernel: usb 2-13: SerialNumber: 21300018C81848AF8FE2185DC72000F5
ноя 25 12:24:57 valik-comp kernel: cdc_acm 2-13:1.0: ttyACM0: USB ACM device
```
Okey. I and I conclude that the firmware at least started.
But as soon as I try to connect to the board via USB at least with a simple terminal program, the board immediately reboots:
```
(octoprint) [r0mik@valik-comp ~]$ miniterm.py /dev/ttyACM0 
could not open port '/dev/ttyACM0': [Errno 5] could not open port /dev/ttyACM0: [Errno 5] Input/output error: '/dev/ttyACM0'
```
the kernel log shows that the board has restarted:
```
ноя 25 12:24:56 valik-comp kernel: usb 2-13: new full-speed USB device number 98 using xhci_hcd
ноя 25 12:24:57 valik-comp kernel: usb 2-13: New USB device found, idVendor=1d50, idProduct=614e, bcdDevice= 1.00
ноя 25 12:24:57 valik-comp kernel: usb 2-13: New USB device strings: Mfr=1, Product=2, SerialNumber=3
ноя 25 12:24:57 valik-comp kernel: usb 2-13: Product: lpc1768
ноя 25 12:24:57 valik-comp kernel: usb 2-13: Manufacturer: Klipper
ноя 25 12:24:57 valik-comp kernel: usb 2-13: SerialNumber: 21300018C81848AF8FE2185DC72000F5
ноя 25 12:24:57 valik-comp kernel: cdc_acm 2-13:1.0: ttyACM0: USB ACM device
===>> ноя 25 12:25:04 valik-comp kernel: usb 2-13: reset full-speed USB device number 98 using xhci_hcd
ноя 25 12:25:04 valik-comp kernel: usb 2-13: USB disconnect, device number 98
ноя 25 12:25:04 valik-comp kernel: usb 2-13: new full-speed USB device number 99 using xhci_hcd
ноя 25 12:25:05 valik-comp kernel: usb 2-13: New USB device found, idVendor=1d50, idProduct=614e, bcdDevice= 1.00
ноя 25 12:25:05 valik-comp kernel: usb 2-13: New USB device strings: Mfr=1, Product=2, SerialNumber=3
ноя 25 12:25:05 valik-comp kernel: usb 2-13: Product: lpc1768
ноя 25 12:25:05 valik-comp kernel: usb 2-13: Manufacturer: Klipper
ноя 25 12:25:05 valik-comp kernel: usb 2-13: SerialNumber: 21300018C81848AF8FE2185DC72000F5
ноя 25 12:25:05 valik-comp kernel: cdc_acm 2-13:1.0: ttyACM0: USB ACM device
```
in the middle you can see the message where it happened (I marked it with an arrow).
because of what can this happen?

anonizmous, 2020-12-01:
now connects, but it looks like something is wrong with the speed.

I've tested all possible speeds from 9600 to 5600000, but it always returns similar junk -
[image]

is nothing interesting in log, just a brute of speeds:
<details><summary>log</summary>

```
2020-12-01 13:40:37,896 - octoprint.startup - INFO - ******************************************************************************
2020-12-01 13:40:37,897 - octoprint.startup - INFO - Starting OctoPrint 1.5.0
2020-12-01 13:40:37,897 - octoprint.startup - INFO - ******************************************************************************
2020-12-01 13:40:37,905 - octoprint.util.connectivity.connectivity_checker - INFO - Connectivity state is currently: online
2020-12-01 13:40:37,905 - octoprint.util.connectivity.connectivity_checker - INFO - Connecting to 8.8.8.8:53 is working
2020-12-01 13:40:37,905 - octoprint.util.connectivity.connectivity_checker - INFO - Resolving octoprint.org is working
2020-12-01 13:40:37,912 - octoprint.plugin.core - INFO - Loading plugins from /home/r0mik/.octoprint/venv/lib/python3.8/site-packages/OctoPrint-1.5.0-py3.8.egg/octoprint/plugins, /home/r0mik/.octoprint/plugins and installed plugin packages...
2020-12-01 13:40:38,113 - octoprint.plugin.core - INFO - Plugin Pi Support Plugin did not pass check, not loading.
2020-12-01 13:40:38,192 - octoprint.plugin.core - INFO - Found 18 plugin(s) providing 18 mixin implementations, 36 hook handlers
2020-12-01 13:40:38,211 - octoprint.server.heartbeat - INFO - Starting server heartbeat, 900.0s interval
2020-12-01 13:40:38,213 - octoprint.server - INFO - Intermediary server started
2020-12-01 13:40:38,217 - octoprint.filemanager.storage - INFO - Initializing the file metadata for /home/r0mik/.octoprint/uploads...
2020-12-01 13:40:38,254 - octoprint.filemanager.storage - INFO - ... file metadata for /home/r0mik/.octoprint/uploads initialized successfully.
2020-12-01 13:40:38,254 - octoprint.server - INFO - Added new permission from plugin action_command_notification: PLUGIN_ACTION_COMMAND_NOTIFICATION_SHOW (needs: "Need(method='role', value='plugin_action_command_notification_show')")
2020-12-01 13:40:38,254 - octoprint.server - INFO - Added new permission from plugin action_command_notification: PLUGIN_ACTION_COMMAND_NOTIFICATION_CLEAR (needs: "Need(method='role', value='plugin_action_command_notification_clear')")
2020-12-01 13:40:38,254 - octoprint.server - INFO - Added new permission from plugin action_command_prompt: PLUGIN_ACTION_COMMAND_PROMPT_INTERACT (needs: "Need(method='role', value='plugin_action_command_prompt_interact')")
2020-12-01 13:40:38,255 - octoprint.server - INFO - Added new permission from plugin announcements: PLUGIN_ANNOUNCEMENTS_READ (needs: "Need(method='role', value='plugin_announcements_read')")
2020-12-01 13:40:38,255 - octoprint.server - INFO - Added new permission from plugin announcements: PLUGIN_ANNOUNCEMENTS_MANAGE (needs: "Need(method='role', value='plugin_announcements_read'), Need(method='role', value='plugin_announcements_manage')")
2020-12-01 13:40:38,255 - octoprint.server - INFO - Added new permission from plugin appkeys: PLUGIN_APPKEYS_ADMIN (needs: "Need(method='role', value='plugin_appkeys_admin')")
2020-12-01 13:40:38,255 - octoprint.server - INFO - Added new permission from plugin backup: PLUGIN_BACKUP_ACCESS (needs: "Need(method='role', value='plugin_backup_access')")
2020-12-01 13:40:38,255 - octoprint.server - INFO - Added new permission from plugin firmware_check: PLUGIN_FIRMWARE_CHECK_DISPLAY (needs: "Need(method='role', value='plugin_firmware_check_display')")
2020-12-01 13:40:38,255 - octoprint.server - INFO - Added new permission from plugin logging: PLUGIN_LOGGING_MANAGE (needs: "Need(method='role', value='plugin_logging_manage')")
2020-12-01 13:40:38,255 - octoprint.server - INFO - Added new permission from plugin pluginmanager: PLUGIN_PLUGINMANAGER_MANAGE (needs: "Need(method='role', value='plugin_pluginmanager_manage')")
2020-12-01 13:40:38,255 - octoprint.server - INFO - Added new permission from plugin pluginmanager: PLUGIN_PLUGINMANAGER_INSTALL (needs: "Need(method='role', value='plugin_pluginmanager_manage'), Need(method='role', value='plugin_pluginmanager_install')")
2020-12-01 13:40:38,255 - octoprint.server - INFO - Added new permission from plugin softwareupdate: PLUGIN_SOFTWAREUPDATE_CHECK (needs: "Need(method='role', value='plugin_softwareupdate_check')")
2020-12-01 13:40:38,255 - octoprint.server - INFO - Added new permission from plugin softwareupdate: PLUGIN_SOFTWAREUPDATE_UPDATE (needs: "Need(method='role', value='plugin_softwareupdate_update')")
2020-12-01 13:40:38,255 - octoprint.server - INFO - Added new permission from plugin softwareupdate: PLUGIN_SOFTWAREUPDATE_CONFIGURE (needs: "Need(method='role', value='plugin_softwareupdate_configure')")
2020-12-01 13:40:38,454 - octoprint.util.pip - INFO - Using "/home/r0mik/.octoprint/venv/bin/python -m pip" as command to invoke pip
2020-12-01 13:40:38,638 - octoprint.util.pip - INFO - Version of pip is 20.2.1
2020-12-01 13:40:38,639 - octoprint.util.pip - INFO - pip installs to /home/r0mik/.octoprint/venv/lib/python3.8/site-packages (writable -> yes), --user flag needed -> no, virtual env -> yes
2020-12-01 13:40:38,640 - octoprint.util.pip - INFO - ==> pip ok -> yes
2020-12-01 13:40:38,670 - octoprint.plugins.softwareupdate - INFO - Loaded version cache from disk
[... cut ...]
  File "/home/r0mik/.octoprint/venv/lib/python3.8/site-packages/pyserial-3.5-py3.8.egg/serial/serialposix.py", line 401, in _reconfigure_port
    raise SerialException("Could not configure port: {}".format(msg))
serial.serialutil.SerialException: Could not configure port: (5, 'Input/output error')
2020-12-01 13:49:13,762 - octoprint.util.comm - INFO - Serial detection: Could not open port /dev/ttyS1, baudrate 2500000, skipping
2020-12-01 13:49:13,762 - octoprint.util.comm - INFO - Serial detection: Trying port /dev/ttyS0, baudrate 2500000
2020-12-01 13:49:13,762 - octopri

mental405, 2020-12-02:
Klipper uses two serial ports. The serial port to the MCU (board) is utilized by the klipper service on the host os to communicate. Any user level communication to the firmware is done via the virtual serial port located at /tmp/printer
If you are trying to connect directly to /dev/ttyACM0 you are going to be getting encoded data from the MCU so it will look like garbage.

anonizmous, 2020-12-02:
/tmp/printer  is a symlink to /dev/serial/by-id/usb-Klipper_lpc1768_21300018C81848AF8FE2185DC72000F5-if00
which in turn is a symlink to /dev/ttyACM0
it is created when the service starts (systemctl start klipper)
I start the clipper manually, so I specify the direct port. which is the same thing.

> If you are trying to connect directly to /dev/ttyACM0 you are going to be getting encoded data from the MCU so it will look like garbage.

klipper just can't connect. I gave the log above (by the way, it uses /dev/serial/by-id/usb-Klipper_lpc1768_xxxxx)

jakep82, 2020-12-02:
Please read and follow the install documents.  Octoprint must be connected to /tmp/printer.

https://www.klipper3d.org/Installation.html#configuring-octoprint-to-use-klipper

jakep82, 2020-12-02:
Also please note that /tmp/printer is not a symlink to the serial port of the MCU.  It's a symlink to a virtual serial port created by the host software at /dev/pts/X.

anonizmous, 2020-12-03:
> Also please note that /tmp/printer is not a symlink to the serial port of the MCU. It's a symlink to a virtual serial port
oh lol,
I'm dumbas!
problem solved, thanks!

closed 2020-12-03 by anonizmous