Klipper3d/klipper #3592: [Klipper] [Documentation Request] Acceleration Calibration

Brokengeiger, 2020-11-30:
As a new user of Klipper I found many of the documentations and external documentations fairly well documented. I managed to my old RigidBot 1 upgrade (now essentially a reliabuild 3) onto klipper in 1 night with another day of calibration. However one aspect seems amiss - I'm stuck printing at 30mm/s.  While I've run into some decent documentations out there for various systems, mostly marlin. It seems this topic has a vast enough difference on Klipper to warrant some critical choice words for acceleration and speed - especially with pressure advance properly configured and still feels like my prints are high quality at 30,  A-OK at 40 but full on failure at 60.  

Would love to be cool like my friends. 

Klipper seems to have a more down to earth explanation and approach that appeals to my frame of thought that may help as a walkthrough of step by step. 

Forgive me if this is intermixed with another topic that I missed - perhaps it can stand alone for itself?

jakep82, 2020-11-30:
It seems like you might be confusing velocity and acceleration.  I typically print at 60-80mm/s velocity with 200mm/s non-print moves, and my acceleration is 4000mm/s^2.  All I did to tune acceleration was turn it up until I started missing steps at 5000, then I dropped it to 4000 to be safe.  None of the tuning for speed or acceleration is specific to Klipper unless you're using input shaping (which is well documented), and it's mostly subjective with regard to acceptable print quality.

grigi, 2020-12-01:
Max printable speed is often limited by how much plastic your hot-end can melt.
The "Calculating Maximum Feedrate" section here: https://teachingtechyt.github.io/calibration.html#accel explains what you need to do nicely.

Then you either configure it in your slicer, e.g. slic3r/prusaslicer/superslicer has it as a filament property (This is correct as some plastics need more energy to melt), or you basically manage your maximum feedrate yourself.

If there is a lot of headroom left, I would recommend to tune input shaping, as it will get rid of a few artifacts, and allow you to get better prints at higher velocities (even with a wobbly printer).

Brokengeiger, 2020-12-01:
This article you just provided me, looks like this is nearly exactly what I might need. There is a few topics in there about jerk too, but was hoping for associations to how this all works with pressure advance to help tie it all together. 

But for now I think this page will do.

grigi, 2020-12-01:
Yes, it was a very good resource for me when I was on Marlin. for Klipper you'd probably want to:
* Basic sanity checks
* Do PID autotune from the klipper docs: https://www.klipper3d.org/Config_checks.html#calibrate-pid-settings
* Extruder flow calibration is basically 1/<marlin-number> (otherwise the same)
* Slicer flow (from that guide)
* Then I would do: https://www.klipper3d.org/Pressure_Advance.html (doesn't matter that things are not perfect, you want to push the melt flow to breaking point in any case)
* Determine the maximum feedrate
* Then https://www.klipper3d.org/Resonance_Compensation.html (which will also get acceleration about right)
* Jerk is the `square_corner_velocity` in the `[printer]` section. I had to drop mine from 5 to 3, see if dropping it helps your printer. Don't increase it unless you have something fantastic.
* Then do the temperature tower (from that guide)
* Lastly tune in retraction settings (from that guide)

The resonance compensation is something unique to Klipper and it's truly amazing what it can do.

Unfortunately for me, my Printer's frame is just not made of high-enough quality parts, but I can now get very good prints at 80mm/s. The limiting factor for me is the melt flow rate of about 8.5mm²/s. So I can only do 60mm/s at 0.28mm layer height or start suffering under extrusion.

Brokengeiger, 2020-12-01:
Great list! Thank you kindly. I know I've done only some of these, but I'll parse over them all again. I've love to move safely from 30 to 80!

Sineos, 2020-12-01:
Maybe to add an experience I have made when moving from Marlin to Klipper:
80 mm/s in Marlin is a whole different story than 80 mm/s in Klipper. 
I have no clue what the real difference is, but on my modified Ender 5, Marlin 120 mm/s is working nicely and the machine is printing in peaceful harmony with itself. 
Klipper's 120 mm/s is tearing the machine apart. To be honest, I trust Klipper's speed statements much more than I do for Marlin's.

Brokengeiger, 2020-12-01:
Yeah it was night and day for me (marlin to klipper) - that's where I was thinking Klipper might need it's own specific set of instruction. But tonight when I get to that list above - I might know better.

dmbutyugin, 2020-12-02:
@Sineos you could probably compare the print times between Marlin and Klipper in those scenarios and judge from there.

Brokengeiger, 2020-12-02:
Last night I only got one test out of the way. I didn't have square_corner_velocity set in my config so it was taking defaults. 
When I went to check the reference wiki I noticed max_accel_to_decel has a description that sounded like it would tear itself apart if I left it too high. My max_accel is 3000, so than my max_accel_to_decel would be 1500. I reduced max_accel_to_decel to 1000 and square_corner_velocity to 3.0. My first print was a known failing print at 60 had succeeded. 

I still have a few tweaks to do as the very next print, slightly more complex, no brim was a total mess at 60, prolly would have been fine at 60 with a brim though. 

#max_accel_to_decel:
#   A pseudo acceleration (in mm/s^2) controlling how fast the
#   toolhead may go from acceleration to deceleration. It is used to
#   reduce the top speed of short zig-zag moves (and thus reduce
#   printer vibration from these moves). The default is half of
#   max_accel.
#square_corner_velocity: 5.0

Sineos, 2020-12-03:
@dmbutyugin I did. Klipper is always faster using the same setting for accel, speed etc.

@Brokengeiger 
From my experience, owning a modified Anet A8 and a heavily modified Ender 5:

- With a typical 0.4 nozzle and average extruder / hot-end and a layer height of 0.2 and with the common layer width of 0.48 speeds between 60 and 80 mm/s should be easily feasible
- Over 80 mm/s will depend on on the above mentioned parameters, mainly due to a physical limitation in the flow of the molten plastics

With the issues you are describing, I would first and foremost check the mechanical aspects of the printer:

- mechanical accuracy of the frame: 90° corners, torqued beams etc.
- bed stability and accuracy
- frame stiffness: if in question add corner stiffening measures
- tighten screws
- check linear movement components: play, friction, smoothness, greasing, etc.
- belt tension

Of course you can tune down accel and speed (btw. accel is more important in cutting down print times than speed) but it will only mask the symptoms but not cure the root cause.

Also as @grigi mentioned, the resonance compensation is amazing. Again, in my humble opinion, this is top end tuning and makes no sense if the foundation is lacking.

closed 2020-12-07 by klipper-gitissuebot

grigi, 2020-12-07:
No, I don't think this ticket should be closed. It's a standing request (and discussion) for what would be required for a comprehensive calibration manual for Klipper.
I'll even volunteer doing it.

KevinOConnor, 2020-12-07:
It's fine (and appreciated) if you wish to add documentation to Klipper.  A PR is probably the best way to discuss that (for example, see #3158), but a github issue ticket is also fine as long as someone is actively working on it.

This particular ticket made reference to an event with the software ("I'm stuck printing at 30mm/s") but did not include the log.  From my perspective, this ticket is invalid.  If you would like to contribute I suggest opening a separate ticket/PR.

Thanks,
-Kevin