Klipper3d/klipper #3593: Get hd44780 (Anet "Zonestar" display) to work on BTT SKR MINI E3 2.0

grigi, 2020-11-30:
Hi, I have been trying to get an hd44780 display working on an BTT SKR Mini E3 2.0 board.

I have done the rewiring as per the docs https://github.com/MarlinFirmware/Marlin/blob/e8177735a00d1d7bb20b439544d3324fe01cfe7e/Marlin/src/pins/sanguino/pins_ANET_10.h#L224-L233
mapping that to the SKR's pinout for it's EXP1 header.
I **did not connect the ADC_KEYPAD_PIN**, as it should not be needed to get the display working (and I have not yet done the 3.3V mod)

The config I went with after quarduple-checking everything is:
``` ini
[display]
lcd_type: hd44780
rs_pin: PB9
e_pin: PA15
d4_pin: PB8
d5_pin: PA9
d6_pin: PA10
d7_pin: PB5
```

The display is powering up, but not initializing. It behaves as if it ONLY got power.

There is precedent for getting this display working on SKR Mini boards in Marlin, so I think they should be electrically compatible.

Thinking it may be a timing issue, I dooubled/halved this: https://github.com/KevinOConnor/klipper/blob/fcb78e50e50dae67d4f335464704e98bff26b8d9/klippy/extras/display/hd44780.py#L15 but to no effect.

Any help would be appreciated.

[klippy.log](https://github.com/KevinOConnor/klipper/files/5617623/klippy.log)

grigi, 2020-12-10:
I gave up on trying to get this working, closing.

closed 2020-12-10 by grigi

postfach, 2021-02-16:
I managed to get this working without too much trouble, I did have to rewire the mainboard end of the ribbon cable.

The display section of my printer.cfg is as follows, I have left the pin aliases for reference.

```
[display]
lcd_type: hd44780
rs_pin: PB9
e_pin: PB15
d4_pin: PA10
d5_pin: PA9
d6_pin: PB8
d7_pin: PA15

[board_pins]
aliases:
    # EXP1 header
    EXP1_1=PB5, EXP1_3=PA9,   EXP1_5=PA10, EXP1_7=PB8, EXP1_9=<GND>,
    EXP1_2=PA15, EXP1_4=<RST>, EXP1_6=PB9,  EXP1_8=PB15, EXP1_10=<5V>

```
I used the pinout from JP16 [here](https://reprap.org/mediawiki/images/7/7d/Melzi-circuit.png) to find the original wiring and moved the wires to the appropriate pins based on [this diagram](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/master/hardware/BTT%20SKR%20MINI%20E3%20V2.0/Hardware/BTT%20SKR%20MINI%20E3%20V2.0-PIN.pdf)

I didn't need to make any other changes.

grigi, 2021-02-16:
I'm pretty sure I did a very good effort on this, but it didn't want to work for me. Note that since I did this there have been several fixes for HD44780 display initialization, which may have resolved it.

I'm glad you got it working :-)

jaanbrotherns, 2021-04-29:
@postfach this is pretty awesome. I'm looking to do the same but don't have your wizardry in pin alignment for the wires. I see you included the JP16 diagram and I have the SKR 2.0 pin out, but I'm not getting which goes to which on the wiring schema. Can you share this? A photo or even a:

PB9 = 2 EXT-A1

..and so forth would be incredible!

postfach, 2021-05-03:
> @postfach this is pretty awesome. I'm looking to do the same but don't have your wizardry in pin alignment for the wires. I see you included the JP16 diagram and I have the SKR 2.0 pin out, but I'm [...]

Sorry for the slow response, I will check tonight exactly what I did and get back to you, I should still have my notes at home.

postfach, 2021-05-04:
@jaanbrotherns 

I changed the physical order of the wires at the board end of the LCD ribbon cable as follows:

```
JP16		EXP1		REWIRE 

SDA		PB5		-		
A1		PA15		SDA		D7
SCL		TX		TX		D5
A2		RST		-		
TX		RX		RX		D4
A3		PB9		A3		RS_PIN
RX		PB8		SCL		D6
A4		PB15		A2		E_PIN
5V		GND		GND		
GND		5V		5V		
```

JP16 is the pins labeled and ordered as they are originally, EXP1 is the pinout for the SKR, REWIRE is the pins with original names in new (working) order.  The final column is how I assigned them in my printer.cfg

Below is a photo of how my cable looks after rewiring.

[image]

jaanbrotherns, 2021-05-05:
Thank you! Couldn't have done it without your help, thanks so much! I've got the LCD up and running, next step: making the buttons actually work.
[image]