Klipper3d/klipper #3594: SKR1.4 & FYSETC TMC2209 v3

robvoi, 2020-11-30:
Hi.
It would be great if someone could share a working configuration for SKR1.4 & FYSETC TMC2209 v3 with sensorless homing.
I have UART running. (just trying x for now)
But the sensorless homing doesn't work. Depending on how I configure the endstop pin (inverted or not) it either always triggers (in stand still) or never. I also tried shorting the diag pin an the endstop trigger pin. Didn't help.

Not worth a lot as I tested all kind of settings - but here my latest config. (which doesn't work/load in this state)

[stepper_x]
step_pin: P2.2
dir_pin: P2.6
enable_pin: !P2.1
step_distance: .0125
#endstop_pin: P1.29
endstop_pin: tmc2209_stepper_x:virtual_endstop
position_endstop: 0
position_max: 235
homing_speed: 50
homing_retract_dist: 0

[tmc2209 stepper_x]
uart_pin: P1.10
microsteps: 16
interpolate: True
run_current: 0.800
hold_current: 0.500
stealthchop_threshold: 250

diag_pin: PXDIAG1
#diag_pin: 1.29
driver_SGTHRS: 250

Any help would be highly appreciated.
Thanks

jakep82, 2020-11-30:
You need to post a log, but I'm not sure why you have the pin as PXDIAG1.  It looks like with that board it would be P1.29 (your commented diag pin line is missing the P).  You've also set the driver_SGTHRS value near the most sensitive.  It might trigger immediately when it tries to start moving.  If none of that helps, you might need to use a multimeter to verify that the diag pin from the driver is connected to the endstop pin.

robvoi, 2020-12-01:
I don't know what was up that day. I tried all combinations of settings.
Using PXDIAG1 was pure frustration.
driver_SGTHRS most sensitive was to make sure it triggers. Just for debugging.

Now everything works. (Once I put in the missing P again)

Thanks for your support.
Robert

For reference - working config with SKR1.4 & FYSETC TMC2209 v3

[stepper_x]
step_pin: P2.2
dir_pin: P2.6
enable_pin: !P2.1
step_distance: .0125
endstop_pin: tmc2209_stepper_x:virtual_endstop
position_endstop: 0
position_max: 235
homing_speed: 50
homing_retract_dist: 0

[tmc2209 stepper_x]
uart_pin: P1.10
microsteps: 16
interpolate: True
run_current: 0.800
hold_current: 0.500
stealthchop_threshold: 250
diag_pin: P1.29
driver_SGTHRS: 250

closed 2020-12-01 by robvoi