Klipper3d/klipper #3595: Ender 5 Plus Touch screen on printer

Fishpig1, 2020-12-01:
So i have got everything setup and working beautifully even down to getting the bed visualiser working perfectly.
My problem i found is the touch screen on the printer itself was working with the exception of the left and middle buttons, the settings button worked until i turned off the printer and when i powered it back on the screen stays at the creality boot logo but everything still prints and works.
is this just a side effect from running Klipper?

Cheers Fishpig

Fishpig1, 2020-12-01:
Don't need to supply a log, just asking a question.

mental405, 2020-12-02:
Many of those touch screens use a serial connection to the board and are essentially gcode terminals. There is presently no support in klipper to open a serial terminal back to the host service via a serial connection on the board. There have been numerous requests for it but so far nobody has taken on the task,

closed 2020-12-08 by klipper-gitissuebot

Fishpig1, 2020-12-25:
I didn't want to post a log file as this was a question and not a problem to be solved via logs.