Klipper3d/klipper #3596: Feature Request: step_distance Feed Rate Lookup Table

DroneMang, 2020-12-01:
Hi,

I have been doing some calibration of extruder step_distance along the lines of Triffid Hunter's Calibration Guide, with a twist.
https://reprap.org/wiki/Triffid_Hunter's_Calibration_Guide#Measure

I have been doing them at or near the max flow rate (using guide below) of my nozzle and setup, **with the nozzle on, and actually extruding**, resulting in about 230-240 max feed rate before grinding or skipping steps on a 0.4mm nozzle and about 100mm/sec max print speed.

https://teachingtechyt.github.io/calibration.html#accel

The benefit of doing the extruder step_distance calibration with the nozzle on and actually extruding is it allows you to have the flow rate set perfectly to 1.0 in the slicer, but when printing smaller features the feed rate is less.  

If I do Triffids calibration at say a lower feed rate of 150, too much filament is being pushed out, and his equation would say the step distance should be adjusted to be less to compensate.

Basically, if you are extruding at a higher feed rate, you need more steps to push a given amount of filament through because of friction etc, which in Triffids test is 100mm.  If I set the feed rate to 150 and ask for 100mm to be extruded (holding temp the same), more than 100mm will be extruded, on the order of 5mm too much.

This is evident on a print where longer extrudes of solid bottom layers may print perfectly, but the short ones are clearly having over extrusion.

I propose a lookup table and set of equations where you do a calibration to see what the step distance should be at your max feed rate or near it, say 240, and the calibration is done with the nozzle on and actually extruding, then you do the same calibration at a lower feed rate say 50, and you find what the step_distance should be there.  Then you can do more calibrations at different feed rates in between if you want, but I think you just need 2 points.  Then you linearly interpolate what the step_distance should be depending on the feed rate called for in the Gcode, and Klipper would dynamically adjust the step_distance on the fly while printing.  This would result in much better infill and less over extrusion.  This could then be done at different temperatures, as the performance is quite different at lower temps.  This would help switching from 0.2 to 0.1 layer heights, since holding the speed constant at 100mm/s but printing at 0.1 results in a feed rate of 120 instead of 240.

This can then be fine tuned using a test print where you print a single layer in a large triangle shape, lets say opening to the right <, and the extrudes go up and down, this would force lower feed rates on the left, and higher feed rates on the right as you get to your max print speed and max_accel_to_decel stops having an effect and pressure gets to fully build in the nozzle.  Then you can adjust your low or high feed rate step distance in the cfg so you get perfect infill across the triangle.

Here is the equation for reference from Triffids Calibration:

new_e_steps = old_e_steps * (100 / distance_actually_moved) … or, old_e_steps * (100 / (120 - distance_to_mark))), so you mark the filament at 120mm, call for 100mm to be extruded, then measure how far off you are from 20mm.

In the calibration there is inaccuracy doing this, because you have to extrude a little, say 40mm, mark the filament at 120mm, then extrude 100mm, and in this time the pressure bleeds off slightly in the nozzle, but this would be taken care of by the calibration print.

Thanks!
-DroneMang

This log is unrelated, just to keep the bot happy!
[klippy7.zip](https://github.com/KevinOConnor/klipper/files/5624461/klippy7.zip)

grigi, 2020-12-06:
Could you try to graph the behavior you are measuring? That way we can see if it's a gradual climb or a sudden one.
e.g. a sample point at a feedrate of 50 (to see how it changes from the 150 value), and some feedrates between 150 and 230. 

If its sudden, I'm thinking the slippage would be a sign of imminent failure,and maybe the solution would be to just limit your max feedrate to just below you notice any slippage?

If it's gradual or closer to linear, then there would be serious merit to being able to calibrate like this.

1N4148, 2020-12-06:
A dynamic flow compensation? Would be nice. But from my experience this would only make sense in the limiting regions of your hotend.

grigi, 2020-12-07:
OK, I can confirm. On my anet hotend, for a flow width of 0.45 I get:
40mm/s: 0.45-0.46
60mm/s: 0.45-0.46
80mm/s: 0.45-0.46
100mm/s: 0.41-0.43
110mms/: 0.35-0.37

So, for me there is no need to the flow rate until probably 90mm/s, but above that it seems to taper off.
I think this is basically where the hotend can't melt the plastic thoroughly. The plastic is also slightly more matt.

It's like at the measured flow rate according to the calibration instructions ends up at ~120mm/s, but in reality its more like at 90mm/s things are great, above that we experience increasing slippage until failure. 

Possibly we should state in our calibration docs about this phenomenon?

DroneMang, 2020-12-07:
What feed rate did you command when you did these?  Are you doing bowden or
direct drive? I wonder if the bowden tube length has an effect?

Just wanted to add that at 0.1 layer height the flow is much less.

If you print at 0.1 layer height and 100 mm/s your feed rate is only F120
for 0.48 extrusion width, vs F240 at 0.2 layer height, so

G91
G1 E100 F120

I'll have to do a table as well to see what I get.

grigi, 2020-12-07:
It was 0.2mm layer height with a 0.45 extrusion width specified in SuperSlicer. The feedrates was listed on the left and the measured extrusion width range was on the right.

I have a 2016 Anet A8 hotend, which is a 30W 12v mk8 hot-end, with a ptfe liner, and a single-gear direct-drive extruder. It's very low-spec, as you can see.

Hmm, so flow is less for you with 0.1mm layer height? Some form of backpressure as the plastic resists getting squashed? Viscosity resistance? That's super interesting.

grigi, 2020-12-07:
On my draft setting of 0.28mm layer and 0.56mm extrusion width, I get the right extrusion at 60mm/s.
Which is way higher flow than for the 0.2mm @90mm/s.

So that single datapoint strongly supports what you said, with more constricted nozzles the effective max flow rate drops, and it increases with less constriction.

grigi, 2020-12-07:
(I forgot to state, 0.4mm nozzle for all)

DroneMang, 2020-12-08:
There does seem to be a linear relationship.  I purposely used very smooth filament to minimize the effect, from full feed rate down to a crawl at F30, there is about a 2.5% difference.

But I have some other filament that is a bit rough, and the difference here from F240 to F120 is 4.75%.  So it does seem to be filament dependent to a degree as to how severe this effect is.  Still a few percent in terms of solid layers and infill does make a big difference in long stretches vs short zigzags.

[Extruder step_distance Feed Rate.xlsx](https://github.com/KevinOConnor/klipper/files/5660834/Extruder.step_distance.Feed.Rate.xlsx)

grigi, 2020-12-08:
That **is** interesting. What constitutes acceptable amount of slippage? :thinking: 

I'm trying to think of how to add this info to my calibration docs, but I feel I don't have a good understanding of it yet.

So far we think we know:
* It's roughly linearly related to flow rate
* It's related to filament type
* It's related to constriction
* It' super-linear when you exceed your actual reliable flow rate.

That's a lot of variables. The last one can be excluded as it's a fail condition.

Klipper is very close to the metal, so it could know the flow rate, but not the filament or constriction.
The slicer could estimate the constriction. And it could know the filament properties if you tell it explicitly.

:thinking: 

At worst we could say that the flow rate should be fine-tuned at half the maximum flow rate, at the typical layer height/width for that nozzle. That way you would be able to minimize the variation.
Considering that "good" filaments radius varies by ~1%, we can't really tune it in much better than that.

How far under ideal extrusion do you start getting layer separation issues? How far above ideal extrusion do you start getting significant tolerance issues?

I suspect many prints are more sensitive to under extrusion than over extrusion, so it might make sense to calibrate at higher than the 50'th percentile speed, but it depends on whether you prioritize detail or strength?

closed 2021-01-06 by github-actions[bot]