Klipper3d/klipper #3599: MCU 'mcu' shutdown: Timer too close / Lost communication with MCU 'mcu'

fsironman, 2020-12-02:
Hello I'm having problems with klipper losing communication with the board (SKR Mini E3 V2.0) due to it disconnecting from the Raspberry Pi.

I changed usb cables (both shielded)
Switched power supply
switched from a pi 3b to a pi zero
taped off VCC on the usb cable

every 1-10 hours im getting the error of lost communication with MCU

in dmesg log I can see that the board disconnects and immediately reconnects:

```
[98822.391791]` usb 1-1: USB disconnect, device number 13
[98822.601658] Indeed it is in host mode hprt0 = 00021501
[98822.811539] usb 1-1: new full-speed USB device number 14 using dwc_otg
[98822.811868] Indeed it is in host mode hprt0 = 00021501
[98823.053684] usb 1-1: New USB device found, idVendor=1d50, idProduct=614e, bcdDevice= 1.00
[98823.053709] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[98823.053722] usb 1-1: Product: stm32f103xe
[98823.053733] usb 1-1: Manufacturer: Klipper
[98823.053743] usb 1-1: SerialNumber: 35FFDC054246313014611857
[98823.072704] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
[184474.088352] usb 1-1: USB disconnect, device number 14
[184474.298263] Indeed it is in host mode hprt0 = 00021501
[184474.508070] usb 1-1: new full-speed USB device number 15 using dwc_otg
[184474.508310] Indeed it is in host mode hprt0 = 00021501
[184474.778305] usb 1-1: New USB device found, idVendor=1d50, idProduct=614e, bcdDevice= 1.00
[184474.778324] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[184474.778333] usb 1-1: Product: stm32f103xe
[184474.778341] usb 1-1: Manufacturer: Klipper
[184474.778348] usb 1-1: SerialNumber: 35FFDC054246313014611857
[184474.799180] cdc_acm 1-1:1.0: ttyACM1: USB ACM device
```
I was reading through some posts and Issues refering to EMI as a possible cause.
Short of completly rewiring everything and swapping out the board I came here to ask if there might be another issue with the board or Klipper.
This Issue only started to manifest after upgrading from version 8 to 9 which could be a coincidence because I wasnt printing that much and long before as I tried now.

After I had the error of Lost communication with MCU I issued a FIRMWARE_RESTART command.
Expecting it to reset the connection.
However it then gave me the MCU 'mcu' shutdown: Timer too close error a few times in a row until it finally worked.
Is this an Indicator of another underlying problem related to Issue 3331?

[klippy.log](https://github.com/KevinOConnor/klipper/files/5632205/klippy.log)

fsironman, 2020-12-04:
I had another occurence of Timer too close error.
Im seeing no apparent Issue in the log.
Am I missing something?

[klippy.log](https://github.com/KevinOConnor/klipper/files/5645379/klippy.log)
file

KevinOConnor, 2020-12-04:
Your second log indicates some kind of system event occurred and the Klipper host software did not get any cpu time for 900ms.  Unfortunately, it's not clear what the cause of that system event is.  If you're familiar with Linux you could try checking the system logs to see if they indicate the source of the problem (eg, voltage issues, heat issues, sd card errors, etc.).

-Kevin

closed 2020-12-09 by fsironman