Klipper3d/klipper #3600: Stepper position error

webers89, 2020-12-03:
Hello.
I have found strange problem with Klipper firmware on my printer.
I can see that if I move X,Y axis in long distances then traveled distance is correct but if I try to print file with small infill where axis has to change direction, it travels incorrect distance. In noticed this problem on Y axis, because I have two stepper motors. One side of Y axis mistake is increasing in plus direction ,second side in minus direction.  And then when I am homing I can clearly see that Y axis difference between both sides are more than 20 mm (after test print of couple of minutes).
I tested similar G code on X axis and I observed the same problem.  When X axis is is traveling back and forth I can clearly see that it doesn't stop in the same position. After each move X position moves in plus direction. 

I am using re-arm board with Geckodrive  external drivers. 

[klippy.log](https://github.com/KevinOConnor/klipper/files/5635015/klippy.log)

webers89, 2020-12-03:
I think I have found problem- 
Gecko stepper driver minimal pulse values are:
STEP pulse LOW 0.5 micro-second, step pulse HIGH 3 ms micro-second.
Is there way to increase step pulse time in Klipper firmware?

KevinOConnor, 2020-12-03:
It can be set when building the micro-controller code.  It's in "make menuconfig", enable "low-level options", and you should find a setting to set the step pulse duration.  Klipper uses the same value for both high and low pulse duration, so just set it to 3us (or larger).

-Kevin

webers89, 2020-12-04:
I increased step pulse time, but it didn't solve the problem.
I found in some manuals for G210X driver that there is parameter 20 micro-second min hold time after Step edge. Probably this is parameter which makes the problem, because it is mostly visible when there are many small direction changes.

I tried to make video of X axis. G code has the same MAX X coordinate in each line, but you can see, that in the beginning X axis stops till the duct tape, but in the end of video it already has passed it more than 10 mm.

[https://www.youtube.com/watch?v=C4NFFiLNqIU&feature=youtu.be](url)

webers89, 2020-12-07:
Is there option to increase time between direction change step pulses?

webers89, 2020-12-11:
> It can be set when building the micro-controller code. It's in "make menuconfig", enable "low-level options", and you should find a setting to set the step pulse duration. Klipper uses the same value for both high and low pulse duration, so just set it to 3us (or larger).
> 
> -Kevin

I tried to increase STEP pulse duration to 20 micro-second, but it didn't help.
Is there way to increase pulse duration between direction changes?

closed 2021-01-09 by github-actions[bot]