Klipper3d/klipper #3602: stm32f103C8 with HID bootloader. I can not let it run

xiaoqianzi15, 2020-12-03:
Hi, Right now I'm using two Skr mini e3 V2 to drive my Voron V1.8 
And I'll start to building my Voron V2.4 so I'm trying to build my own board with stm32f103
One for step motor driver and endstops only and another one for heat bed, fans etc.
But I have questions about how to let my Blue pill board work.
I did the following steps.
1:---make menuconfig
   enable extra low-level config options
   stm32
   stm32f103
   2KiB bootloader(HID Bootloader)
   clock 8Mhz
   Use USB for communication
   USB ids
   PC13  GPIO pins to set at micro-controller startup
2:---make
3:copy "klipper.bin" to my computer and flash it by "STM32CubeProgrammer"  ---Download the file to my Blue Pill
-----------------------------------------------------------------------------------------------------------------
But my Raspberry pi can not find the board (I use "ls /dev/serial/by-id/*")
And my computer(windows10 desktop) warning
"The last USB device you connected to this computer malfunctioned,and windows does not recognize it"

Please help, What did I do wrong?
I almost tried every boot load I can find in "Bootloader offset"

Arksine, 2020-12-05:
If you are flashing klipper.bin directly to the board with STM32CubeProgrammer then you aren't using a Bootloader.  If you want to flash this way then you need to select "no  bootloader" for the `Bootloader offset` option.

xiaoqianzi15, 2020-12-06:
> If you are flashing klipper.bin directly to the board with STM32CubeProgrammer then you aren't using a Bootloader. If you want to flash this way then you need to select "no bootloader" for the `Bootloader offset` option.

Thanks, it works.

closed 2020-12-28 by xiaoqianzi15