Klipper3d/klipper #3605: Problem with "BL touch failed to raise probe" error - redux

WizardOfYendor1, 2020-12-06:
I having the exact same problem again, randomly, as documented in issue #2848.  What's changed since the last time is I decided I was getting too far behind on updates and I updated the firmware and software to the latest and greatest version(s) - then this issue popped up again. 

I haven't done too many prints since the last time this issue happened but everything was smooth sailing until now until the latest updates.

I am able to manually issue the commands BLTOUCH_DEBUG COMMAND=pin_down and pin_up via the terminal without issue. The probe always responds appropriately. 

This is happening when I try to level the bed... which I use a macro to do. This error occurs on the 1st step which is a gcode, G28 command. In fact, if I merely issue the G28 command from terminal, it fails with the exact same error. 

This is messages that get returned back when the G28 command is run.

00:02:04 Failed to verify BLTouch probe is raised; retrying.
Failed to verify BLTouch probe is raised; retrying.
00:02:04 BLTouch failed to raise probe
00:02:04 BLTouch failed to raise probe

Any suggestions for troubleshooting?

WizardOfYendor1, 2020-12-06:
Added log file....

I read thru and followed some stuff here: https://www.klipper3d.org/BLTouch.html. 

In the initial tests section, it goes thru testing the probe. In my case, query probe *always* reports "triggered" regardless of the pin position. When the probe pin is down, there's no light on. If I push it up slightly it turns to red... then turns offs when I stop pushing.  When pin is up, the probe is red. 

For the most part, I've printed a lot of stuff on my printer had never had an issue homing the Z axis. I've had this issue popup in the past and randomly go away... then everything was fine for months until this last software upgrade.  Never had an issue like this with the original Marlin firmware. 

[klippy.log](https://github.com/KevinOConnor/klipper/files/5648119/klippy.log)

BlackStump, 2020-12-06:
This section is important to verify that you have a **functional** BLTOUCH
>Run BLTOUCH_DEBUG COMMAND=pin_down, verify that the pin moves down, run BLTOUCH_DEBUG COMMAND=touch_mode, run QUERY_PROBE, and verify that command reports “probe: open”. Then while gently pushing the pin up slightly with the nail of your finger run QUERY_PROBE again. Verify the command reports “probe: TRIGGERED”.

If your BLTOUCH always reports triggered then you either have a wiring problem, a configuration problem or a faulty BLTOUCH.

WizardOfYendor1, 2020-12-06:
Following those steps it says triggered... instead of open.  

One strange thing that happened last night is I left the printer on and walked away for a few hours. When I came back, query probe started reporting open. So I thought my problem was magically solved... tried to home, and failed with the same error. 

So today I went thru the trouble of removing the C7 capacitor on the zstop pin (Ender3). No change. It always reports triggered. 

Like I said, it's been working fine for the past few months. Did a print a few weeks ago.  I suppose the probe could've gone bad at the exact same time that I did a software/firmware update - but that would be an amazing co-incidence. 

I'll see if I can hook my oscilloscope up to the sense wire and see what it puts out.

WizardOfYendor1, 2020-12-06:
So I disconnected the sense connector that goes into the zstop header on the Ender MCU (1.1.4). I left the control wiring connected (so I could issue pinup and down commands).  

I put my oscilloscope on the sensor wire connector (probe on white, ground on black). With the pin down... there is no signal (well, other than some very low voltage noise). When I touch the end of the pin the red light triggers and I get an immediate signal triggered on my scope (square wave).  I believe this tells me my bltouch is working correctly. So to be clear, the MCU is not connected to the white/black wires in any way.... just direct to my scope. 

One thing I find odd is, with the sensor wire completely disconnected (floating), query_probe still reports "triggered" with the pin up or the pin down. I tried resetting the probe, using "touch_mode" etc.. no change.

Do "triggered" mean a logic high on the sensor pin? If so, is it possible the internal pullup resistors are not enabled or something? i.e. not being pulled high/low (floating). 

I'm not sure if this MCU is "5V tolerant",  if it is, I supposed I could try that mode.  Finding it hard to find any schematics or pinouts for my MCU, this is my config for klipper. I presume the pin numbers are correct.... I didn't change them (or any of the config) between software updates.  I tinkered with probe_with_touch_mode to see if I could get something different to happen. 

Again, appreciate the help....  I know this looks like a hardware problem and I wouldn't be wasting your time if it occur exactly when I upgraded the klipper software/firmware.  

I suppose the pin on the microcontroller could be fried. I've seen this weird stuff on Atmel chips before (arduino) when I've accidentally fried one. 

Again, it's just strange to me that it was working.... then software/firmware upgrade, then not working. 

my current config:
[bltouch]
sensor_pin: ^PC4
control_pin: PA4
pin_move_time: 0.500
probe_with_touch_mode: False
pin_up_reports_not_triggered: True
pin_up_touch_mode_reports_triggered: True
x_offset: -44 
y_offset: -14
#z_offset: 1.27
speed: 2

WizardOfYendor1, 2020-12-07:
I've tested, with my oscilloscope,  the BL touch sensor pin in parallel with the MCU pins.... i.e. BLTouch is connected to the MCU and my scope probe is across the pins/wires.  No change. There's no signal/voltage on the zstop pins until I push the bl probe pin.... then red light comes on and a square wave pulse is seen. 

Query probe always reports "triggered" even with the BLTouch disconnected.... disconnected and power cycled - still reports triggered. 

Is the expected behavior that if there is nothing connected to the zstop header (open),  that query_probe command should always report "probe:open"?  Seems in my case, the exact opposite is happening. 

So I don't think it is a wiring or probe problem.... unless there is something I've missed. 

I'm not sure if it is an MCU problem....but given there is no voltage on the pins, I suspect it's OK. When I've fried arduino pins in the past, they usually latch up.  

I'm making a big assumption that a logic high signal on those pins is supposed to set the "triggered" flag when the probe is "down". And thus it should be "open" when the probe is "up". 

It seems like a software issue (microcontroller code or otherwise) but I have hard time accepting that because many others would be reporting this issue.  I'm trying to figure out how the triggered flag gets set but I'm not terribly familiar with python. 

Any more troubleshooting ideas or steps would be welcome? I'm considering just putting the Marlin 2.0 firmware on there and see if everything works again.

BlackStump, 2020-12-07:
There is so many versions of BLTOUCH then there are the CLONES [maybe you should be reading through the original manufactures documentation that is for the version you have](https://www.antclabs.com/manual)
I have 2 BLTOUCH's 1-original and 1-clone that works as intended but it is a mute point because of the amount of variation and quality of some of the clones out in the market. Maybe some lobbying from users to get the manufacture to test there products with Klipper would be beneficial.
From my experience it would suggest it is not a Klipper issue.  One can only looks back through issues posted about BLTOUCH.

WizardOfYendor1, 2020-12-07:
I actually did read thru that when I removed the C7 cap on the MCU. 

I believe I have a genuine Bltouch, I did make the effort when I bought it to ask if it is was a genuine version because I knew of all the clones..... but that doesn't mean a whole lot to unscrupulous sellers.  I haven't taken it apart to check the PCB to verify it, 

The only reason I've even posting about it here is because besides the briefly similar issue I had before, the BLTouch has been running like a champ for many months now and dozens of prints/bed levels.... everything was fine until the exact moment I upgraded the klipper software/firmware. Then it stopped working completely at that moment. No changes were made to the configuration either. 

I really don't understand why it always reports "triggered" with the sensor wire completely disconnected from the MCU. 

Maybe I'll try rolling back to a prior version to see if it makes any difference.

WizardOfYendor1, 2020-12-08:
I did verify I have a genuine BLtouch 3.1 (smart).  

What seems strange to me is that even with the bltouch completely disconnected (literally all the wires disconnected and the bltouch is completely free), is that klipper is reporting it in a "triggered" state. Even after power cycle of the MCU, restarting octopi etc...  triggered.. always.... even without the bltouch connected in anyway. 

Whether it's connected or not, the errors are the same. At startup klipper says:
Recv: // Failed to verify BLTouch probe is raised; retrying.
Recv: // Failed to verify BLTouch probe is raised; retrying.

Then query_probe reports triggered regardless of pin up or down... forever and always. 

I guess I can get a new bltouch. I guess it could of fried at the exact moment in time as the software/firmware update.

FanDjango, 2020-12-08:
> I put my oscilloscope on the sensor wire connector (probe on white, ground on black). With the pin down... there is no signal (well, other than some very low voltage noise). When I touch the end of [...]

The following is valid for a BLTouch V3.1:

Nice try, but you should add a 4.7k resistor pull up to 5V in that scenario. The BLTouch in its default config has an Open Drain (Google it!) output that can only pull down to GND and on a trigger it will float the signal line. Difficult to see anything sensible that way.

If you want to test it without a resistor, you would need to issue a probe command to set the probe from OD mode to 5V mode, it will then provide it's own (weak) pullup.

Also, if you don't go into SW mode after the deploy, you will only get a single 10ms pulse on trigger. Difficult to see that on a scope. If you go into SW ( = Switch ) mode, it will be more like a normally closed to GND switch that gets opened. If you don't want to do that, you should provoke the probe to into error mode, by deploying and then pushing in the pin until it starts blinking, which will then make the signal line go high permanently until a reset is issued. You should see that on the scope then.

> What seems strange to me is that even with the bltouch completely disconnected (literally all the wires disconnected and the bltouch is completely free), is that klipper is reporting it in a [...]

You have defined `^PC4` which means connecting nothing to that pin means it will pull up by itself to HIGH potential (no switch connected to pull it to GND) and so it will be TRIGGERED. Test PC4 by shorting it to GND. You should see it no longer being TRIGGERED.

WizardOfYendor1, 2020-12-09:
" Test PC4 by shorting it to GND. You should see it no longer being TRIGGERED." 
@FanDjango   
So I did exactly that... on the ender 3 1.1.4 board I have. From what I've determined,  the zstop pins (the ^PC4 + gnd) should be shorted together. One pin is ground (verified with ohm meter). 

I left the 3 wire connector connected so the probe would function and respond to commands. 

So not even trusting the wiring, I disconnected the zstop "sensor" wire connector that runs up to the bltouch. I put 2 dupont jumper wires on the pins and hooked them together (thus it is in a constant shorted to gnd state).  pin_down, query_probe = triggered... pin_up, touch_mode et al didn't matter always triggered.... power cycled everything, just for fun, everything = still triggered.  (following the steps in the documentation of course). 

Granted, I clearly don't fully understand how how this thing works with klipper but shouldn't the zstop pins shorted together force "probe: open"? 

If it should, then it's got to be a board problem or a software problem.  I guess the board could've fried at the exact moment I upgraded the software - just would be an amazing co-incidence.  

I tried older versions of klipper (.08) at the tag level and various random commit points. Flashed all that, verified the "new"(older) version is working/running (klipper spits it out when it starts up)..... and I have the same problem. 

If shorting the sensor pin to ground should make klipper report "open" instead "triggered", then I guess the board is hosed or I have the wrong pin defined in the configuration. 

I have not tried changing it from open drain to 5V mode - because I am not sure if the board is 5V logic tolerant... I find conflicting information.... so I never wanted to try it lest I risk frying the board/pin.  I do, BTW, "force" it to be in OD mode in the config. I know it's the default but figured what the hell... just to be sure.

FanDjango, 2020-12-09:
> Granted, I clearly don't fully understand how how this thing works with klipper but shouldn't the zstop pins shorted together force "probe: open"?

Forget the probe for a moment and do the same (with dupont jumpers), for example, with the x-stop pin. You could even use the x-stop pin in the probe section for a moment and try there. Try to compare / isolate the problem.

WizardOfYendor1, 2020-12-09:
Hmmm that's a GREAT idea! why didn't I think of that :-)

Ok so I simply changed my config for the Zaxis probe (bltouch config)  to use the XAxis endstop pin. Restarted everything. 

With the XAxis microswitch not pressed, query_probe reports = probe:open.  With the microswitch pressed.... it reports triggered. 

So that's tells me it is almost certainly a MCU board problem.... this presumes I am actually using the correct z stop pin in my config. I am a little uncertain about this as I cannot find any documentation on the pinouts on the ender 3 1.1.4 board. However, I didn't change the config at between software updates (not to my knowledge anyways). 

I don't think it could be the firmware because everyone would be having this problem if it was a GPIO pin misconfigured in arduino etc.. 

Unless you have any other suggestions to dig deeper, looks like its time to replace my MCU board.  What an amazing co-incidence this pin fried when I upgraded the software.... strange but it happens. 

Abject lesson in correlation is not necessarily causation.....

Also, sorry for wasting anyone's time.... as a software dev myself, I really hate when software gets blamed for hardware problems :-).

FanDjango, 2020-12-09:
I do believe it is PC4 for the ZSTOP on that board. Yes, you might have cooked a single input pin of the MC without compromising the rest of the chip. Can happen. You could try the DEPLOY/SW-MODE and the other probe tests with the black wire on GND and the white wire on the XSTOP. It would confirm that the probe is still ok and just PC4 is broke.

WizardOfYendor1, 2020-12-09:
Not sure what you mean by "DEPLOY/SW-MODE".... 

I did connect up the old microswitch for the zstop (and of course changed the config back to be the correct pins) and I get the same thing.... always triggered. The zstop microswitch is NC, so same effect as the pin jumper. 

Is there any nifty way to query the pin logic state directly? i.e. bypass the query_probe command and just get the logic state? This would confirm beyond any doubt the GPIO pin is nuked. 

I am concerned how the pin got nuked in the 1st place. e.g. the bltouch probe is hosed up and fried the pin with high voltage or something.

FanDjango, 2020-12-09:
> Not sure what you mean by "DEPLOY/SW-MODE"

pin-down, then touch mode. It will then behave like a switch that is normally closed (to GND).

> Is there any nifty way to query the pin logic state directly?

`QUERY_ENDSTOPS` maybe?

WizardOfYendor1, 2020-12-09:
Ah, Ok.. gotcha. 

Query endstops does the same thing... firmware is still in the middle. 

Send: query_endstops
Recv: x:open y:open z:TRIGGERED

I wonder if I could write a little arduino sketch that would poll and dump out the pin state to serial. Easy thing to do for arduino devices not sure if that's really possible with this MCU (though they are both using Atmel chips).

FanDjango, 2020-12-09:
> I am concerned how the pin got nuked in the 1st place. e.g. the bltouch probe is hosed up and fried the pin with high voltage or something.

Hard to say. Static Electricity, Rubber Shoes, Carpet and you got near the BLTouch or its connection wires?

> I wonder if I could write a little arduino sketch that would poll and dump out the pin state to serial. Easy thing to do for arduino devices not sure if that's really possible with this MCU (though they are both using Atmel chips).

Yes, that's easy to do but not really needed. You can trust the query_endstops output.

Check out the following:

Remove all cables from the X-STOP. query it. Should be "TRIGGERED", right? Now attach the X-STOP SWITCH, trigger it by pushing the nozzle to the home position (slowly) or pushing the endstop switch with your finger. Query again. It will say "open". But really, it is "closed". I kind of hate the "open" and "TRIGGERED" verbiage. It doesn't really reflect the facts. But it sill proves the XSTOP (PC2) is working. So if PC4 is "triggered" always, you could check that the endstop plug is not wobbly, that the endstop signal line of PC4 actually reaches all the way to the ATMEGA128 chip. I can see if I can find out the pin number for you, if you want.... If that Z-STOP plug socket has lost solder connection under the board due to mechanical stresses etc. you will get "TRIGGERED"....

FanDjango, 2020-12-09:
[image]

[image]

WizardOfYendor1, 2020-12-09:
I'll take a look at that but honestly, it's a $35 part for a new and improved board. I feel pretty confident the MCU is hosed all I really want to be sure of is the bltouch isn't broken and going to fry the pin again. 

Regardless, I don't have much standing to claim this is a software issue.  

I'll check that out and/or replace the board then I'll update and close the issue. 

Thanks for all the troubleshooting help.

wesk-ms, 2020-12-10:
I don't think it's your board.  As alluded to in the fourth post in this thread and in the documentation, just issuing a PIN_DOWN or PIN_UP command won't result in a changed state.  You have to issue the PIN_DOWN and follow it with a COMMAND=touch_mode, then push the pin up slightly, THEN issue a QUERY_ENDSTOPS command WHILE the pin is partially pushed up in order to see the state change.

Also, your bltouch should be reporting "open" when querying the endstop after issuing a  PIN_DOWN or PIN_UP command.  To fix this, invert the sensor pin in your config -- !^PC4.

Lastly, your original issue of "...failed to raise probe" is most likely a timing issue.  Comment out the "pin_move_time" line in your config to see if that helps.

To summarize my suggested next steps:
Hook everything back up as you had it previously.
Comment out the "pin_move_time" line in your config.
Invert your sensor_pin by adding a "!".
Reboot, restart, etc...
Try to home the z axis.  (This way, you don't have to mess with all of the manual commands)

If it still fails, post the errors you are receiving with the updated config.

WizardOfYendor1, 2020-12-12:
I had already ordered the new board when I saw your comment so I just decided replace it with the new 4.2.7 board.  I'm not sure why it would've worked in the past without a "!" inverting the sensor pin. I didn't even know about the ! nor was it in my config (didn't change between software updates). So I really think the board was hosed (that pin anyways). 

Now I'm trying to figure out how to hook up the bltouch to the new board. There's a dedicated header for it but I've have it spliced into pin 27 on my display ribbon cable.  As usual, find the pinout/mappings for this board is seemly impossible. 

If anyone feels generous on a pointer on what the pin mappings are for the 4.2.7, I would appreciate it. Trying to find a place to connect up the control pin and then of course the sensor/zstop (got that one figured out I think). 

I realize we've wandered off the path of my original issue, that's a non software issue. So I will close this in a few days. 

BTW for those who are contributors...  I would suggest adding this to the "generic-creality-v4.2.7.cfg" so the display works. I dug around to find this and it seems to be working on my printer OK. 
[display]
lcd_type: st7920
cs_pin: PB12
sclk_pin: PB13
sid_pin: PB15
encoder_pins: ^PB14, ^PB10
click_pin: ^!PB2

closed 2020-12-12 by WizardOfYendor1

FanDjango, 2020-12-12:
Happy to hear that