Klipper3d/klipper #3608: Steppers are skipping steps since last updated

LazaroFilm, 2020-12-06:
Since the last updated that required a MCU reflash, all my steppers are missing steps on simple moves, it sounds as if the steppers are trying to go faster than they can. 
I'm using TMC2208 in UART mode. I have not changed any printer.cfg settings before it stopped working.
I'm thinking that is *may* have something to do with `[ ] Specify a custom step pulse duration` in the new MCU config. 

Klippy.log ->  http://paste.c-net.org/MammalCordless (I did not choose that link lol)

LazaroFilm, 2020-12-06:
Bad bot, there is a klippy.log but here it is again.
[klippy.log](https://github.com/KevinOConnor/klipper/files/5649419/klippy.log)

robthide37, 2020-12-24:
are you using the rotation metric to define distance ?

LazaroFilm, 2020-12-27:
No, i'm not. It was not released when I posted this issue. I'll try to update with rotation metrics when I get back to the printer.

cristianku, 2021-01-14:
hi, similar issue here , I have downgraded Klipper to 0.9.1 - 106d1d2 , lets see if it helps

https://github.com/KevinOConnor/klipper/issues/3773

kaktus770, 2021-01-17:
Hello, I have same problem. I try use Klipper with Duet 2 WiFi, stepper motors tmc 2660 in UART mode. It's a clean install. Stepper motors XYZ 0.9, on extruder 1.8. Rotating, but with very little torque.

[klippy.log](https://github.com/KevinOConnor/klipper/files/5825657/klippy.6.log)

kaktus770, 2021-01-17:
The problem was solved by commenting  "idle_current_percent"

LazaroFilm, 2021-01-19:
I'm going to close this since things are now working fine after updating everything and changing to rotation. Not sure exactly what did it but it's working

closed 2021-01-19 by LazaroFilm