Klipper3d/klipper #3610: TMC2209 spreadsheet - value optimising. Missing CS value in klipper (or docs)

BelgianGuyWithALaser, 2020-12-07:
I'm finetuning klipper for my TMC2209 drivers in combination with my steppers.
However, while filling in the TMC2209 spreadsheet from trinamic to find my optimal value's I noticed there's no mention in klipper or the documentation for the CS value (current scaling). In marlin this was reported back from the eeprom to fill in.
Any chance that we could find out what this value should be?
[image]
[image]

jakep82, 2020-12-07:
TMC_DUMP will show you the CS value.

https://www.klipper3d.org/G-Codes.html#tmc2130-tmc2660-tmc2208-tmc2209-and-tmc5160

Sineos, 2020-12-07:
Documentation here: https://github.com/KevinOConnor/klipper/blob/master/docs/Config_Reference.md#tmc2209

ShohninDmitriy, 2020-12-07:
It is necessary to select CS so that the shunt resistance coincides approximately with the actual value of ~ 0.101 Ohms either in the 1st column or in the second

[image]

robthide37, 2020-12-07:
But then how do we set the current scaling ?

ShohninDmitriy, 2020-12-08:
> But then how do we set the current scaling ?

Change the CS value by substituting a value from 0-31 so that Rsense is equal to 0.101 Ohms +-3%

ShohninDmitriy, 2020-12-08:
1. Enter the highlighted system parameters and settings   like clock frequency and supply voltage as well as motor parameters. 
2.   Check the resulting sense resistor value (C46 or C47) and modify the current   **scale setting (CS in C26)** if you want to use a different **sense resistor value**
3.   **Check that all hints fileds show OK and modify toff setting (C20) if desired**.
4.   Use the resulting **hysteresis values (F32, F33) as a start value for parameter**   optimization

robthide37, 2020-12-08:
> 1. Enter the highlighted system parameters and settings   like clock frequency and supply voltage as well as motor parameters.
> 2. Check the resulting sense resistor value (C46 or C47) and modify the current   **scale setting (CS in C26)** if you want to use a different **sense resistor value**
> 3. **Check that all hints fileds show OK and modify toff setting (C20) if desired**.
> 4. Use the resulting **hysteresis values (F32, F33) as a start value for parameter**   optimization
> 
> as shown in the last picture Rsense using VSENSE=1 = 0.100 ohms

I am familiar with the spreadsheet i jsut am not aware how to change the cs value in my klipper config or on the 2660s. I was asking if you know where that value is ?

ShohninDmitriy, 2020-12-08:
These settings are not available in any firmware, it is calculated automatically when setting Irms

BelgianGuyWithALaser, 2020-12-08:
> These settings are not available in any firmware, it is calculated automatically when setting Irms

problem is, once you change your hysteresis value's and do a new TMC_dump. the CS_actual dropped (or went higher) so if you would "fill in" the new CS value you would get different hysteresis value's again.
Unless somehow you could say in klipper that the CS_actual = a hardcoded number

ShohninDmitriy, 2020-12-08:
In idle mode, when you remove the TMC dump, the current is set to **hold_current** and in motion mode to **run_current**, set both values to the same and it will show correctly, although this is calculated in the table and not read from the driver

ShohninDmitriy, 2020-12-08:
This CS value depends on the current that the driver sets at the moment in idle mode it is low in operation mode it is high, so high is calculated in the table according to the Rsense at the operating point.

BelgianGuyWithALaser, 2020-12-08:
> This CS value depends on the current that the driver sets at the moment in idle mode it is low in operation mode it is high, so high is calculated in the table according to the Rsense at the operating point.

I have done the TMC dump while it was idle. after changing the hysteresis settings I did a new TMC dump to see if it read the value's correctly and noticed that the CS_actual dropped.
so it's not that it's a "static" number. so as said, even when setting the "optimal value's" from the spreadsheet then, if I'd double check, the CS_actual value dropped and it's not "optimal" anymore

ShohninDmitriy, 2020-12-08:
Все я устал обьяснять,  the [Datashee](https://www.trinamic.com/fileadmin/assets/Products/ICs_Documents/TMC2209_Datasheet_V103.pdf)t

robthide37, 2020-12-09:
Ok so you are setting the current scaling to make the tables match up to your sense resistor

FHeilmann, 2020-12-14:
Be advised that klipper, when setting the current, automatically switches the `VSENSE` field to achieve better accuracy.
Relevant code here: https://github.com/KevinOConnor/klipper/blob/master/klippy/extras/tmc2130.py#L123
See the datasheet page 39 (orange box) to see why this is recommended.

The `VSENSE` field implements a scaling feature with the current fields.
When `VSENSE` is `true`, the CS register scales from 0% to ~55% max current.
When VSENSE is `false`, the CS register scales from 0% to 100% max current.

Consequently, if someone asked for a current value below ~55% of the max current, greater accuracy can be achieved by setting `VSENSE` to `true`, and recomputing the CS value.

Sineos, 2020-12-14:
> Consequently, if someone asked for a current value below ~55% of the max current, greater accuracy can be achieved by setting `VSENSE` to `true`, and recomputing the CS value.

Interesting. Thanks for sharing this information. 
Does this have an effect in reality? Isn't the typical run current always over 55% of the max current? Apart from that, how does the driver know the max current?

FHeilmann, 2020-12-14:
well there is a maximum current that can be achieved with the given sense resistor that is on the board. See here:

[image]

Given a maximum RMS current of 2A, I'd argue there are a lot of people that run their motors at 1A or lower, especially in direct drive extruder motor scenarios or pancake extruders.

Sineos, 2020-12-14:
Ah, k. So this refers to the theoretical current depending on driver chip and R-Sense value. So for my 5160 drivers with R-Sense of 0.075 Ohm, the motors are indeed always running below 55% of the RMS current (3.11 Amp). 

What does this mean in respect to the calculation sheet?
If my run-current is below the 55% of the drivers RMS current, does this mean I have to aim for a different CS value to do the tuning according to the Excel sheet?

robthide37, 2020-12-25:
> Ah, k. So this refers to the theoretical current depending on driver chip and R-Sense value. So for my 5160 drivers with R-Sense of 0.075 Ohm, the motors are indeed always running below 55% of the [...]

No you want to set your cs value to what it is so that the other values are calculated properly . you can do a dump with idle current set to 100 to verify that you have the right value figured out

Sineos, 2020-12-26:
My 5160 values:
[image]
[image]

For the following TMC dump, I set `hold_current: 3`, since Klipper will throw an error for values over 3:
```
Recv: // ========== Write-only registers ==========
Recv: // COOLCONF:   00000000
Recv: // GLOBALSCALER: 0000005c GLOBALSCALER=92
Recv: // IHOLD_IRUN: 00061f0e IHOLD=14 IRUN=31 IHOLDDELAY=6
Recv: // PWMCONF:    c40c001e PWM_OFS=30 pwm_autoscale=1 pwm_autograd=1 PWM_REG=4 PWM_LIM=12
Recv: // TPOWERDOWN: 0000000a TPOWERDOWN=10
Recv: // ========== Queried registers ==========
Recv: // GCONF:      00000000
Recv: // CHOPCONF:   12418105 toff=5 hend=2 tbl=3 tpfd=4 MRES=2(64usteps) intpol=1
Recv: // GSTAT:      00000005 reset=1(reset) uv_cp=1(Undervoltage!)
Recv: // DRV_STATUS: 810e0048 SG_RESULT=72 CSACTUAL=14 stallGuard=1 stst=1
Recv: // FACTORY_CONF: 0000000e FACTORY_CONF=14
Recv: // IOIN:       30000042 REFR_DIR=1 SD_MODE=1 VERSION=0x30
Recv: // LOST_STEPS: 00000000
Recv: // MSCNT:      00000356 MSCNT=854
Recv: // MSCURACT:   007d012a CUR_A=-214 CUR_B=125
Recv: // OTP_READ:   0000000e OTP_FCLKTRIM=14
Recv: // PWM_SCALE:  0000000d PWM_SCALE_SUM=13
Recv: // PWM_AUTO:   0000001d PWM_OFS_AUTO=29
Recv: // TSTEP:      000fffff TSTEP=1048575
```

It seems the values are accepted and the stepper also work quite good. I have the impression that they get less hot.

One remaining question:
In the `printer.cfg`, I effectively set `run_current: 1.1`. Does the calculation work with the specification values of the stepper:
- Icoil (peak)[A] := 2.1
- Icoil   (RMS)[A] = 1.485

or should I aim at the configured maximum of `run_current: 1.1`?

Sineos, 2020-12-29:
> One remaining question:
> In the `printer.cfg`, I effectively set `run_current: 1.1`. Does the calculation work with the specification values of the stepper:
> 
>     * Icoil (peak)[A] := 2.1
> 
>     * Icoil   (RMS)[A] = 1.485
> 
> 
> or should I aim at the configured maximum of `run_current: 1.1`?

I contacted Trinamic on this issue. Result:

- `Icoil (peak)[A]` and `Icoil   (RMS)[A]` should match the CS, Rsense and Vsense value as per Rsense table
- If the stepper is operated at different currents, then the effective real-world current shall be used, i.e. Klipper's `run_current`

Given my understanding is correct, this brings me to the point where the TMC5160 are a complete design flaw for common 3D printers. The typical Rsense of 0.075 Ohm results in "optimal" working conditions at Icoil(RMS) = 3.11A. 
So running at a RMS current somewhere around 1A +/- 0.5A should severely limit the drivers to scale the current.

closed 2021-01-28 by github-actions[bot]