Klipper3d/klipper #3613: Ender 3 Pro - 8bit - board v1.1.4

er9xuser, 2020-12-08:
When using configuration file printer-creality-ender3pro-2020.cfg i get error:

Unknown value 'PB12' in enumeration 'pin'
This type of error is frequently caused by running an older
version of the firmware on the micro-controller (fix by
recompiling and flashing the firmware).
Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Protocol error connecting to printer
Klipper state: Not ready

For get it to work i have to use printer-creality-ender3-2018.cfg. Just for info.

Thanks

er9xuser, 2020-12-08:
I've deleted klipper now. I will get 4.2.7 board and I'll test again with new board. Do you have maybe instructions for 32bit board?

closed 2020-12-15 by klipper-gitissuebot

crawler, 2021-01-17:
@er9xuser you need to use the Ender 3 2018 config template for the 1.1.4 8bit board.
https://github.com/KevinOConnor/klipper/issues/3742#issuecomment-761772207