Klipper3d/klipper #3614: Feature: support half-stepping rotary encoders

Whitehawk2000, 2020-12-08:
My encoder needs two rotations for one step in the menu. 
I have been able to correct this by following guides in these two issues:

#730
#1727

But due to recent changes in the file-structure of klipper the i can't get these workarounds to function anymore.

modmt, 2020-12-27:
For me the same issue. I used the workaround (handmade implementation of nickbrennan01) and it worked great. But with new structure it does not work anymore.

Would it be possible to integrate this feature in klipper as default?

closed 2021-01-25 by github-actions[bot]

thecrocodille, 2021-04-11:
> For me the same issue. I used the workaround (handmade implementation of nickbrennan01) and it worked great. But with new update it does not work anymore.
> 
> Would it be possible to integrate this feature in klipper as default?

Hi, the main reason it's not working after the last update is migration of buttons block from 'klippy/extras/display/menu.py' to 'klippy/extras/display/menu_keys.py'.
So, you have to find 'Register rotary encoder' block in 'menu_keys.py' and add changes from issue #730  to this file.