Klipper3d/klipper #3618: problem defining virtual endstop for sensorless homing with manual stepper

stableconnection, 2020-12-09:
I defined the endstop_pin according to the sensorless homing file provided and it gives me an 'Unknown pin chip name' error. (I tried with basically every underline combination and it throws the same error. I probably made an idiotic mistake somewhere but i can't figure it out. Thanks in advance.
The config section:

[manual_stepper w]
step_pin: P0.1
dir_pin: !P0.0
enable_pin: !P0.10
step_distance: .005 # .0025
endstop_pin: tmc2130_manual_stepper_w:virtual_endstop
position_endstop: 0
position_max: 2.52
homing_retract_dist: 0
homing_speed: 2

[tmc2130 manual_stepper w]
cs_pin: P1.1
spi_software_miso_pin: P0.5
spi_software_mosi_pin: P4.28
spi_software_sclk_pin: P0.4
diag1_pin: ^!P1.26
microsteps: 16
run_current: 0.320 
hold_current: 0.200
stealthchop_threshold: 0
driver_SGT: 5

stableconnection, 2020-12-09:
maybe i am approaching my problem the wrong way...the point of all this is to integrate the stepper for the toolchange mechanism and in this case it has no endstop switch but should home sensorless
just to clarify if someone knows a better way to solve my problem

stableconnection, 2020-12-09:
[klippy.log](https://github.com/KevinOConnor/klipper/files/5666426/klippy.log)
just so the bot is happy :)

kakou-fr, 2020-12-10:
an example that works for me (for example, define tmc... before the stepper section):  
```
[tmc2209 manual_stepper idler_stepper]
uart_pin: mmboard:PB7
microsteps: 16
run_current: 0.800
hold_current: 0.800
interpolate: True
sense_resistor: 0.110
stealthchop_threshold: 100
diag_pin: ^mmboard:PA1 # Y+
driver_SGTHRS: 75

[manual_stepper idler_stepper]
step_pin: mmboard:PB13
dir_pin: mmboard:PB14
enable_pin: !mmboard:PB12
# 25 step/mm
step_distance: .04
velocity: 100
accel: 500
endstop_pin: tmc2209_idler_stepper:virtual_endstop
homing_retract_dist: 0
```

stableconnection, 2020-12-11:
thanks a lot, apparently the issue is resolved, config doesnt throw an error anymore

closed 2020-12-11 by stableconnection