Klipper3d/klipper #3620: Unknown command: queue_soft_pwm_out

Oranginator, 2020-12-09:
Hi together,

I updated klipper to the most recent build  with:
`cd ~/klipper && git pull && sudo service klipper restart`

Afterwards I could not connect to the printer using octoprint. It always gives back the error:

>  19:51:33
>  Unknown command: queue_soft_pwm_out
>  This type of error is frequently caused by running an older
>  version of the firmware on the micro-controller (fix by
>  recompiling and flashing the firmware).
>  Once the underlying issue is corrected, use the "RESTART"
>  command to reload the config and restart the host software.
>  Protocol error connecting to printer
>  Klipper state: Not ready

>  19:51:33
>  Unknown command: queue_soft_pwm_out

I already updated the firmware via:
`make clean`
`make menuconfig`
`make`

This didn't solve the problem. Before the update everything was fine.

Thanks for your help, best regards
Steffen

[klippy.log](https://github.com/KevinOConnor/klipper/files/5668046/klippy.log)

theopensourcerer, 2020-12-10:
Looks like you didn't flash the new firmware to the MCU.

Oranginator, 2020-12-10:
> 
> 
> Looks like you didn't flash the new firmware to the MCU.

Interessting, I had it always flashed with SD-Card named FIRMWARE.CUR and it worked. After renaming it to FIRMWARE.BIN it worked.

Thank you. Case closed.

closed 2020-12-10 by Oranginator