Klipper3d/klipper #3622: Ender 5 Pro - BL Touch 3.1 - Board 4.2.2 (STM32)

chriscaj, 2020-12-10:
Hi,

I'm trying to configure klipper, but I'm havving some problems on BL Touch (original) configuration.

The commands pin_down and pin_up are working well. When I use the command BLTOUCH_DEBUG COMMAND=touch_mode and then Send: QUERY_PROBE, I get probe: TRIGGERED.

I tried all combinations of pin_up_touch_mode_reports_triggered and pin_up_touch_mode_reports_triggered.

The last thing I tried was using the branch work-bltouch-20201209, but anything changed. 

Could'd you help me?

Thank's
Christian

[klippy.log](https://github.com/KevinOConnor/klipper/files/5669416/klippy.log)

Temperature
Control
GCode Viewer
Terminal
Timelapse
Klipper
Changing monitoring state from "Offline" to "Detecting serial connection"
Performing autodetection with 7 port/baudrate candidates: /tmp/printer@115200, /tmp/printer@250000, /tmp/printer@230400, /tmp/printer@57600, /tmp/printer@38400, /tmp/printer@19200, /tmp/printer@9600
Trying port /tmp/printer, baudrate 115200
Connecting to port /tmp/printer, baudrate 115200
Handshake attempt #1 with timeout 2.0s
Connected to: Serial<id=0xabdbd050, open=True>(port='/tmp/printer', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=2.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Recv: ok
Send: N0 M110 N0*125
Changing monitoring state from "Detecting serial connection" to "Operational"
Send: N0 M110 N0*125
Recv: ok
Recv: ok
Send: N1 M115*39
Recv: ok FIRMWARE_VERSION:v0.9.1-105-g582b29cd FIRMWARE_NAME:Klipper
Send: N2 M21*18
Recv: ok
[...]
Send: BLTOUCH_DEBUG COMMAND=pin_down
Recv: // Sending BLTOUCH_DEBUG COMMAND=pin_down
Recv: ok
[...]
Send: BLTOUCH_DEBUG COMMAND=pin_up
Recv: // Sending BLTOUCH_DEBUG COMMAND=pin_up
Recv: ok
Send: BLTOUCH_DEBUG COMMAND=pin_down
Recv: // Sending BLTOUCH_DEBUG COMMAND=pin_down
Recv: ok
[...]
Send: BLTOUCH_DEBUG COMMAND=touch_mode
Recv: // Sending BLTOUCH_DEBUG COMMAND=touch_mode
Recv: ok
[...]
Send: QUERY_PROBE
Recv: // probe: TRIGGERED
Recv: ok
[...]
Send: G28
Recv: // Failed to verify BLTouch probe is raised; retrying.
Recv: // Failed to verify BLTouch probe is raised; retrying.
Recv: !! BLTouch failed to raise probe
Recv: ok
[...]
Send: BLTOUCH_DEBUG COMMAND=reset
Recv: // Sending BLTOUCH_DEBUG COMMAND=reset

FanDjango, 2020-12-11:
I've checked your pin definitions and so on. Looks ok. I've got an Ender 5 but alas I have replaced the board with a different one. But I remember the ^PB1.

```
[bltouch]
sensor_pin = ^PB1
control_pin = PB0
x_offset = -42
y_offset = -10
z_offset = 0
probe_with_touch_mode = True
pin_up_touch_mode_reports_triggered = False
```

Remove the last one for now. It should not be needed. 

You implied a BLTouch V3.1?

`probe_with_touch_mode = True` is a good idea but your problem lies at an earlier phase of what is happening.

What is happening: 

Right at initialisation of the `bltouch.py` module, it tries to `pin_up`. It then checks the status of the probe.

For some reason, your probe seems to be triggered, even if the pin is up. Shouldn't be like that, so you get the error: Failed to raise the probe (although it is raised). But that's ok, it indicates a problem further down the line.

The problem is: Your probe seems to be triggered **even in those cases, where it should be pulling down `PB1` to a low voltage level**, like an endstop switch (which is normally closed when not triggered) being closed to ground. But it isn't doing that, ever. Neither with `pin_down` nor with `pin_up`.

You really need to check your `PB1` terminal on the board with something other than the sensor wire of the BLTouch. Firstly, you could leave the terminal unattached to anything. And you could then short it to GND manually (and carefully). But you could temporarily put a real endstop switch on in - use the X endstop for example and QUERY that.

FanDjango, 2020-12-11:
What I am saying, is that you need to verify that Klipper can read PB1 for normal physical things like a real switch before one gets further down the line.

You are the second to have this problem right now and we need to get to the bottom of this somehow.

FanDjango, 2020-12-11:
Is this it?

[image]

wesk-ms, 2020-12-12:
Invert the sensor pin with !^PB1 in your config and see if that fixes everything.

chriscaj, 2020-12-14:
> Is this it?
> 
> [image]

I'm using G,V IN (with blue, red and yellow). And the other wires are at Z

chriscaj, 2020-12-14:
> Invert the sensor pin with !^PB1 in your config and see if that fixes everything.

I did that.

After BLTOUCH_DEBUG COMMAND=pin_down, BLTOUCH_DEBUG COMMAND=touch_mode, QUERY_PROBE, now I allways get probe: open.
[klippy.log](https://github.com/KevinOConnor/klipper/files/5688655/klippy.log)

FanDjango, 2020-12-14:
If you have not plugged it in where it say G ( = GND, black) and OUT ( = white, probe-signal), but instead you have plugged it in on the Z-MIN stop, **then shouldn't you say ^PA7 instead of ^PB1?**

But I would recommend using the pins that they have provided (PB1). There will be a better response, no capacitor on the signal line, perhaps. I haven't got the schematic.

wesk-ms, 2020-12-14:
> > Invert the sensor pin with !^PB1 in your config and see if that fixes everything.
> 
> I did that.
> 
> After BLTOUCH_DEBUG COMMAND=pin_down, BLTOUCH_DEBUG COMMAND=touch_mode, QUERY_PROBE, now I allways get probe: open.
> [klippy.log](https://github.com/KevinOConnor/klipper/files/5688655/klippy.log)

Great.  So, if you gently push up on the pin and run the query_probe command, does it report triggered?

chriscaj, 2020-12-17:
> If you have not plugged it in where it says G ( = GND, black) and OUT ( = white, probe-signal), but instead you have plugged it in on the Z-MIN stop, **then shouldn't you say ^PA7 instead of ^PB1?**
> 
> But I would recommend using the pins that they have provided (PB1). There will be a better response, no capacitor on the signal line, perhaps. I haven't got the schematic.

Perfect!
I plugled in G and OUT at BL_Tinstead Z-MIN and it worked!

Thanks!

closed 2020-12-17 by chriscaj