Klipper3d/klipper #3625: Feature Request: Support directories from gcode files directory, to make file selection via display easier

brandonheller, 2020-12-11:
Happy recent Klipper user here, trying to understand if there's a way to list directories on a display.  M20 doesn't seem to mention it, and using master from yesterday didn't show dirs:
https://github.com/KevinOConnor/klipper/blob/master/docs/G-Codes.md

FWIW, I'm using a Voron Zero Display, which is a ~4x16 char display.

This is the main functionality I've used with a Duet/PanelDues for years, so I assumed it would be present.

Does this functionality exist in any form?  What areas of code would be places to start?  Thanks.

brandonheller, 2020-12-11:
One other related question - From the v0.9 release notes (https://github.com/KevinOConnor/klipper/blob/master/docs/Releases.md):

> The LCD display and menu are now configurable using the Jinja2 template language.

Is there any documentation on this, or on display usage in general?  Aside from sample configuration files (https://github.com/KevinOConnor/klipper/blob/master/config/sample-lcd.cfg) I couldn't find anything, but I'd love to see what goes on different display types and learn what the expected functionality even is.  For example, in the SD menu, there's a Start Printing option, plus files shown, but I'm not sure how these relate.

brandonheller, 2020-12-11:
Is there some other place to track feature requests then?  I don't see any persistent location noted here:

https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md

brandonheller, 2021-01-07:
Trying again @KevinOConnor - how does one make a feature request to Klipper?

KevinOConnor, 2021-01-07:
FWIW, I don't see value in tracking "feature requests", as without an active contributor working on implementing a feature, my experience is that those requests gather for years, become stale, and make it harder to search for topics that are under active work.

As for printing from files in a sub-directory, my understanding is that it is now possible using the "mainsail" or "fluidd" interfaces to Klipper (or via another interface that uses Klipper's [api server](https://www.klipper3d.org/API_Server.html).  It is not possible to print from a sub-directory using the old M20 style G-Code commands, and I don't think it's a good idea to try and retrofit such functionality into these old arcane commands.

-Kevin

brandonheller, 2021-01-07:
Thanks Kevin - that does answer my question.

The value in having feature requests is exactly this - you get a
persistent, referenceable thread with pointers for those who might want to
take on a feature, who aren't you.  And by posting here, I learned why I
didn't see a supported G-code for listing directories, yet I can do that in
Mainsail.  And I have a place to ask why G-codes are even relevant to using
a directly connected display, like the 4-line LCD I have, if it's all in
Klipper source.  I haven't dived into the source yet, but that would be my
next question.  And with this thread, someone down the line can get access
to the answer for why something that seems missing may actually be hard, or
not make sense.

Lots of projects have email groups, an official Discord, or (like Duet) a
web-based discussion board for these kinds of things that encourage
development.  Is there an intended place to go for questions?  Looking at
the readme.md on the GitHub, there's no link to something similar, so I
have to wonder if there's no desire for a developer community, or if I just
don't know where it lives.  I love lots of what I've seen so far and I'm
using Klipper happily on a V0 with Mainsail, and I'd like to maybe
understand the resonance testing code a bit more, or just better understand
things like why an I2C error seems to halt the system.

Thanks.

KevinOConnor, 2021-01-07:
I understand, but in my experience those benefits are outweighed by the disadvantages mentioned in my message above.

Contact information is available at: https://www.klipper3d.org/Contact.html .  I'm not sure how that got missed as it's provided in several parts of the documentation.  Indeed, it was provided when this ticket was opened and again by the githubissue bot above.

-Kevin

brandonheller, 2021-01-07:
(no tone here, trying to explain)

I didn't think to click on the link in the automated response above,
which says "for information on obtaining the log..", because I don't need
information on obtaining the log.  There's no relevant log.  So I wouldn't
think to click on that.

I thought a feature request would make sense to submit to GH issues,
because 8 other open issue in the Klipper GH have FR in the title:
https://github.com/KevinOConnor/klipper/issues?q=is%3Aissue+is%3Aopen+FR

And on the main web page, klipper.org, as well as the GH readme, there is
no direct link to Contact or Mailing List.  If I click on Documentation, it
has a Contact link, but I'm not looking for documentation.  I'm looking for
Community or Mailing List or Get Involved or Learn More.

I'll join the mailing list, but my 2c if you want more community
involvement (which should be huge given how many people know the
languages involved and use printer software now) - put a clear link on the
front pages + add a note about FRs and mailing list to the auto-responder.

closed 2021-02-06 by github-actions[bot]