Klipper3d/klipper #3629: Why use RPi as a secondary MCU?

Bald-79, 2020-12-13:
after you add the lines:
[mcu host]
serial: /tmp/klipper_host_mcu
klipper no longer connected.
I did everything as it says here
https://github.com/KevinOConnor/klipper/blob/master/docs/RPi_microcontroller.md
what's wrong?
[klippy.log](https://github.com/KevinOConnor/klipper/files/5684731/klippy.log)

dmbutyugin, 2020-12-16:
If you did the parts [Install the RC script](https://github.com/KevinOConnor/klipper/blob/master/docs/RPi_microcontroller.md#install-the-rc-script) and built the [microcontroller code](https://github.com/KevinOConnor/klipper/blob/master/docs/RPi_microcontroller.md#building-the-micro-controller-code) for Linux process and ran `make flash`, maybe you need to restart the RPi?

Bald-79, 2020-12-16:
> If you did the parts [Install the RC script](https://github.com/KevinOConnor/klipper/blob/master/docs/RPi_microcontroller.md#install-the-rc-script) and built the [microcontroller [...]

make flash does not work for me because skr 1.4, I was flashing through a memory card.  the script is copied to the desired folder and added to startup.  Well, I rebooted it 20 times.

dmbutyugin, 2020-12-16:
Well, you do need to follow the guide exactly. That `make flash` actually 'flashes' an RPi: copies the Linux microcontroller code to the appropriate place. And it won't require an SD card.

Note that next time you need to update the SKR, you will need to run `make menuconfig` again, and choose your board there, run make clean / make / make flash as suggested by the update guide.

Bald-79, 2020-12-17:
> Well, you do need to follow the guide exactly. That `make flash` actually 'flashes' an RPi: copies the Linux microcontroller code to the appropriate place. And it won't require an SD card.
>
> Note [...]

My board is not connected via yusb, but /dev/ttyAMA0. which one should I indicate FLASH_DEVICE=? error without it.

Sineos, 2020-12-17:
> My board is not connected via yusb, but /dev/ttyAMA0. which one should I indicate FLASH_DEVICE=? error without it.

Using the RPI as secondary MCU has absolutely nothing to do with the **printer controller board** that you are using or how you connect the board.

When you use `make menuconfig` and choose `linux` then `make` and `make flash`, a new piece of software will be install on your Raspberry. This piece of software allows Klipper to use the RPi like a second board, e.g. for additional sensors etc.

Your printer controller board, the connection of your printer controller board, settings etc. is totally unaffected by this nor related to this.

Bald-79, 2020-12-17:
> Using the RPI as secondary MCU has absolutely nothing to do with the **printer controller board** that you are using or how you connect the board.
>
> When you use `make menuconfig` and choose `linux` [...]

Thank you. happened. I did not choose "Linux process". sorry i'm a fool

Bald-79, 2020-12-17:
everything worked out. it's a pity that gcod cannot be sent with the board off, because nothing prevents it from being executed if you need to change the state of the raspberry legs