Klipper3d/klipper #3630: More of a shout for help on BL Touch Sovol SV01

parahooners, 2020-12-13:
So iv got this far,

And now i am Trying to Config the printer to work with the BLTouch  sensor, and i cant seem to find the right setting, to be honest now im not sure that i even have the right pin out for the board... iv read most the guides but probable need an idiots guide to this below is a copy of the config file and the error i get is below follow by the set up..

Now i know the sensor works as iv got it going in merlin, iv been through the Ino files for marlin but dont seem to be able to find the pin map out .....

Recv: // Failed to verify BLTouch probe is raised; retrying.
[...]
Send: probe
Recv: // Printer is not ready
Recv: // The klippy host software is attempting to connect.  Please
Recv: // retry in a few moments.
Recv: !! Printer is not ready
Recv: ok
Recv: // Failed to verify BLTouch probe is raised; retrying.
[...]
Recv: // Klipper state: Ready

# This file contains common pin mappings for Sovol Sv01 printer. To
# use this config, the firmware should be compiled for the AVR
# atmega2560.

# See docs/Config_Reference.md for a description of parameters.

[stepper_x]
step_pin: ar54
dir_pin: ar55
enable_pin: !ar38
step_distance: .0125
endstop_pin: ^ar3
position_endstop: 0
position_max: 300
homing_speed: 50

[stepper_y]
step_pin: ar60
dir_pin: ar61
enable_pin: !ar56
step_distance: .0125
endstop_pin: ^ar14
position_endstop: 0
position_max: 255
homing_speed: 50

[stepper_z]
step_pin: ar46
dir_pin: !ar48
enable_pin: !ar62
step_distance: .0025
endstop_pin: probe:z_virtual_endstop
position_max: 300

[bltouch]
sensor_pin: ^PC5
control_pin: PA5
x_offset: -47.0
y_offset: -5.0
z_offset: 1.65
speed: 5.0
samples: 2
sample_retract_dist: 8.0

[homing_override]
set_position_z:0
gcode:
 G1 Z10 F600
 G28 X Y
 G1 X166 Y120 F6000
 G28 Z

[extruder]
step_pin: ar26
dir_pin: !ar28
enable_pin: !ar24
step_distance: .0024
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: ar10
sensor_type: EPCOS 100K B57560G104F
sensor_pin: analog13
control: pid
pid_Kp: 31.147
pid_Ki: 2.076
pid_Kd: 116.803
min_temp: 0
max_temp: 265

[heater_bed]
heater_pin: ar8
sensor_type: ATC Semitec 104GT-2
sensor_pin: analog14
control: pid
pid_Kp: 72.174
pid_Ki: 1.816
pid_Kd: 717.224
min_temp: 0
max_temp: 110

[filament_switch_sensor my_sensor]
switch_pin: ar2
pause_on_runout: True

[fan]
pin: ar9

[mcu]
serial: /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_AB0K4CJ6-if00-port0
pin_map: arduino

[printer]
kinematics: cartesian
max_velocity: 300
max_accel: 3000
max_z_velocity: 5
max_z_accel: 100

[display]
lcd_type: st7920
cs_pin: ar16
sclk_pin: ar23
sid_pin: ar17
encoder_pins: ^ar33, ^ar31
click_pin: ^!ar35

parahooners, 2020-12-13:
[image]

FanDjango, 2020-12-13:
Post your Marlin configs somewhere and I will look at them to see what you had for Marlin

closed 2020-12-20 by klipper-gitissuebot