Klipper3d/klipper #3631: Ender 3 + MKS GEN L: Unable to read tmc uart 'stepper_x' register IFCNT

Zwen1111, 2020-12-13:
Can somebody help me with this error:
` Unable to read tmc uart 'stepper_x' register IFCNT`

I tried to do everything in the [video](https://youtu.be/5bIzcA8PBgY) of TeachingTech. I copied his printer.cfg file and changed a few settings that have another name these days. I don't have BLTouch so I removed that block of settings. If you are interested in the config, I attached it as a .txt. 

Also I added the klippy.log file. I don't have any idea what is causing the problem. I tried to find any other issues with the same kind of problem, but nothing helps. I hope somebody of you could 😃 

[klippy.log](https://github.com/KevinOConnor/klipper/files/5685188/klippy.log)
[printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/5685189/printer.cfg.txt)
[image]
[image]
[image]

NBouquain, 2020-12-13:
From the picture, you have two MKS pins per TMC UART pin, but in you config file only one is declared.
You need to add the _tx_pin_ parameter in each TMC2208 sections.

If your steppers are connected as per the video, the configuration should be as follow (tx_pin should be the one with the resistor). Note that your TMC2208 section for the extruder is incorrectly named stepper_extruder, extruder is enough.

[tmc2208 stepper_x]
uart_pin: ar63  #   The pin connected to the TMC2208 PDN_UART line. (RX)
tx_pin:   ar40  #   The TX pin if separate

[tmc2208 stepper_y]
uart_pin: ar64  #   The pin connected to the TMC2208 PDN_UART line. (RX)
tx_pin:   ar59  #   The TX pin if separate

[tmc2208 stepper_z]
uart_pin: ar65  #   The pin connected to the TMC2208 PDN_UART line. (RX)
tx_pin:   ar42  #   The TX pin if separate

[tmc2208 extruder]
uart_pin: ar66  #   The pin connected to the TMC2208 PDN_UART line. (RX)
tx_pin:   ar44  #   The TX pin if separate

Zwen1111, 2020-12-20:
> From the picture, you have two MKS pins per TMC UART pin, but in you config file only one is declared.
> You need to add the _tx_pin_ parameter in each TMC2208 sections.
>
> If your steppers are [...]

At first I thought that this would fix it, but unfortunately it didn't. Anything I could do to find out if the wiring is working as it should? 

The LCD isn't showing anything, that is not good as well. Any solution?

NBouquain, 2020-12-20:
The LCD not showing anything is a typical behavior in case of Klipper incomplete initialization due to an error, in your case the TMC2208 configuration failure.

Just for confirmation, have you soldered to two pads on the bottom of each driver to enable UART mode as described here? https://www.bigtree-tech.com/products/bigtreetech-tmc2208-v3-0.html

Zwen1111, 2020-12-21:
> The LCD not showing anything is a typical behavior in case of Klipper incomplete initialization due to an error, in your case the TMC2208 configuration failure.
>
> Just for confirmation, have you [...]

It looks like there is already a resistor soldered on it. I tried to wire it without a resistor, but it didn't work

[image]

NBouquain, 2020-12-21:
Weird, perhaps it is a 0 Ohm resistor to replace the solder joint, in that case no soldering is required.
Based on this picture, the UART should be available on the MS3 pin (instead of the PDN pin).

If the wiring has not been modified compared to your first post, you are currently connected to the PDN pin (ie the middle one), 
could you try to connect to the the left pin ?

Zwen1111, 2020-12-21:
Should I revert the changes to the configuration file that you suggested?

NBouquain, 2020-12-21:
Not for now, it should work regardless of the presence or not of a resistor on the driver.
Once your setup is working, you could try to replace your wires with a single dupon wire (without inline resistor), and the configuration file should then be updated to remove  the 'tx_pin'

Zwen1111, 2020-12-21:
The error remains unfortunately...

Zwen1111, 2020-12-21:
I solved it! I came across [this](https://www.youtube.com/watch?v=6RcrgmNvyeA) video that explained two of the three jumpers were needed:
[image]
On my board all three of the jumpers were connected. I removed the one that was missing. I wired it up like @NBouquain explained. So the cables had to be on the left (MS3) pin.

closed 2020-12-21 by Zwen1111

NBouquain, 2020-12-21:
All right, it was so obvious that I forgot about that one :-(
To avoid any interaction in UART mode you should remove all the jumper. They are no longer required since you can change the stepper microsteps through the 'microsteps:' parameter