Klipper3d/klipper #3634: Repeatable firmware crash, unable to finish print

Sch-LikA, 2020-12-14:
Dear Klipper team,

First thanks a lot for your work, I really enjoy Klipper !

I recently setup Octoprint on a OrangePi Zero (4 cores) with a current Klipper. My printer is an Ender 3 pro with a few upgrades.

I have been trying to print the "CE3PRO_pet-displayBase.gcode" file generated with Cura, but printing stops at around 39 minutes, OctoPrint sending the dready 

_"Your printer's firmware reported an error. Due to that the ongoing print job will be cancelled. Reported error: Lost communication with MCU 'mcu'"_

error, and terminal loops with :

> "Send: M105
> Recv: ok B:0.0 /60.0 T0:0.0 /205.0
> Send: M105"

> FIRMWARE_RESTART

gets the printer back, with a dead print ...

Here is the slightly edited klippy.log (the original was ~40Mo) :

[klippy.log](https://github.com/KevinOConnor/klipper/files/5687390/klippy.log)

Here is the gcode file trying to be printed (just for the records, I tried printing it 90' to see if it would help) 

[CE3PRO_pet-displayBase.zip](https://github.com/KevinOConnor/klipper/files/5687398/CE3PRO_pet-displayBase.zip)

Thanks for helping !

KevinOConnor, 2020-12-17:
Unfortunately, there's little we can do with a partial log file.  That said, this sounds like the FAQ: https://www.klipper3d.org/FAQ.html#i-keep-getting-random-lost-communication-with-mcu-errors

-Kevin

Sch-LikA, 2020-12-26:
Thanks a lot for the answer. I used another USB header on my OrangePi and did not connect +5v, user a much shorter USB cable and updated to latest git, so far so good !

closed 2020-12-26 by Sch-LikA