Klipper3d/klipper #3635: Scheduled digital out event will exceed max_duration

Afterschnitzel, 2020-12-14:
Hi.
I have disconnecting issues randomly during the print after updating klipper. Can you please help me?
Enisy Rambo - custom MK3S printer.

 19:34:11
Previous MCU 'mcu' shutdown: Scheduled digital out event will exceed max_duration
19:34:12
Previous MCU 'mcu' shutdown: Scheduled digital out event will exceed max_duration
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown

[klippy.log](https://github.com/KevinOConnor/klipper/files/5690888/klippy.log)

KevinOConnor, 2020-12-15:
The log indicates a large number of messages are being lost between the host and micro-controller.  It looks like you're using a Prusa variant of the einsy rambo which has a known bug in the firmware of the usb-to-serial chip.  My understanding is that board requires some updates to run successfully with Klipper.  @Arksine may have more info.

-Kevin

Afterschnitzel, 2020-12-16:
Thank you Kevin. I already notified the Einsy board has performance issues. I am unable use more than 16 microstepping on extruder using 3.5:1 gear ratio and 0.8°stepper. 
So I ordered BTT002 board and I will see B-)

Arksine, 2020-12-16:
Kevin is correct.  Prusa flashes a custom version of "usbserial" firmware on the Atmega 32u2.  Unfortunately this firmware corrupts serial transfers and it gets progressively worse over time.  If you want to use the Einsy with Klipper you can flash a [modified version](https://github.com/PrusaOwners/mk3-32u2-firmware) of the Firmware.  This does require a programmer (or an Arduino flashed with the ArudinoISP sketch).  If you aren't comfortable doing this waiting on the BTT002 is probably your best bet.

Afterschnitzel, 2020-12-16:
Thank you guys for your advices. I will try to tune Einsy before my BTT002 arrive :-)

Afterschnitzel, 2020-12-20:
Hi. 
I played with Arduino Leonardo this weekend and I succesfully flashed 32u2. So I will see if the problem is solved or not :-)

Afterschnitzel, 2020-12-21:
So problem was solved, thanks again.

closed 2020-12-21 by Afterschnitzel

dvoelkel, 2021-01-03:
Had the same problem, found this hint,
followed the solution, and fixed the problem.

First print died after 90 minutes, Now at 120 minutes and still printing.

Thanks mates :)

dvoelkel, 2021-01-03:
Narf. Died after 150 minutes :-( Same problem :-(

Afterschnitzel, 2021-01-05:
Hi. Maybe you didn' reflash 32u2 properly. But I don't know how to check it.
I think this solution solved my problem, I did 5 Hour print without issue. But in next 2 days after flash my BTT002 arrived and I am not using Einsy Rambo anymore. So I didn't make any longer prints on it.
I must day that BTT002 is great print quality improvement with Klipper on MK3S.

dzhaparoff, 2021-04-11:
@Afterschnitzel Hi! Sorry for offtop, my question is about BTT002 + Klipper. How is it? It is worth to upgrade from stock prusa einsy to BTT002? How reliably BTT002 works with Klipper? All features implemented and working properly?

Afterschnitzel, 2021-04-11:
> @Afterschnitzel Hi! Sorry for offtop, my question is about BTT002 + Klipper. How is it? It is worth to upgrade from stock prusa einsy to BTT002? How reliably BTT002 works with Klipper? All features implemented and working properly?

Hi. Yes, BTT002 is worth replacement for obsolete 8bit Einsy Rambo. But I reccomend to print Zaribo electronic case, because in stock box there is conflict with driver's heatsinks.

dzhaparoff, 2021-04-11:
> Hi. Yes, BTT002 is worth replacement for obsolete 8bit Einsy Rambo. But I reccomend to print Zaribo electronic case, because in stock box there is conflict with driver's heatsinks.

Thank you! Yep, i already have big thick reversed Einsy-box with more space for better cable management and RaspberryPI.

Afterschnitzel, 2021-04-11:
> Thank you! Yep, i already have big thick reversed Einsy-box with more space for better cable management and RaspberryPI.

Don't forget to add system fan. I didn't do that and fry BTT002 heater bed mosfet in enclosure.

dzhaparoff, 2021-04-11:
Oh, thanks for the advice! Such a strange thing, original einsy's mosfets are cool enough.