Klipper3d/klipper #3637: ADXL345 Invalid ID

BehemothSensei, 2020-12-15:
Hi, after i set up the adxl345 to the Rpi and installed all the 
dependency, when i query the probe, i got this message:

Send: ACCELEROMETER_QUERY
Recv: !! Invalid adxl345 id (got 8f vs e5)

In the first time the "got" section was ff, the f1 and now i got 8f.
Can some one help me figure this out?
[klippy.log](https://github.com/KevinOConnor/klipper/files/5695040/klippy.log)

olofolshall, 2020-12-16:
Try (at your own risk) disconnecting the ground wire in between the adxl345 and the pi as weird as it seems I had the exact same issue as you and nothing had worked for me and others until we disconnected the ground wire.

grigi, 2020-12-17:
I see on the Discord there is a discussion about that it's likely electrical interference, and the cable is too long.
Mine is 28awg ribbon cable of ~90cm length, and works fine.

olofolshall, 2020-12-17:
Yeah we thought that it was EMF interference as well but even with a really short cable the problem persisted but the removal ground wire made it work. But still could be emf since Klipper has a pretty high polling rate.

Best Regards
/Olof

dmbutyugin, 2020-12-17:
Well, EMF interference doesn't explain why disconnecting the ground may help some folks. The fact that disconnecting the ground helps sometimes means that something is really, really off in those cases. Unfortunately, most of the folks who solved the problem that way didn't want to investigate it further.

olofolshall, 2020-12-18:
I’ll gladly help to investigate it further, just didn’t have time to do then. I’ll be back with my printer after Jan 10.

Best Regards
/Olof

dmbutyugin, 2020-12-19:
OK, thanks, the help would be appreciated.

dewi-ny-je, 2020-12-21:
Is it a Chinese 2 dollars module or a module from a reputable source like adafruit, costing almost 20 dollars?

Has anyone tested the Chinese modules Vs the reputable ones?

I have a bad experience with Chinese DS18B20 so I wanted to warn you.

dmbutyugin, 2020-12-21:
I have both the Adafruit one, and a some cheap modules from Aliexpress. Both work well. Of course, it is not possible to guarantee that all cheap modules will be good. However, this problem is a bit more peculiar, so it _probably_ cannot be explained by cheap module of a bad quality.

Point is: for the adxl345 to function properly it needs power. Power is delivered via VCC & GND wires together. Basically, one wire isn't enough, there needs to be a difference in voltage and current going in both ways. When the ground is disconnected, the module has no power anymore. It means that the ground is 'delivered' somehow else. And that's what's currently unexplained. But given that it happened to a few folks (when disconnecting the ground helped), it means that whatever happened there can be (accidentally) repeated by several people.

As a one theoretical possibility, maybe the adxl345 gets screwed into some metal part of the printer; and that somehow gets adxl345 board (either one of its pins, or its ground) connected to the metal frame of the printer, which may be connected to earth or the ground of the printer's own PSU, which may have an offset vs the ground of RPi power supply, and this may cause issues. 

But it is impossible to determine this by guessing, it is necessary to look into it in more details.

olofolshall, 2020-12-22:
My mount is screwed into a metal carriage with is connected to the printer with linear rails so it’s definitely a possibility in my case. But if I think I got the same issue even when it was hanging loose. I’ll definitely buy a second module and help with some testing in January. Probably easiest to resolve in discord with maybe a video chat so you can see what I’m doing.

Best Regards
/Olof

chrisger123, 2020-12-24:
I have this issue as well. But I think it is more a software issue.
I have two different ADXL345 now and checked the wiring. But still get 0 as id.

But when I remove the check for the Id in the code, I see changing ids every time, so I think it is still an issue with the parsing the response (similar to the issue for the 65 Bit Pi clone that should be fixed - maybe?).
Even if I'm a developer, this is where my knowledge ends at the moment since I'm normally using Java and the last time I developed something with C was about 20 years ago :-(

(btw. I'm using a Pi 4)

When I use the commandline tool I get values without any problems (with both ADXLs):
So technically and from the wiring point of view it should work.

pi@octopi:~/klipper $ sudo adxl345spi
time = 0.000, x = 0.062, y = 0.172, z = 1.328
time = 0.200, x = 33.004, y = 66.008, z = 72.000
time = 0.400, x = 2.141, y = 36.250, z = 36.141
time = 0.601, x = 64.008, y = 2.141, z = 66.258
time = 0.801, x = 36.141, y = 64.008, z = 36.141
time = 1.001, x = 66.008, y = 36.250, z = 66.008
time = 1.201, x = 36.141, y = 2.141, z = 36.250
time = 1.401, x = 66.008, y = 64.258, z = 2.141
time = 1.601, x = 36.141, y = 36.141, z = 64.008
time = 1.802, x = 66.008, y = 66.258, z = 36.250
time = 2.002, x = 36.141, y = 36.141, z = 2.141
time = 2.202, x = 66.008, y = 66.258, z = 64.258
time = 2.402, x = 36.141, y = 36.141, z = 36.141
time = 2.602, x = 66.008, y = 66.258, z = 66.258

dmbutyugin, 2020-12-24:
> I have this issue as well. But I think it is more a software issue.

No, it is still a hardware issue:

> When I use the commandline tool I get values without any problems (with both ADXLs):
> So technically and from the wiring point of view it should work.
>
> pi@octopi:~/klipper $ sudo adxl345spi
> time = [...]

That tool does not perform the check of the ID, that's why it returns something. But as you see, it reports invalid acceleration values. The values it prints are in `g`, so the first line is OK, but then the second line (time = 0.200, x = 33.004, y = 66.008, z = 72.000) reports absurdly high acceleration of 270'000 mm/sec^2 on X axis, 540'540 mm/sec^2 on Y axis, and 589'680 mm/sec^2 on Z axis. So it is merely a data read error, which is not detected by that tool. And it has these read errors very frequently (almost every time).

So, the ID check simply tests the reliability of the connection. There is no point in disabling it. It is not a parse failure.

dmbutyugin, 2020-12-24:
> My mount is screwed into a metal carriage with is connected to the printer with linear rails so it’s definitely a possibility in my case. But if I think I got the same issue even when it was hanging [...]

Thanks. I'd suggest to use a voltmeter to measure the voltage across several points and report the results. I would at least test the following: first get the adxl345 chip up and running (and measuring the acceleration) by disconnecting the ground. Then measure the following values (making sure not to short anything in the process):

- voltage between the VCC (or 3V3, depending on which one is used for power) pin and the GND pin (which is now disconnected)
- voltage between the VCC and ground wire (coming from RPi or MCU)
- voltage between GND pin and ground wire

robthide37, 2020-12-25:
I had a carriage i was testing that i printed out of cf pc and it would cause prob also just the screws need be small enough to not touch the small resistors. i used foam as a spacer and wrapped plastic on the bolts. I also bought a nuch of the skinny ones that are multi voltage and i saw that they have 5v and 3,3 pins which apparently you need to connect both of them. Anyway the poeple with the ground disconnect i bet if they measure axis noise with the acclerometer in teh air it will be fine but will be very high when mounted. Last is i wonder if the usb cable needs to be blocked to not transmit power on the rpi as that may cuase some type of interference since i saw banding on my paneldue until i did that. I know all over the place but i have been helping alot of people set this up and i also bought a bunch of acceerometers to test them and it does seem like the spi connection acts  as a ground cause the ones with led light up as soon as the you connect the mosi and cs pins at the same time. I have a scope and voltmeter and 5 diff type of 345s and also have multiples of each just to rule out flukes lmk if their is anything you would like me to do

chrisger123, 2020-12-25:
OK. Found my problem. I had only one thing left regarding possible hardware issues - the power supply.
And it was at least my problem. Seems like I had a problem with different ground levels or something due to the external power supply of the PI.
When I used a stepdown module (set to 5.2V) and powered the PI from the printer power supply (that also powers the printers mainboard and so on,) it worked.

dmbutyugin, 2020-12-25:
@chrisger123, so your adxl345 was connected to RPi? And did you notice, did the sensor touch anything metal on the printer? Or how would different ground levels affect it otherwise?

@robthide37 Sorry, I didn't get everything from your post. Some comments for what I understood:

>  i used foam as a spacer and wrapped plastic on the bolts.

Ideally, the accelerometer must be mounted on a stiff mount (to detect all vibrations), so foam as a spacer may not work very well. But screws indeed should be small (e.g. m2.5 or even m2), and one could also use nylon washers.

> i saw that they have 5v and 3,3 pins which apparently you need to connect both of them

If the sensor board has 2 voltage pins, either of them should be connected, but only one is needed. I don't know what happens if you connect the two, probably it would use one of them.

> it does seem like the spi connection acts as a ground cause the ones with led light up as soon as the you connect the mosi and cs pins at the same time.

That's strange. CS should be high when inactive (but it goes low on read, so it cannot work as a power source), and MOSI, well, I don't think it can act like that either. I have [both](https://github.com/KevinOConnor/klipper/blob/master/docs/img/adxl345-fritzing.png) versions of ADXL345, and none of them work if I forget to connect the power (VCC _and_ GND pins) - and I did forget to connect the power a few times. But the mounts I made for ADXL345s insulate them electrically from the printer frame.

> I have a scope and voltmeter and 5 diff type of 345s and also have multiples of each just to rule out flukes lmk if their is anything you would like me to do

So, you can try to measure the voltages across several pins as I wrote in the previous post. You can also test the case when the ADXL345 is 'powered' only by CS and MOSI pins.

chrisger123, 2020-12-26:
@ dmbutyugin, yes, the adxl345 was connected to a PI which used a normal external USB Power Supply. Afterwards I switched to the internal 24V power supply of the printer with a stepdown module.
I don't know exactly what the problem is, but I could imagine that there was maybe a different voltage potential between the PI and the SKR board (which both are connected by a USB cable) or just some kind of supply noise.

thebucketmouse, 2021-01-09:
Same problem here, invalid ID 0. If I disconnect the ground from the sensor I get invalid id FF. Wish I had my oscilloscope on me to diagnose further but it's in storage.

robthide37, 2021-01-09:
> @chrisger123, so your adxl345 was connected to RPi? And did you notice, did the sensor touch anything metal on the printer? Or how would different ground levels affect it otherwise?
>
> @robthide37 [...]

Their are some real crap boards on amazon that have 3.3 and 5v and they use 3.3 signaling and 5v to power the board and they only work if both are connected i was very confused by it as well as i assumed teh 5v was in front of a buck converter

robthide37, 2021-01-09:
> @chrisger123, so your adxl345 was connected to RPi? And did you notice, did the sensor touch anything metal on the printer? Or how would different ground levels affect it otherwise?
>
> @robthide37 [...]

yas i will measure continuity across pins and also i get the led to light as soon as i connect anything even with no ground but to my shock my pc cf printed carriage adds interference my noise goes from around 20 to high 100s its crazy

closed 2021-02-07 by github-actions[bot]

j0seph1387, 2021-02-19:
Hello everybody,
i have struggled with the same problem for the past few days. with a short 20cm dupont cable I occasionally had a connection to the adxl345 with other, longer cables no chance.
yesterday I grabbed an old ethernet cable, cat5, and crimped it (1 x bed -> arduino nano, 1 x extruder -> rpi, both about 60-70cm). I plugged in the cables and there are no more problems with the connection to the two adxl345. I assume that it was / is an emv problem, and the shielded cable is the solution, possibly from the power supply

MCTaylor17, 2021-02-21:
Had this problem on RPi 3B+.  Detatching the ground pin changed id from `0` to `ff` which was still invalid.  Changing from ground pin 20 to ground pin 9 fixed the problem.  YMMV.

lowfat32, 2021-03-26:
Definitely a cable issue for me as well.  I tested connectivity from the pi to the back of the adxl345 and all tested good. But kept getting the wrong ID. I used a 3ft USB2.0 cable that I cut in half. Wasn't good enough I guess. Ended up using a CAT5E cable at about 2.5ft and it works.

JanKolibri, 2021-04-09:
I still struggled with this same issue, but no luck.
I tried the cat5 cable (1,5 Meter) & also the short dupont 10 cm version. Using the 3,3 V or the 5 V from the RPi or the SKR1.3. Did you guys also use the isolating GND of the cat5 for the ADXL345 as GND? Or does using 5,2 V (with stepdown) + 3,3 V both connected do the trick? 
Also - I noticed SDO is suppose to be on logic 1, but it is always 0 (connected independently to 3,3 V). Could this be it?
Thanks for your help in advance.

lowfat32, 2021-04-09:
I'd try w/ just the standard 6 inch dupont jumper cables and see if it you can get it working. Then move on to making a longer cable.

JanKolibri, 2021-04-09:
Thank you. Yes - I tried those already, with no change. 
I tried different voltage sources for 5 V and 3,3 V, using the ones from the RPi and the SKR1.3. I did the same with GND and - as recommended - disconnecting it. Still nothing.
No changes on SDO, SCL, SDA and CS with the RPi. Last thing on the list would be using a buck converter for 5,2 V.

ZorK2K, 2021-04-09:
So I found probably one more solution.
For me worked to disconnect the raspicam.
Then there was no more problem at all.

JanKolibri, 2021-04-09:
Sorry - still nothing.
I attached the current state of my built (Fig.: (A)), with the used ADXL345 (Fig.: (B)) and the used connection with RPi (Fig.: (C))
Disconnecting the RPi-cam didn't do the trick. I also tried disconnecting the fan and the relais for the PSU.
Next thing was changing the Baudrate from 250000 to 115200, because I found some sources that claimed this being the problem. Still nothing.
Any further ideas, please?

[image]

dmbutyugin, 2021-04-09:
FYI, there was recently a case discussed on the Klipper discord, where the clone board in the same form-factor as yours was forced into I2C mode by shorting one of the pins (SDO or SDA?) to GND. The solution there was to unsolder one of the resistors. I suggest you to check that out. And find the schematics of your specific board.

KevinOConnor, 2021-04-09:
> forced into I2C mode by shorting one of the pins (SDO or SDA?) to GND.

FYI, the SDO pin (aka MISO pin) had a pulldown resistor with a low resistance, and thus the host was only ever able to read zeros.  (It seems the device was only intended to use i2c mode.)  Unsoldering that pulldown resistor fixed the issue for that user.

-Kevin

robthide37, 2021-04-09:
Those don't work you need the single row header ones.  I tried prob 6 or 6
types for my patrons and I had no luck with the dual voltage split ones.
Also the dual voltage is supposed let you power with 5v and it will give
you 3.3v power its just a buck converter iirc

JanKolibri, 2021-04-09:
Thank you, you guys! It worked.

I removed the pull-down resistor as suggested (Fig.: (A), 3) and left the connection open (Fig.: (B), 4). The ADXL345 was detected without any hesitation. No disconnecting of GND cables, RPi camera or extra buck converters needed. I tried the short dupont cables first, but the cat5 was no problem either.  

Before I had it running though, I validated the functionality of the unchanged ADXL345 with this tutorial: https://pimylifeup.com/raspberry-pi-accelerometer-adxl345/

Hope this helps for future pitfalls. (ツ)╯

[image]

Deneteus, 2021-04-25:
I want to point out as I have been testing this myself that you can get this ID error if you try to use 3.3v instead of 5v to power the following board over a 1 to 2ft distance. The only issue I saw was that I could get the ADXL345SPI Utility to capture samples but once I did I couldn't get the board to be recognized via Octoprint.  

[image]

[image]

[image]

The instructions left out that entire section about loading the MCU and I had to manually install PIGPIO to get the ADXL345SPI tool to work to verify the sensor was working. https://github.com/nagimov/adxl345spi

[image]

EsserPrototyping, 2021-07-24:
Maybe it helps somebody here..
I have the same blue board mentioned above this post and could not get it running in any VCC combinations (3.3V or 5V) from the PI.
However, in the end it worked with 3.3V VCC - but I had to power the PI with a seperate USB power supply and not via a buck converter which was fed by the printer power supply. I do think the sensor is very sensitive to GNG noise, which could easily come from the buck converter or printer PSU.

michaelbeljaars, 2022-01-02:
I am facing the same problem as many of you here. I have tried all suggestions, but I have not been able to get my ADXL345 to work with Klipper. I have tried 3 sensors, all of which give me the same error (Invalid adxl345 id (got 0 vs e5).). I tried shortening my cable to only 10 cm, no luck either. Also the suggestion of a different power supply for the Raspberry Pi (I was running mine using a separate 5V power supply, changed to a Rasberry Pi microusb power adapter) didn't make a difference.

I do get representative values using the ADXL345SPI tool (https://github.com/nagimov/adxl345spi), but as stated somewhere, this tool does not check the ID of the sensor. Nonetheless it proves that the SPI communication works and that I am able to read the sensor.

[image]

robthide37, 2022-01-02:
do you have any other peripherals on the rpi ?

michaelbeljaars, 2022-01-02:
> do you have any other peripherals on the rpi ?
> […](#)

No, there are no other peripherals connected to the pi.

Sineos, 2022-01-03:
According to my experience, the error `Invalid adxl345 id (got 0 vs e5)` is a cable issue or a dead chip.
The values shown by `adxl345spi` can be misleading. A good indication that `adxl345spi` is tricking you is a high repetition of the same values on all axes as in your screenshot above.

Another point on the `adxl345spi` tool or maybe even general: It happened on my side that the ADXL or the RPi somehow got stuck and would not work although the cable was OK. On such occasions it was necessary to reboot the RPi. Cutting power to the ADXL did not solve it.

This is a dry-run of my working ADXL:
```
ubuntu@ubuntu ~/adxl345spi $ sudo ./adxl345spi
time = 0.000, x = -0.922, y = 0.336, z = 0.250
time = 0.201, x = -0.875, y = 0.375, z = 0.250
time = 0.401, x = -1.000, y = 0.359, z = 0.258
time = 0.602, x = -0.953, y = 0.297, z = 0.242
time = 0.803, x = -0.922, y = 0.328, z = 0.258
time = 1.003, x = -0.906, y = 0.336, z = 0.211
time = 1.204, x = -0.945, y = 0.367, z = 0.266
time = 1.404, x = -0.898, y = 0.328, z = 0.281
time = 1.605, x = -0.930, y = 0.336, z = 0.258
time = 1.805, x = -0.922, y = 0.289, z = 0.242
time = 2.006, x = -0.906, y = 0.352, z = 0.250
time = 2.206, x = -0.922, y = 0.336, z = 0.273
time = 2.407, x = -0.906, y = 0.352, z = 0.242
time = 2.607, x = -0.891, y = 0.359, z = 0.266
time = 2.808, x = -0.961, y = 0.352, z = 0.281
time = 3.008, x = -0.961, y = 0.336, z = 0.266
time = 3.209, x = -0.922, y = 0.336, z = 0.234
time = 3.409, x = -0.922, y = 0.305, z = 0.242
time = 3.610, x = -0.938, y = 0.352, z = 0.250
time = 3.811, x = -0.922, y = 0.359, z = 0.250
time = 4.011, x = -0.906, y = 0.352, z = 0.227
time = 4.212, x = -0.930, y = 0.328, z = 0.242
time = 4.412, x = -0.891, y = 0.367, z = 0.273
time = 4.613, x = -0.977, y = 0.320, z = 0.289
time = 4.813, x = -0.891, y = 0.375, z = 0.266
25 samples read in 5.03 seconds with sampling rate 5.0 Hz
Done
```

Also see https://github.com/Klipper3d/klipper/pull/5059

michaelbeljaars, 2022-01-03:
> According to my experience, the error `Invalid adxl345 id (got 0 vs e5)` is a cable issue or a dead chip.
> The values shown by `adxl345spi` can be misleading. A good indication that `adxl345spi` is tricking you is a high repetition of the same values on all axes as in your screenshot above.

It seems unlikely to me that all 3 sensors have this problem because of a dead chip. Moreover, it does respond to movement, the values change with moving the sensor. I will look into using a shielded upt cable to see if that makes any difference.

Also, I see repeating numbers in the output that you've shared too. I think this might be due to the limited resolution of the output of the adxl345 tool.

michaelbeljaars, 2022-01-03:
Here you can see how your data is distributed

[image]

dallerman, 2022-01-03:
Have the exact same issue here.

Some shaking of the chip gives this result:
pi@megazero:~ $ sudo adxl345spi
time = 0.000, x = -0.078, y = 0.961, z = 0.430
time = 0.200, x = -0.117, y = 1.359, z = 0.633
time = 0.400, x = -0.062, y = 0.453, z = -0.680
time = 0.601, x = 0.109, y = 1.141, z = 0.664
time = 0.801, x = -0.688, y = 0.367, z = 0.000
time = 1.001, x = 0.047, y = 1.055, z = 0.406
time = 1.201, x = -0.047, y = 0.281, z = -0.812
time = 1.402, x = 0.469, y = 1.578, z = 0.883

In klipper i just get Invalid adxl345 id (got 0 vs e5).
The klipper log for this ecvent is as follows:
API Dump Helper start callback error
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/extras/motion_report.py", line 39, in _start
    self.startstop_cb(True)
  File "/home/pi/klipper/klippy/extras/adxl345.py", line 426, in _api_startstop
    self._start_measurements()
  File "/home/pi/klipper/klippy/extras/adxl345.py", line 376, in _start_measurements
    % (dev_id, ADXL345_DEV_ID))
CommandError: Invalid adxl345 id (got 0 vs e5).
This is generally indicative of connection problems
(e.g. faulty wiring) or a faulty adxl345 chip.
Invalid adxl345 id (got 0 vs e5).
This is generally indicative of connection problems
(e.g. faulty wiring) or a faulty adxl345 chip.

michaelbeljaars, 2022-01-03:
ok, so I have just upgraded mi Raspberry Pi 3B+ using `sudo rpi-update` with the latest firmware and it turns out that that solved the problem. Klipper now detects the ADXL345 without problem and I am calibrating resonances as I typ

c0m3d1an, 2022-02-15:
CAT5 Solved it for me!!!

acunningham-bard, 2022-03-12:
CAT5 wired as described also solved my issues.

Opzet, 2022-03-24:
Reliability may be due to adxl345 floating pins.. Datasheet makes mention of this on page 10. 
https://www.analog.com/media/en/technical-documentation/data-sheets/ADXL345.pdf

[image]

More detail here on breakout and carrier pcb's

https://www.best-microcontroller-projects.com/adxl345.html

            CS is pulled high by a 10k resistor (I2C active). Can be 4k7 on some boards.
            ADDR(SDO) is pulled low by a 4k7 resistor.
            SCL is pulled high to 3V3 by a 4k7 resistor.
            SDA is pulled high to 3V3 by a 4k7 resistor.
            VDDIO is connected to VS and both connect to 3V3.

[image]

MaGe-1, 2022-04-20:
I had this issue, too.

Solved with the following method:
I used a twisted pair cable (cat 5).

I used as twisted pair:
vcc/gnd
Mosi/miso
CLK/CS

I think, that you get EMI issues from the clk in other combinations.

[michaelbeljaars](https://github.com/michaelbeljaars) solution should be better, if you have enough space for cables.

dmit2k, 2022-05-27:
> Solved with the following method: I used a twisted pair cable (cat 5).
> 
> I used as twisted pair: vcc/gnd Mosi/miso CLK/CS
> 
> I think, that you get EMI issues from the clk in other combinations.

I can confirm that in most cases this problem is caused by CLK interference. Had exactly the same while using flat unshielded ribbon cable. Once I separated the CLK wire from other wires that worked fine and Klipper detected the chip.

HaroldSP, 2022-05-27:
Hi everyone!

I had the same issue, and instead of twisting cables or adding some extra isolation between the screws and the chip,
I saw that I have my raspberry pi 4 case fan being connected to the 4th and 6th pin.

So I have tried the easiest first step - disconnecting a case fan - and it worked!

Hope you are all doing fine and gonna make this eventually work.

besselfunct, 2022-08-08:
Klipper Version: v0.10.0-278

I don't really want to re-open this issue, but I've reached the limit of my troubleshooting skills.

I have a nice ADXL345 board from Adafruit (specifically [this one](https://www.adafruit.com/product/163)).

I continue to get the `Invalid adxl345 id (got ff vs e5).` error after the following:

* Wired using CAT5e cable, with twisted pairs {SCL+CS; SDA+SDO; VCC+GND}
* Removing the ground
* Updating the RPi firmware
* Quadruple-checking wiring
* Using a very short cable (< 20cm)
* Using +5V instead of +3.3V

Moreover, when I hook the sensor up to an Arduino Uno, I get reasonable values.

I have also tried using this [python script](https://github.com/wrh2/ADXL345) to read the ADXL345 chip, and I get the following output:
```
XYZ: [0.0, 0.0, 0.0]
XYZ: [-9.220419, 0.0, -2.142504]
XYZ: [-9.258678, 0.038259, -2.142504]
XYZ: [-9.335196, 0.0, -2.142504]
XYZ: [-9.220419, 0.114777, -2.142504]
XYZ: [-9.258678, 0.076518, -2.142504]
XYZ: [-9.220419, 0.076518, -2.1807630000000002]
XYZ: [-9.258678, 0.114777, -2.142504]
XYZ: [-9.18216, 0.076518, -2.065986]
XYZ: [-9.258678, 0.038259, -2.142504]
XYZ: [-0.726921, 0.535626, 11.554218]
XYZ: [-2.525094, 0.191295, 8.952606000000001]
XYZ: [-1.071252, -0.459108, 9.296937]
XYZ: [0.076518, 0.306072, 9.641268]
XYZ: [1.798173, 0.153036, 9.526491]
XYZ: [0.459108, 0.267813, 9.603009]
XYZ: [0.38259, 0.267813, 9.641268]
XYZ: [0.38259, 0.306072, 9.603009]
XYZ: [0.38259, 0.306072, 9.603009]
XYZ: [0.344331, 0.267813, 9.679527]
XYZ: [0.42084900000000003, 0.38259, 9.641268]
XYZ: [0.42084900000000003, 0.229554, 9.679527]
XYZ: [0.42084900000000003, 0.306072, 9.641268]
XYZ: [0.306072, 0.306072, 9.679527]
XYZ: [0.459108, 0.306072, 9.641268]
XYZ: [0.38259, 0.267813, 9.641268]
XYZ: [0.42084900000000003, 0.267813, 9.603009]
XYZ: [0.38259, 0.306072, 9.603009]
XYZ: [0.42084900000000003, 0.344331, 9.641268]
XYZ: [0.38259, 0.344331, 9.603009]
XYZ: [0.38259, 0.306072, 9.641268]
XYZ: [0.38259, 0.306072, 9.641268]
```
This indicates to me that the chip is not bad, nor is the SPI hooked up incorrectly.

An excerpt of my `printer.cfg`:

```
[mcu rpi]
serial: /tmp/klipper_host_mcu

[heater_fan my_nozzle_fan]
pin: rpi:gpio16
heater: extruder
heater_temp: 50.0
fan_speed: 1.0

[adxl345]
cs_pin: None

[resonance_tester]
accel_chip: adxl345
probe_points:
    100, 100, 20 # an example
```

I can confirm that the `[heater_fan]` macro is working, so it seems like the Raspberry Pi MCU is configured correctly.

Am I missing something obvious here?

JanKolibri, 2022-08-08:
No worries. Thanks for asking. (^‿^)╯
Is the shielding of the CAT cable on ground too?

besselfunct, 2022-08-08:
When you say shielding, are you referring to anything other than the plastic outer sheath? That's not connected to ground in my case. I have my ADXL345 in a 3D-printed bracket currently which should be non-conductive, and I'm powering my Pi with an external power brick separate from my printer.

JanKolibri, 2022-08-08:
If you are using a CAT5 cable, those twisted cables are wrapped in a foil, which is part of the shielding. The foil is shortend with an unisolated separate cable without any isolation. This component prevents the electromagnetic interaction between the signals.

Gilabite, 2022-08-08:
@JanKolibri actually thats not quite right. CAT5 is twisted pair and unshielded, CAT5e is shielded twisted pair

besselfunct, 2022-08-08:
I must have used a regular CAT5 then.

I'll try to find a CAT5e cable with appropriate shielding. In the mean time though, why am I able to get data from the sensor using the python script? I would think it would be all/nothing when it comes to EMF for something like SPI.

JanKolibri, 2022-08-08:
@Gilabite Thank you for correcting me (•‿•)

mmlange, 2022-08-08:
> @JanKolibri actually thats not quite right. CAT5 is twisted pair and unshielded, CAT5e is shielded twisted pair
Actually, Cat5e _can_ be shielded, but the main physical difference between 5 and 5e is twists per centimeter.   For a quick overview, see https://community.fs.com/blog/quick-view-ethernet-cables-cat5-cat5e-and-cat6.html

Source: my day job is network security

besselfunct, 2022-08-10:
I'm wondering if I don't get interference when I'm running the python script because I'm polling it infrequently. I found the code for the ADXL in [klipper/klippy/extras/adxl345.py](https://github.com/Klipper3d/klipper/blob/master/klippy/extras/adxl345.py). This seems to indicate that Klipper wants to poll at 3200Hz(?). So I might be fine at lower polling rates, but get EMI when I try to go higher? I'm not super familiar with SPI, so I don't know if this is a reasonable assumption.

JanKolibri, 2022-08-10:
I am not sure, how deep you want to dive into the physics of this. But I can ensure you, EMI at this frequency is reasonable (e.g. skin effect, proximitty effect, induction). (ಠ‿ರೃ You also have to consider; this is an analog signal. So your very best solution would be coaxial cable for each individual wire.

MaGe-1, 2022-08-10:
Can you share a picture?
Is the python Script running on the same device With the exact same cables and Ports? And both Times via spi
? 

You can try add a small cap from vcc to gnd
Or solder an additional ground Wire.

The Signal is digital with gnd reference. Perfect is scl/gnd miso/scl and so on...

besselfunct, 2022-08-10:
Unfortunately I'm working from home today, so I don't have an easy picture to share, but I'll try to add one later.

The python script is running on the same device with the exact same ports, and is running via SPI. No configuration change is made from running `ACCELEROMETER_QUERY` (which fails) and the python script (which appears to work).

I'll see if I can find a capacitor. Something like 10uF?

MaGe-1, 2022-08-10:
Strange, Strange, Strange.
Yeah 10 uF electrolyte With 100n ceramic Sounds reasonable as a test.
Maybe the output voltage does not Stay stable if you repeat it too fast.

Habe you tried a Better cable? For example cat 6 or cat 7.

[24 more replies]