Klipper3d/klipper #3638: BLtouch deploys then goes dark when trying to home, comes up Communication error

trooper55, 2020-12-15:
when trying to run a g28 y0 x0 z0 the printer will home verything but the z. it hangs after deplying the probe it goes from red probe deployed blue light then out. Then sits there no z motor movement at all. 

[klippy.log](https://github.com/KevinOConnor/klipper/files/5695111/klippy.log)

FanDjango, 2020-12-15:
BLTouch original (V3.1?) or clone?
Please remove  `pin_up_touch_mode_reports_triggered = False`, not needed for original BLTouch
Have you worked the stufff in [](https://github.com/KevinOConnor/klipper/blob/master/docs/BLTouch.md)?
The blue light goes out in klipper after commands, because the PWM is shut off until the next command.
Need more info

trooper55, 2020-12-16:
I apologize I am very new to this. any info you need please let me know. I do a lot of mechanical work so i can describe stuff fairly well. So I was able to home all 3 axis and ran just fine with a print minus me not setting the z offset right. I replaced the filament feed tube from the factory to the fancy blue one. I accidently pulled the plug out of the bltouch. I plugged it back in checked for any other issues and found non then started to re do the calibration for the z offset but once trying to home the z nothing. the motors don't even seem to engage. I took out the line of code you suggested and tried again and nothing.  When commanded to home the z I can still turn the z lead screw with no resistance.  As far as the blue light deal i Just feel like i remember the blue light being on solid tell touched before I unplugged it. Sorry Im just grasping at straws I also have an ender3 v2 that I cant communicate with after trying to install klipper so im down to no working 3d printers ahahah my own fault. 

Controller board i think is the factory 4.2.2 board im running on pi 4 2gb.

FanDjango, 2020-12-16:
> I took out the line of code you suggested and tried again and nothing.

I was not expecting that to solve the immediate problem but perhaps later ones

>  When commanded to home the z I can still turn the z lead screw with no resistance

Well, that points to the configuration in klipper for the z stepper being wrong, or the z stepper wiring being wrong. If the z stepper won't move, how should a probe ever get to the point of doing something?

Consider changing the title of the issue to "Z won't home, stepper does not move / engage". It's not the fault of the BLTouch.

trooper55, 2020-12-16:
I reloaded the original printer config file in the directory and still nothing. working on trying to figure out how to test the stepper motors them self.

FanDjango, 2020-12-16:
These are the ones that need to be checked:
```
[stepper_z]
step_pin = PL3
dir_pin = PL1
enable_pin = !PK0
```

trooper55, 2020-12-16:
[stepper_z]
step_pin: PL3
dir_pin: PL1
enable_pin: !PK0
step_distance: .001266
endstop_pin: probe:z_virtual_endstop
position_max: 400
position_min: 0
homing_speed: 10.0

FanDjango, 2020-12-16:
Yes, but are they right?

FanDjango, 2020-12-16:
> So I was able to home all 3 axis and ran just fine with a print minus me not setting the z offset right

Let's go back a step.

It worked before, and now, no more Z movement? So then the pin definitions would not be the problem.

FanDjango, 2020-12-16:
In the log:
`Failed to home z: Timeout during homing`

Ok, I would test removing the `[bltouch]` section (temporarily):
```
[bltouch]
sensor_pin = ^PD3
control_pin = PB5
x_offset = -43
y_offset = -8
z_offset = 0
speed = 3.0
pin_up_touch_mode_reports_triggered = False
```
and put the `^PD3` definition into to stepper_z section instead of the virtual endstop.

I would then remove the plug(s) of the bltouch completely.

This would be a configuration of a normal z-axis with a fake endstop (^PD3). Its a lie, of course, and you should manually move the nozzle up quite far, then turn on the printer, try to home and if it moves, turn it off immediately to avoid damage, because the ^PD3 endstop is fake, just for testing.

If moves now, a problem with the bltouch needs to be checked. If it still does not move, need to test more things.

trooper55, 2020-12-16:
testing now, Recv: // Klipper state: Ready
Send: M105
Recv: ok B:20.6 /0.0 T0:21.2 /0.0
Send: G91
Recv: ok
Send: G28 X0 Y0
Recv: ok
Send: G90
Recv: ok
Send: M105
Recv: ok B:20.6 /0.0 T0:21.2 /0.0
Send: G91
Recv: ok
Send: G28 Z0

14:04:04 Firmware version: v0.9.1-115-g77add956
14:04:04 Connected to host via /tmp/printer @0bps
14:04:05 Firmware version: v0.9.1-115-g77add956
14:04:17 Move out of range: 180.000 180.000 410.000 [0.000]
14:04:55 Klipper state: Disconnect
14:04:59 Klipper state: Ready
14:05:52 Too many consecutive timeouts, printer still connected and alive?

Guessing I have an issue with the board or my stepper motors. great.

trooper55, 2020-12-16:
any easy way to go back to factory firmware quick just to check its not software?

FanDjango, 2020-12-16:
> 4:04:17 Move out of range: 180.000 180.000 410.000 [0.000]

Huh? Its not even trying

trooper55, 2020-12-16:
nope after taking out the probe code and setting ^PD3 probe plugged in or out homes x and y centers to home z then sits

FanDjango, 2020-12-16:
Don't give up too early.

`position_max = 400` is in your Z. Why does he want to go to to 410? Homing upwards? Lemme check a few things...

trooper55, 2020-12-16:
[stepper_z]
step_pin: PL3
dir_pin: PL1
enable_pin: !PK0
step_distance: .001266
endstop_pin: ^PD3
position_endstop: 400
position_max: 400
position_min: 0.0
homing_speed: 15.0

FanDjango, 2020-12-16:
Add a homing dir.

`homing_positive_dir: false`

FanDjango, 2020-12-16:
I see the problem now

FanDjango, 2020-12-16:
#homing_positive_dir:
#   If true, homing will cause the stepper to move in a positive
#   direction (away from zero); if false, home towards zero. It is
#   better to use the default than to specify this parameter. The
#   default is true if position_endstop is near position_max and false
#   if near position_min.

trooper55, 2020-12-16:
14:13:16 Connected to host via /tmp/printer @115200bps
14:13:16 Firmware version: v0.9.1-115-g77add956
14:13:16 Firmware version: v0.9.1-115-g77add956
14:13:26 Klipper state: Disconnect
14:13:29 Klipper state: Ready
14:15:38 Failed to home z: Timeout during homing
14:15:41 Klipper state: Disconnect
14:15:56 Klipper state: Ready
14:16:17 Failed to home z: Timeout during homing

FanDjango, 2020-12-16:
You need to rethink the positions you are defining there.

Is it true that your endstop is 400 AND your pos max is 400?

Describe your printer. You home downwards, otherwise you would not be using the probe

So mine is like this:

```
[stepper_z]
step_pin: P2.2
dir_pin: P0.20
enable_pin: !P0.19
step_distance: .00125
endstop_pin: probe: z_virtual_endstop
position_min: -25
position_endstop: 2.1
position_max: 225
homing_speed: 15
homing_retract_dist: 2.5
homing_retract_speed: 30
second_homing_speed: 2.5
homing_positive_dir: false
```

FanDjango, 2020-12-16:
I suggest:

Your next test with

```
[stepper_z]
position_min: -25
position_endstop: 2.1
position_max: 400
homing_speed: 15
homing_retract_dist: 2.5
homing_retract_speed: 30
second_homing_speed: 2.5
homing_positive_dir: false
```

trooper55, 2020-12-16:
[stepper_z]
step_pin: PL3
dir_pin: PL1
enable_pin: !PK0
step_distance: .001266
endstop_pin: ^PD3
position_endstop: 400
position_max: 400
position_min: 0.0
homing_speed: 15.0
homing_positive_dir: false

Im just using the defaults that came with the klipper files. as the worked off the bat.  and working from there.

FanDjango, 2020-12-16:
But not the BLTouch yet. Get ready to turn off the printer

trooper55, 2020-12-16:
14:15:41 Klipper state: Disconnect
14:15:56 Klipper state: Ready
14:16:17 Failed to home z: Timeout during homing
14:19:58 Klipper state: Disconnect
14:20:01 Klipper state: Ready
14:20:03 Must home axis first: 10.000 0.000 0.000 [0.000]

with what you posted

[stepper_z]
step_pin: PL3
dir_pin: PL1
enable_pin: !PK0
step_distance: .001266
endstop_pin: ^PD3
position_min: -25
position_endstop: 2.1
position_max: 400
homing_speed: 15
homing_retract_dist: 2.5
homing_retract_speed: 30
second_homing_speed: 2.5
homing_positive_dir: false

FanDjango, 2020-12-16:
Does your printer home upwards (nozzle towards bed) or downwards (nozzle away from bed)

trooper55, 2020-12-16:
no movement on any of the changes. just x and y movement

FanDjango, 2020-12-16:
Can we do this conversation on discord? I think we have a communication problem

FanDjango, 2020-12-16:
It your printer WOULD WORK WELL, would it home upwards (nozzle towards bed) or downwards (nozzle away from bed)

FanDjango, 2020-12-16:
>  I do a lot of mechanical work so i can describe stuff fairly well

Describe better. You are letting me do a lot guessing

trooper55, 2020-12-16:
sure where have a link to chat.

And everything you make a suggestion I log into and edit the printer.confg file save it. log into octoklipper run the restart command. use the control tab to home x and y axis. then I tell it to home the z axis the head moves to the center of the ender 5 plus and sits there tell it communication errors out. I have taken the electrical connections apart and reseated them in hopes of a lose connection.

FanDjango, 2020-12-16:
https://discord.gg/QsmxAFgn

FanDjango, 2020-12-16:
Spent a lot of time on discord.

Ok, we've cleared up most of the stuff. Most of the above discussion can be ignored and @trooper55 will close this and open up a new one with up-to date info on what is actually going on.

One thing that threw me was "position_endstop" --> being replaced by "z_offset" when you define the `[bltouch]`. 

In the official docs:

```
position_endstop:
#   Location of the endstop (in mm). This parameter must be provided
#   for the X, Y, and Z steppers on cartesian style printers. 
```

and 

```
#homing_positive_dir:
#   If true, homing will cause the stepper to move in a positive
#   direction (away from zero); if false, home towards zero. It is
#   better to use the default than to specify this parameter. The
#   default is true if position_endstop is near position_max and false
#   if near position_min.
```

but specifiying this when you also specify `[bltouch]` section causes:
```
!! Option 'position_endstop' is not valid in section 'stepper_z'
```

KevinOConnor, 2020-12-17:
Okay.  FYI, there is a mention of this at: https://www.klipper3d.org/Config_Reference.html#bed-probing-hardware .  There was also a recent change documented at: https://www.klipper3d.org/Config_Changes.html

-Kevin

trooper55, 2020-12-17:
went back to oem firmware and tried to home nothing. looks like the board chose to die.

closed 2020-12-17 by trooper55