Klipper3d/klipper #3639: z1 stepper doesn't move

Mckevinho, 2020-12-16:
hi,

I'm setting up a klipper config file for my printer but i have a problem.
My Z axis has 2 motors and each motor has his own tmc2130 driver. i use a MKS gen L V2.1 board.
the stepper_z motor moves but the stepper_z1 doesn't move.
i don't think it's an hardware problem because if i change the motor / cable / driver and even the board, i still have the same problem.
But i don't know what i do wrong in the config.

hopefully someone can help me

Kind regards
Kevin

[klippy.log](https://github.com/KevinOConnor/klipper/files/5701477/klippy.log)

FanDjango, 2020-12-16:
Are you sure this is correct (from your config)?
```
[stepper_z1]
step_pin = ar36
dir_pin = ar34
enable_pin = !ar30
```
Have plugged in z1 to the E1 socket?
Why not use the normal pin numbers for the LPC1768 instead of the arduino numbers?
I see there is a sample config for the MKS SGEN L in the config directory. It uses 
```
[stepper_z1]
step_pin = P0.1
dir_pin = P0.0
enable_pin = !P0.10
```
but you need to change ALL of your pin definitions to the Pxxx ones.

Mckevinho, 2020-12-16:
Hi fanDjango,

Thx for reply. 
Yes, I placed the z1 to the E1 socket. 
I need to use the arduino pinnumbers because the board has an atmega2560 chip (see pictures).

[image]
[image]
[image]

FanDjango, 2020-12-16:
Ok, sorry. So many boards... But anyway, the E1 pins used for [stepper_z1], are you sure of them?

Mckevinho, 2020-12-16:
first i've swapped the motors and the cables from socket, again only the motor on the z socket was turning and not the E1 socket.
second i swapped the drivers, same results.
last, i took a second board (same model) and also here was only the z socket working and not the E1 socket.

All the other things like x and y motor and the heaters working fine with the arduino pinnumbers.
on the schematic from the board i read the next numbers for the E1 socket.

EN = D30
STEP=D36
DIR=D34
CS=D21
MISO=D50
MOSI=D51
SCLK=D52

FanDjango, 2020-12-16:
Ok, I agree. checking the tmc2130 stuff now.

FanDjango, 2020-12-16:
Ok, to isolate the problem I have a suggestion:
Just for testing, put the z1 stepper cable into x, and swap the config definitions. That's not much work, just change the section names. It will prove that "2 steppers for Z" will work synchronously (or not). 
Because perhaps CS (E1) is NOT D21? So if Z on Z / Z1 on X works, its not a general problem with two TMC2130 steppers on Z. If that also does not work, we need to look at the TMC2130 code.

Mckevinho, 2020-12-16:
ok, i did the test. so i swapped stepper_x en stepper_z1 in the config. Z1 was not moving.
[klippy.log](https://github.com/KevinOConnor/klipper/files/5702634/klippy.log)

FanDjango, 2020-12-16:
Am I blind or why can't I see the change in [stepper_z1] and [tmc2130 stepper_z1] in the new klippy.log? Is it a new klippy log?

Mckevinho, 2020-12-16:
on the bottom of the log with the last restart firmware, you will see that the first stepper is z1 en the last stepper is x.

FanDjango, 2020-12-16:
blind. ok. sorry

FanDjango, 2020-12-16:
So to recapitulate, also for @KevinOConnor:

You have TMC2130 stepper_z working. You have TMC2130 stepper_z1 NOT moving when plugged into E1.

TMC2130 stepper_x works fine - proving that the pin numbers are ok for x. 

Changing the configs to use the stepper_x position as a second Z stepper as a test gives the same result: The stepper_z1 does not work. Can't be the pin numbers of E1 then.

Normal config stepper_z1 in E1:
```
[stepper_z1]
step_pin = ar36
dir_pin = ar34
enable_pin = !ar30
step_distance = .0025

[tmc2130 stepper_z1]
cs_pin = ar21
spi_software_miso_pin = ar50
spi_software_mosi_pin = ar51
spi_software_sclk_pin = ar52
microsteps = 16
run_current = 0.650
hold_current = 0.450
stealthchop_threshold = 250
```

stepper_z1 plugged into X axis:
```
[stepper_z1]
step_pin = ar54
dir_pin = !ar55
enable_pin = !ar38
step_distance = .0125

[tmc2130 stepper_z1]
cs_pin = ar63
spi_software_miso_pin = ar50
spi_software_mosi_pin = ar51
spi_software_sclk_pin = ar52
diag1_pin = ^!ar3
microsteps = 16
run_current = 0.800
hold_current = 0.500
stealthchop_threshold = 250
```
[image]

To be safe, move the diag1 pin def away, but I don't think that breaks the test.

Do you have any other, old stepper modules laying around that you could use alternatively for further tests, like 4988s or something. Meanwhile I am sure that Kevin O'Conner has already seen this issue.

Mckevinho, 2020-12-16:
i did the test without diag1. same result

FanDjango, 2020-12-16:
I have no new ideas right now.

KevinOConnor, 2020-12-17:
Be sure to follow the steps at https://www.klipper3d.org/Config_checks.html to verify the motor and follow the debugging steps outlined there.  If using a tmc2130 also be sure the DUMP_TMC command ( https://www.klipper3d.org/G-Codes.html#tmc2130-tmc2660-tmc2208-tmc2209-and-tmc5160 ) is able to obtain sane results from the driver.

-Kevin

closed 2021-01-15 by github-actions[bot]