Klipper3d/klipper #3640: Printer not connecting to Klipper Robin Nano v1.2

SpaceEggs24, 2020-12-16:
Hello,
I've been trying to set up Klipper on Octoprint for almost full two days now and nothing seems to work. 
I installed Fluiddpi first and everything was working fine but i am not ready to let go of the Octoprint plugins i used before just yet. 
So i used my old Octoprint install to run with klipper but it did not work. Today i set up a completely new installation of Octoprint and tried everything i could/found on the internet and asked on the discord channel. Still no success. 
Menuconfig was configured as described in the Robin Nano config example. 
Today i tried flashing the .bin file i used with the fluiddpi installation and with this file the printer did beep while trying to connect to it which it didnt with the .bin from my Octoprint installation. 
BUT still no success and the console displayed the following message:

```Unknown command: queue_digital_out
This type of error is frequently caused by running an older
version of the firmware on the micro-controller (fix by
recompiling and flashing the firmware).
Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Protocol error connecting to printer
Klipper state: Not ready

So this is where i'm at now (really frustrated) i hope you can help me..

I attached my most recent logfile. Btw printer.cfg issues are solved i copied the example configuration for the Robin Nano v1.2 to the Pi to make sure its not my config from fluidpi install causing this issue. (Yes i added the serial port under MCU)
[klippy.log](https://github.com/KevinOConnor/klipper/files/5702488/klippy.log)

Thanks in advance.

SpaceEggs24, 2020-12-16:
I tried the Kiauh script now and its the same issue as with Octoprint. Could this be an incompatibility issue of the Robin Nano v1.2 board with the newer versions of Klipper?

SpaceEggs24, 2020-12-16:
I'm also having the issue on the Fluiddpi installation that the Z offset value created by the PROBE_CALIBRATION command doesnt have any effect on the first layer. Otherwise i would just use that installation for now..

ShohninDmitriy, 2020-12-16:
https://github.com/KevinOConnor/klipper/blob/master/docs/Installation.md#building-and-flashing-the-micro-controller

SpaceEggs24, 2020-12-16:
@ShohninDmitriy What about those two links? I have done whats described there multiple times now. Klipper is up to date and i have compiled the firmware as described and flashed it via SD card on my printer. The correct serial port is also in my printer.cfg. I have done this for two days now. Something of the newer Klipper versions doesnt seem to work with the Robin Nano board.

SpaceEggs24, 2020-12-16:
@KevinOConnor I don't know if you have the time but would be really helpful if you could have a quick look at this.

Coffee0297, 2020-12-16:
I can confirm theres an issue from commit 422386e to commit 4c2d074
ie. the robbin takes a flash and works on commit 422386e
and if i upgrade to commit 4c2d074 it dont update the mcu.

/Tonny

KevinOConnor, 2020-12-17:
You'll need to make sure the klipper version for the host matches the klipper version for the micro-controller.  When the host software is restarted (`sudo service klipper restart`) you'll see a line in the klipper log like the following: `Git version: 'v0.9.1-118-g4c2d0749'`.  When the host connects to the micro-controller you'll see a line like the following: `Loaded MCU 'mcu' 77 commands (v0.8.0-625-gf6453983-20200603_230402-Sapphire / gcc: (15:7-2018-q2-6) 7.3.1 20180622 (release) [ARM/embedded-7-branch revision 261907] binutils: (2.31.1-11+rpi1+11) 2.31.1)`.  You will continue to get errors until those two versions match.

There hasn't been any recent changes to the robin or stm32 code, so this sounds like the robin bootloader is just being finicky.  Double check the files being transferred and be sure to closely follow the steps at the top of the config/generic-mks-robin-nano.cfg file.

-Kevin

SpaceEggs24, 2020-12-17:
@KevinOConnor Hello Kevin, thanks for answering! 
Unfortunately i'm not to experienced with Linux commands and coding. 
What would you recommend to do to get the latest build of Klipper to work with the Robin Nano v1.2? Which commands do i have to execute?

SpaceEggs24, 2020-12-17:
@KevinOConnor By the way the issue is that the Board won't even connect and the serial connection attempt always times out as seen in the log.. I'm not sure if i will even get the 
```
Loaded MCU ...
```
line you described but i'll check for that later.

robthide37, 2020-12-25:
He is saying that the stm code did not change so you have a mismatch version. I would do a fresh instal maybe not with kiuah tbh cause the simpler you make things the better . But anyway do a fresh make build and flash with the sd card all from a fresh gitpull and i bet it will work

closed 2021-01-23 by github-actions[bot]