Klipper3d/klipper #3641: Feature request: multiple "Extruder" definitions in a single printer.cfg

axl12ap, 2020-12-17:
Hello,
my printer has 3 different interchangeable "tools", eachone has: an heater cartridge, a sensor NTC type, a different nozzle / throat, material fan, thermistor definition and PID values.

Each time I want change my tool (due to the filaments requirements specs) I nee to edit my printer.cfg and change the section in extruder, editing:

- sensor_type
- pid kp / ki / kd
- nozzle_diameter
- pressure_advance
- pressure_advance_smooth_time

So I tried to define multiple predefined extruder, but there's a problem in klipper because each extruder require a different step motor and different pla_fan pins.

Is there a chance to add something to set at "runtime" these values??

axl12ap, 2020-12-17:
> Hi @axl12ap,
>
> It did not look like there was a Klipper log file attached to this ticket. The log file has been engineered to answer common questions the Klipper developers have about the software [...]

That's not an issue, that's a FR... feature request... I'm sorry for the wrong topic

KevinOConnor, 2020-12-19:
This looks like a dup of #2171 .

-Kevin

axl12ap, 2020-12-21:
> This looks like a dup of #2171 .
> 
> -Kevin

Thanks Kevin for your attention,
and yes.. may be looks like "shared motor for extruder", because the main issue is that... you can have only one motor DEF for only one extruder DEF. But in addition of this request, the material fan definition also must require 1:1 association with the extruder, so this's an addition.

It would be nice to be able to define multiple predefined extruders (with their own parameters) and then activate them when needed with an "ACTIVATE" macro definition (which already exists).

At the moment, however, it is not possible to do this due to the limitations of 1: 1 motor and material fan associations.

closed 2021-01-19 by github-actions[bot]