Klipper3d/klipper #3644: BTT SKR mini e3 v2: Unable to read tmc uart 'stepper_x' register IFCNT

rngSplat, 2020-12-18:
Getting the error "Unable to read tmc uart 'stepper_x' register IFCNT" in klipper. I read this could be a bad driver, so I flashed back to Marlin to confirm and debug. It works as expected in Marlin, but klipper cannot connect. Doesn't seem to be a hardware issue.

[klippy.log](https://github.com/KevinOConnor/klipper/files/5713157/klippy.log)

rngSplat, 2020-12-18:
I tried switching the order of the sections in the config. It appears to be throwing the error on whatever the first one in the config is.

al3ph, 2020-12-18:
Yeah I get this error on and off, usually restarting klipper gets it working, or switching it on and off again.

al3ph, 2020-12-18:
Sounds like the board is just not responding properly.

rngSplat, 2020-12-18:
Flipping it on and off has not helped so far.

If it's an issue of the driver not responding appropriately, why doesn't Marlin see the same issue? TMC debug came back clean.

ThomasW69, 2020-12-30:
I have the same issue on a BTT SKR1.4 Turbo with the BTT TMC2209 v1.2 drivers. For me it looks like Klipper isn't supporting these drivers in a stable way yet.

closed 2021-01-29 by github-actions[bot]