Klipper3d/klipper #3645: Klipper is great but also maddening with errors

JustSid, 2020-12-18:
First up, I really hope I don't come across as super rude. I really like Klipper, but what I find incredibly frustrating is that every time I so much as look at it wrong, it will disconnect and I need to reload the firmware in order to clear error bits and status. I get it from a software development perspective, error handling is tough, especially for things where there isn't a good way to handle it.  I get the philosophy, but I find it taken to the extreme, where it no longer serves a good purpose.

Two examples to that end, the first one is bed levelling. My bed is 310x310mm worth of bent trash. At times, it goes below 0, see also the attached klippy log. Of course, Klipper can not possibly go there because it will run through the end stop. The big problem is that every time the slicer slices a move across such a region, Klipper will throw up its hands and then just bail on me. So then I gotta restart everything and also re-level the bed or clamp the offsets by hand to 0. Why is Klipper even accepting the mesh to begin with?

The second example I have was when I tried to figure out the Z offset for my BL Touch sensor. I started the probing (which, by the way, is a fantastic feature!), but when it came to adjusting the Z offset I wasn't sure exactly what the syntax was to adjust it. So I just typed in `TESTZ ` and... Klipper threw up its hands and decided its done. Is that necessary? I know it wasn't a valid command, but there is a better way to signal that to the end user. To get back to where I was, I had to reload the firmware, home the printer, then move the nozzle back to the middle of the bed.

It's been a week with Klipper, and I really really want to like it. But I haven't been able to do a single print yet, because it always fails on something stupidly small. I wish Klipper wasn't so hostile to the user, because it really makes the experience a frustrating one. It's small things that make software truly great, and I strongly believe user experience is among it. With that out of the way, my question is: Is this a written in stone philosophy? Ie, is looking at Klipper wrong always going to result in an error and there are no plans to provide a better user experience? Or would you accept pull requests to make the UX better in this regard?

[klippy.log](https://github.com/KevinOConnor/klipper/files/5713219/klippy.log)

BlackStump, 2020-12-18:
Are you using Octoprint?

JustSid, 2020-12-18:
Yes

BlackStump, 2020-12-18:
Then did you config it correctly with Klipper as per Klipper Instructions, if not then It is most likely Octoprint that is giving you all the grief with disconnects

JustSid, 2020-12-18:
I followed the [installation guide](https://www.klipper3d.org/Installation.html) on the Klipper website. It didn't seem like there was a lot to configure on the OctoPrint side of things based on that, the only other thing I did was install the OctoKlipper plugin.

BlackStump, 2020-12-18:
imho Octoprint is a pita, not Klipper friendly with it's default settings

JustSid, 2020-12-18:
I have that setting, but it doesn't really change much, does it? The only difference is that I don't have to re-connect to the printer. But it doesn't really change any of the other behaviour, eg. when seeing any bad input giving up altogether. I don't see how this is something that OctoPrint is doing, if all it does is relay what the printer says?

BlackStump, 2020-12-18:
I would suggest to use a webui that has been designed for klipper
[Mainsail](https://github.com/meteyou/mainsail)
[Fluidd](https://github.com/cadriel/fluidd)

JustSid, 2020-12-18:
But my gripe isn't with OctoPrint, but with how Klipper handles errors it encounters. These errors are the same whether I use OctoPrint, Mainsail or access Klippy via bitbanging requests via wet string. Or do these web interfaces do a pre-pass over data passed to Klippy and filter out problematic input so that Klippy doesn't go into error mode?

BlackStump, 2020-12-18:
Klipper does nothing with a unknown command, it states in the console "unknown command" and that command is ignored to my knowledge.

BlackStump, 2020-12-18:
[image]

still connected, still fully functional

JustSid, 2020-12-18:
Alright, this is the unfortunate (for me) bit where I have to admit that I'm a huge idiot. This is OctoPrints fault, not Klippers! Turns out I can't read so I didn't realize I had to restart OctoPrint for the behaviour setting to change. Sooo, you are quite right, this isn't actually putting Klipper into a bad state, just OctoPrint. Still haven't gotten a print done, but I feel much more motivated to further mess with this. Thanks for taking a look at this!

closed 2020-12-18 by JustSid