Klipper3d/klipper #3646: Flashing Alfawise U30 Pro

Julfried, 2020-12-18:
I recently wanted to flash my Alfawise U30 Pro and i was wondering if anyone has done this already. According to the old Marlin Firmware the board was configured as a Ramps 1.3 with an ATMEGA 2560. So i thought this should not be a problem. 

So now I tried and flashed the board, but the connection does not work. I am not sure what I did wrong so maybe someone could help me out? I attached the clippy log file (hopefully I did this correct)

Appreciate your awesome work. Thank you.

[klippy.log](https://github.com/KevinOConnor/klipper/files/5714956/klippy.log)

foreign-sub, 2021-01-05:
Hi,
I think your issue could be:
`CONFIG_SERIAL_BAUD=250000`

Try again using:
`CONFIG_SERIAL_BAUD=115200`

Cheers.

Julfried, 2021-01-05:
Replacing serial = /dev/ttyACM0 solved it for me. 
Thanks for your help.

closed 2021-02-04 by github-actions[bot]