Klipper3d/klipper #3649: Input Shaper Docs - minor issue

DruckiMcDruck, 2020-12-18:
Greetings,

i recently tried to use the ADXL345 on my Rpi to auto-tune the input shaper. However while setting the pi up as secondary mcu, i noticed that the ``make flash`` command would always fail. I've flashed my primary mcu before on this setup and thus applying a simple ``make clean`` before using ``make`` did the trick. 

Maybe you can add this to the docs, as a few users already encountered this issue?

KevinOConnor, 2020-12-18:
FYI, with recent build changes, a make flash should no longer be needed when changing architectures (commit 85300a4).  If you still see a problem with the very latest code then you should copy-and-paste the exact build output here.

-Kevin

DruckiMcDruck, 2020-12-23:
I tried this again on the newest build (was using a old about ~2 weeks old before). Looks like the issue ist gone 👍 Closing here, as it seems it is now obsolete.

closed 2020-12-23 by DruckiMcDruck