Klipper3d/klipper #3650: ADXL345 on a SKR E3 Mini V1.2 - how to flash the Board

AcJoker2, 2020-12-18:
Hi there,

I'm trying to get an ADXL345 to work on a SKR E3 Mini v1.2.

Unfortunately I don't get any further with the instructions.

1. According to the example for the SKR, I have to "GPIO pins to set at
 micro-controller startup "to"! PC13 "."
If I do that, I get this error when compiling with "make" ("make flash" does not work because you have to flash via SD).

  Building out / compile_time_request.o
Unknown initial pin 'PC13'
make: *** [Makefile: 88: out / compile_time_request.o] Error 255

If I don't enter anything instead of "! PC13", it works.
But I don't get any Klipper.bin to flash via SD.

If I keep doing this, Klipper will run

2. According to the instructions, I have to enter that

"[adxl345]
cs_pin: rpi: None "

The firmware no longer starts with this value
This message comes up

"Unknown value 'none' in enumeration 'pin'
This type of error is frequently caused by running an older
version of the firmware on the micro-controller (fix by
recompiling and flashing the firmware).
Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Protocol error connecting to printer
Klipper state: Not ready
Unknown value 'none' in enumeration 'pin' "

It seems to be due to an old / wrong firmware on the board.
But how can I flash the board? At point 1 no hex file is generated.

If I enter GPIO21 (or another one), everything runs normally at first.
But the results of the sensor don't make sense.

Test the sensor

"Send: ACCELEROMETER_QUERY
Recv: // adxl345 values ​​(x, y, z): -262151.367800, -262151.367800, -262151.367800 "

It would be nice if there was a way to get the ADXL345 to work on a SKR E3 Mini V1.2

Thank you and best regards
joker

AcJoker2, 2020-12-18:
Log is now added

Board freshly flashed, but with the usual SKR E3 Mini settings.

Restart, configured with GPIO21
Connection works
GPIO Config changed to none, as in the instructions
Firmware cannot be loaded

dmbutyugin, 2020-12-18:
@AcJoker2 Can you try to sync to the latest code? There were some fixes in the build rules. Then I would suggest you to make a firmware and flash it in the following order (not sure if the order matters, but just to give it a try):

- SKR E3 Mini
- Linux process (for RPi)

`rpi:none` should work, it is the correct way.

AcJoker2, 2020-12-19:
Hi dmbutyugin,

thank you for your help.
Just to be sure, you mean bringing Klipper up to date as described here?

https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#how-do-i-upgrade-to-the-latest-software

I did that on December 15th including the Octoprint update to Python 3, but I can try again.

What do you mean by flashing the SKR E3 mini first and then the Linux process?
I can use

cd ~ / klipper /
make menuconfig
sudo service klipper stop
make

to create a klipper.bin and then flash it to the printer via SD.
But I don't see the Linux process flashing here. What did you mean?

Unfortunately, there is already the first problem with make menuconfig.
What do I do with this setting?
"GPIO pins to set at micro-controller startup" to "! PC13". "
And then no Klipper.bin is generated that I could flash to the printer via SD.

Or do you mean I should first create a firmware for the board as usual and then switch to the Linux process and thus only flash the Pi?

AcJoker2, 2020-12-19:
Additional information

On December 15th I updated Octoprint and Klipper to Python 3.
But now I still get warnings about Python 2.7.
Is that because Python 2.7 AND 3 are installed?

https://octoprint.org/blog/2020/09/10/upgrade-to-py3/

dmbutyugin, 2020-12-19:
> Just to be sure, you mean bringing Klipper up to date as described here?
> 
> https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#how-do-i-upgrade-to-the-latest-software

Yes, and please do that again now.

> What do you mean by flashing the SKR E3 mini first and then the Linux process?

You should do

```
cd ~/klipper/
make menuconfig
```

Select all the parameters for your SKR E3 mini (including ! PC13 pin), then

```
sudo service klipper stop
make
```
 
> to create a klipper.bin and then flash it to the printer via SD.

Then, you basically need to do [this section](https://github.com/KevinOConnor/klipper/blob/master/docs/RPi_microcontroller.md#building-the-micro-controller-code): run `make menuconfig` again, but this time choose 'Linux process' (I suppose you will need to remove that ! PC13 pin config at this point, because RPi doesn't have and doesn't need it), and then run
```
sudo service klipper stop
make flash
sudo service klipper start
```

This will create a microcontroller binary for your RPi and will "flash" it (just copy it to the appropriate location). Then you can reboot the RPi and it should work from that point.

AcJoker2, 2020-12-19:
So, everything set up again.
New SD, fresh Octroprint, fresh Klipper.
Everything works.
I can also see the files spidev0.0 and spidev0.1 under / dev /
Therefore, SPI should be active.
Host_MCU now seems to run properly with Klipper, because I don't get an error message with the settings described in the instructions, Klipper is running.
I find / tmp / klipper_mcu_host

BUT, "ACCELEROMETER_QUERY" still does not have correct values.

Send: ACCELEROMETER_QUERY
Recv: // adxl345 values (x, y, z): -262151.367800, -262151.367800, -262151.367800
Recv: ok

How can I test the sensor?

I bought this sensor.

https://www.amazon.de/gp/product/B07DJ4V5VH/ref=ppx_yo_dt_b_asin_title_o04_s00?ie=UTF8&psc=1

AcJoker2, 2020-12-19:
It works :-)

First I checked the cabling, everything ok.
Since I still had a second sensor here, I connected it directly with short jumper wire. And now everything works.

Before that I had used a LAN cable and soldered jumper wire to the ends. I checked the cable afterwards and it seemed ok.
Maybe it was too long for SPI too?
I can't find any reliable information about the maximum line length with SPI.
If there is a short limit, maybe that could be included in the description.

Sorry for the trouble, the error might have been the update to Python 3 or an error from me because of the lack of clarity regarding flashing.
And then a non-working sensor.

[image]
[image]

dmbutyugin, 2020-12-19:
Since you've been getting some bogus repeatable numbers, it would seem that either the sensor or the connection was broken. Good that it's been fixed. LAN cables typically have twisted pairs, so I wonder how you have connected different wires within the same pair. Anyway, it might be that the cable had some issues rather than being too long. Anyway, I wouldn't go above 1 meter for sure.

Separately, the automatic suggestions seem OK.

rlatimer4, 2021-01-12:
Sorry to revive an old Issue.  But I wanted to post this for anyone googling this error, and thinking you just need to start fresh.  I recently went down this rabbit hole, and discovered that I had "cs_pin: rpi:none" instead of "cs_pin: rpi:None" and that fixed it.   Check that before you do a fresh OctoPi install.

closed 2021-02-12 by github-actions[bot]

Thelvaen, 2021-07-18:
> It works :-)
> 
> First I checked the cabling, everything ok.
> Since I still had a second sensor here, I connected it directly with short jumper wire. And now everything works.

Hey, I don't want to reopen the issue, but would you mind sharing the wiring you used for that?

gbkwiatt, 2022-01-05:
Also your config ?  
I am trying to create config for SKR MINI E3 V2, the only difference is that PA13 is PA14. but rest should work. 
However I am struggling to find any documentation on how to use on board SPI1 instead of RPI SPI.

vajonam, 2022-05-09:
On an BTT SKR Mini V1.2 I was able to get it working 

`PC7` is a unused neopixel ping for chip select pin

```
[adxl345]
axes_map: x,-y,z
cs_pin: PC7
spi_bus: spi1
```

[image]

Wiring  based on pinout above. 

| ADXL|MCU |
|---|---|
|SDO | MISO|
|SDA | MOSI|
|SCL | SCK| 
|5V | 5V|
|GND | GND|
|CS | PC7|

However I did notice that that the MCU didn't boot up with the ADXL connected, it seems to work with it disconnected and then once it starts up and get the `klipper: ready`, you can connect up the ADXL345 and the `ACCEL.._QUERY` should return correct.

vajonam, 2022-05-09:
@Thelvaen  see above