Klipper3d/klipper #3651: Tripping out at higher temperatures.

Hairyloon, 2020-12-18:
I'm trying to print with PETG, which wants a higher temperature than I usually use, and it keeps tripping out and complaining that the extruder is not heating at the expected rate. That would normally suggest to me a hardware problem, except that the temperature graph on Octoprint looks perfectly normal to me... apart from the bit where it nosedives after it switches itself off...
It does it quite consistently above about 212.
[klippy.log](https://github.com/KevinOConnor/klipper/files/5718447/klippy.log)

grigi, 2020-12-19:
I had an issue on my Mk8 hotend where the 30W heater was struggling to keep up. By doing a PID tune at a higher temperature it improved things at higher temperatures.
so, e.g. pid-tune at 230C

Hairyloon, 2020-12-19:
I'm trying PID calibrations, to see if it helps, if only for the diagnostic and this is a plot of one cycle which completed against another where it shutdown.
[image]

Hairyloon, 2020-12-19:
> 
> 
> I had an issue on my Mk8 hotend where the 30W heater was struggling to keep up. By doing a PID tune at a higher temperature it improved things at higher temperatures.
> so, e.g. pid-tune at 230C

I'd thought of that, but it keeps tripping out. I'm trying it incrementally...

Hairyloon, 2020-12-19:
I'm not sure if this is normal, but the error it gives when it stops is:

> Recv: B:22.4 /0.0 T0:213.7 /222.0
> Recv: // Klipper state: Shutdown
> Recv: !! pid_calibrate interrupted
> Recv: ok

I'd usually expect:

> Recv: // Heater extruder not heating at expected rate
> Recv: // See the 'verify_heater' section in docs/Config_Reference.md
> Recv: // for the parameters that control this check.
> Recv: //
> Recv: // Once the underlying issue is corrected, use the
> Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
> Recv: // config, and restart the host software.
> Recv: // Printer is shutdown
> Recv: !! Heater extruder not heating at expected rate

Which it gives me if I try any other command after it's tripped.

matthieu42morin, 2020-12-19:
What I think would help is checking if the thermistor is doing alright (connection, is the glass bead broken?).
Did you choose the right thermistor when you set it up? 
I got one for free with my heater for E3D Volcano and switched to the old thermistor, because it was probably broken... also reported similar stuff like in your issue. So consider switching thermistors

Hairyloon, 2020-12-19:
It has been working perfectly fine printing PLA and it has worked perfectly fine before, so I am confident that it's the right thermistor, or right enough at least. 
And the thermistor is giving the readings shown in the graph, so it looks like it is working properly.

I have recently re-installed klipper, though it may be coincidence that this problem starts now.

Hairyloon, 2020-12-20:
It turns out that the cooling fan shroud was broken.  I cannot fathom how this gave the symptoms I've been experiencing, except insofar as it could cause erratic cooling, but with no fan, it has just run a PID calibration at 240 without any problem.
I'll leave this open for now, in case it doesn't work when I put it back together...

Am still puzzled by the graphs though...

Hairyloon, 2020-12-20:
OK, so back together and I'm back to having the same problem...

Hairyloon, 2020-12-21:
Either the coincidence pixies are out in force, or this is getting stranger...
I'm running Octoprint in "Safe mode" and it appears to be working fine... I don't understand how that could make a difference to the fault I am observing...

grigi, 2020-12-21:
So, I have been building a new hotend and had a slew of heater not heating at expected rate errors.
This was caused by when the fan turns on, the hotend has an immediate drop of ~5°C, and then when it's almost back to normal temp, it kicks out with the error.

I managed to solve it by adding some impromptu insulation to the hotend.

I feel that Klipper may need to be a little less sensitive on that error?

I have also gotten it on my heated bed in the middle of a print, which is quite annoying.

grigi, 2020-12-21:
You could make it less sensitive by adjusting `check_gain_time` for your hotend: https://www.klipper3d.org/Config_Reference.html#verify_heater

Hairyloon, 2020-12-27:
I may have found a problem, but I remain unconvinced that it is *the* problem. I printed out a new fan shroud and ducts, and in the process of installing them, I found a dodgy connection on the fan, so I have now fixed that. I don't see how that can have caused my symptoms though.
However, I am now running a print with no sign of trouble so far... except that when the cooling fan came on (to cool the print) the temperature started dropping significantly. 
My first thought is to adjust the ducts a bit, but I am wondering if we can calibrate the PID to allow for the cooling fan, bearing in mind that sometimes it is on, and sometimes it it off, but given that it does know which it is.

Sineos, 2020-12-27:
> My first thought is to adjust the ducts a bit, but I am wondering if we can calibrate the PID to allow for the cooling fan, bearing in mind that sometimes it is on, and sometimes it it off, but given that it does know which it is.

I have insulated my hotend with silicone and did the PID calibration with 100% fan on (2x 5015 radial fans). Never noticed a any temperature drops. IMO no need to make it complex via software.

Hairyloon, 2020-12-27:
> I have insulated my hotend with silicone...

Interesting idea. Can you go into more detail? Or perhaps a picture?
Mine simply wouldn't run a PID calibration with the fan on, though I suppose I could change some settings in the config...

Sineos, 2020-12-28:
[image]

[image]

First one is a simply fiber based insulation just wrapped around the HE.
Second one is a silicone insulation in the original shape of the hotend. For many hotends you can buy these pre-made. I did print a casting mold and used liquid silicone to do my own.

IMO it is essential to insulate the HE for fast heat up times and proper PID behavior.

Hairyloon, 2020-12-28:
Super. Could I trouble you for a picture of the other side?

Sineos, 2020-12-28:
TBH, I just took some random pictures from the internet. Especially the pre-made silicon "socks" depend on your HE type.
http://bit.ly/38E3NMJ

Hairyloon, 2021-01-25:
I put the problem aside for a bit and went back to using PLA: it's only an issue above about 225, but I'm trying again to print a thing in PETG so I'm back on this...

I haven't yet tried the silicone sock, partly because I think that might mask the problem and I think I'd like to work out what's actually going on because it has me puzzled and it seems to be getting worse... which may at least make it easier to find the problem.
I was thinking perhaps there is a problem with the thermistor, but I think happens too consistently.

Meanwhile, is there some way to lift the head when it goes into thermal shutdown?
It's easy enough to pick up a print where it stopped, except that if you not quick about it it'll either melt a mess into the work or weld itself to it, or quite often both.
Also, is there some way to stop the messages when it shuts down?
It is usually helpful to see what it was doing before it shut down, but the terminal window fills up too rapidly with the repeating error message.

Hairyloon, 2021-01-25:
I've had a slightly different result today:
[image]
Could that spike be something as simple as noise on the line? If so, what's the best way to deal with it?

Hairyloon, 2021-02-05:
I made a sock for it out of an old silicone HDD enclosure tied on with a bit of wire. It seems to work. 
Doesn't explain the problems, but I'll worry about that if they come back...

closed 2021-02-05 by Hairyloon