Klipper3d/klipper #3652: klipper update lost communication with BTT SKR mini e3 v2

okkman, 2020-12-19:
[klippy.log](https://github.com/KevinOConnor/klipper/files/5718923/klippy.log)

I updated klipper, moonraker, fluid and mainsail and lost communication with my skr mini e3 v2.  The SKR 1.4 still communicates fine but ls /dev/ shows no serial connections when only the mini is connected.  I reflashed a few times trying different communication settings but none have worked.  This communication problem also seems to appear on the e3 turbo with the most recent build of klipper.

MLoprieno, 2020-12-19:
I'm sure I'm in the same boat, similar error message to you. My config is fine, worked well before I downloaded and flashed the latest klipper. It looks like the last commit I had sync'd was d53365f9 back on Oct 13.

[klippy.log](https://github.com/KevinOConnor/klipper/files/5719781/klippy.log)

DMESG:
[214173.795151] usb 1-1.3: new full-speed USB device number 4 using xhci_hcd
[214174.115447] usb 1-1.3: device descriptor read/64, error -32
[214174.335439] usb 1-1.3: device descriptor read/64, error -32
[214174.555281] usb 1-1.3: new full-speed USB device number 5 using xhci_hcd
[214174.655415] usb 1-1.3: device descriptor read/64, error -32
[214174.875389] usb 1-1.3: device descriptor read/64, error -32
[214174.995390] usb 1-1-port3: attempt power cycle
[214175.675211] usb 1-1.3: new full-speed USB device number 6 using xhci_hcd
[214175.675458] usb 1-1.3: Device not responding to setup address.
[214175.895424] usb 1-1.3: Device not responding to setup address.
[214176.115215] usb 1-1.3: device not accepting address 6, error -71
[214176.215215] usb 1-1.3: new full-speed USB device number 7 using xhci_hcd
[214176.215412] usb 1-1.3: Device not responding to setup address.
[214176.435414] usb 1-1.3: Device not responding to setup address.
[214176.655196] usb 1-1.3: device not accepting address 7, error -71
[214176.655489] usb 1-1-port3: unable to enumerate USB device

pi@octopi:~ $ sudo vcdbg log msg
debug_sym: vc_mem_copy: Unable to open '/dev/fb0': No such file or directory(2)

debug_sym: vc_mem_copy: Unable to open '/dev/fb0': No such file or directory(2)
[repeats...]

And I also lost /dev/serial/by-id/usb-Klipper_lpc1768_0C10000F229C3DAF195A9D5CC12000F5-if00

pi@octopi:~ $ ls /dev/serial*
/dev/serial0  /dev/serial1

Arksine, 2020-12-20:
Did you run "make clean" before the build?  If not, do so, rebuild, reflash, and see if it helps resolve the issue.

okkman, 2020-12-20:
yes I did, every time because I have two different boards.

okkman, 2020-12-20:
And I should have mentioned, the update caused no problems at all with the other board, an SKR 1.4.

MLoprieno, 2020-12-21:
So... I've discovered the source of my issue. Quick story: back when I first got my SKr 1.3, I used an arduino to flash the bootloader, then I loaded up the Klipper firmware. In the make menuconfig settings, not knowing which setting to use, I had unchecked 'Target board uses Smoothieware bootloader', figuring that I had already put one on the board. This seemed to work well. This continued to work with Klipper builds in March through May.

Fast forward to December and I decide to update Klipper. I left the menu as-is (bootloader disabled) and it wouldn't boot. Added it back in and now everything works well. This is for both release 9.1 and the current build. Do note that I used an SD card and firmware.bin to flash the device each time - I've never been able to get the build-in flash tool to work.

okkman, 2020-12-21:
I'm glad you got it figured out.  Unfortunately, the smoothieboard option only applies (I think) to the LPC series but in any case, does not apply with the STM32 microcontroller used on the SKR minis.

okkman, 2020-12-22:
some progress.  I had mixed up my boards and configured the GPIO pins wrong as this board is the mini e3 v1.2, not the v2.0 I have in another printer.  Now I get the following:

[image]

closed 2021-01-20 by github-actions[bot]