Klipper3d/klipper #3657: 2 or more Klipper on one Pi

LuckySunny85, 2020-12-19:
Hi, 
I have OctoPrint and Klipper running on a Raspberry with multiple instances. The setup went well. However, I can't control the second printer ( both Ender 3 with a SKR mini) in OctoPrint - /tmp/printer2/ can't be entered or selected as a port. In addition, I can control the first printer from the second instance if I select /tmp/printer/ there.
I have done this several times now with the Pi3b and the Pi4 and always end up with the same error.

[klippy.log](https://github.com/KevinOConnor/klipper/files/5719990/klippy.log)

With kind regards 

LuckySunny

BlackStump, 2020-12-19:
[did you read this section?](https://github.com/KevinOConnor/klipper/blob/master/docs/FAQ.md#can-i-run-multiple-instances-of-klipper-on-the-same-host-machine)

greyhound3, 2020-12-22:
Have you read this http://thomas-messmer.com/index.php/14-free-knowledge/howtos/79-setting-up-octoprint-for-multiple-printers ? Works perfectly for me.

LuckySunny85, 2020-12-26:
The installation is not the problem. The installation went well.
But i dont get a connection to the secound printer. 
In the secound Octoprint it looks like this, i cant take printer2 as a port
[image]
If i delet tmp/printer it looks like this.
[image]

1N4148, 2020-12-26:
Looks like your second instance wasn't started with the right parameter -l /tmp/klippy2.log -I /tmp/printer2

closed 2021-01-24 by github-actions[bot]