Klipper3d/klipper #3660: Filament change based on last used filament

modmt, 2020-12-20:
Hello all,

I use a DIY solution for 6in1 filament changer. I do not use it for printing with different colors, I only use it for change the filaments before printing.
The filament I change with gcode macro, e.g. for slot 4 I use gcode „l_4“, for unload „un_4“.
Works great so far, but I want to save the latest state of used filament because I do not want to unload the filament after each print. 
I see there is an option to save variables in a file, that works for me.
But now I want to check before each print if the last used filament is the right one or there must be a filament change.
I there a possibility to do an if condition in the program or something like this?
Like: If saved value is not the required filament then do a gcode macro?

Another question if it is possible to show the saved variable (number of filament slot)  in octoprint screen?

Thank you for your help!

zellneralex, 2020-12-20:
Yes, there is a way to store variables to a file. Please like here https://github.com/KevinOConnor/klipper/blob/master/docs/Command_Templates.md#save-variables-to-disk for details

modmt, 2020-12-21:
Thank you for your reply. 
Saving the variables works good.
But now I do not know how to make an if condition. 
If filament A is already loaded then do nothing, if filament A is the requested one and filament B is loaded then do the macro for unloading filament B and loading filament A...

zellneralex, 2020-12-21:
Here is an example how a if looks like in a gcode macro

{% if 'temperature_sensor bed' in printer %}
       {% set actTemp = printer['temperature_sensor bed'].temperature %}
    {% else %}
        {% set actTemp = printer.heater_bed.temperature %}
    {% endif %}
    
The syntax is Jinja2 so search for it if you need more details

modmt, 2020-12-22:
Thank you very much.
That information is exactly what I am looking for.

closed 2020-12-22 by modmt